letscontrolit/ESPEasy #4974: [Test build] Testers for new upcoming build (feb '24)

TD-er, 2024-02-12:
I've just uploaded the [latest GH Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/7870689803) [here](https://td-er.nl/ESPEasy/latest/)

So I would like to summon the very appreciated testing crew :)
@fly74 @ghtester @chemmex @alex-kiev @iz8mbw @Wookbert @Dickesplumpel

Just as a reminder:
- ESP32-xx LittleFS builds are based on ESP-IDF 5.x code
- ESP32-xx SPIFFS builds are based on ESP-IDF 4.4

Known issues to be fixed before new release:
- [ ] ESP32 does seem to take upto 2 minutes to connect to WiFi when it needs to scan to (re)connect (mainly on LittleFS builds)

I also would like to know what you guys think about the new download "artifacts" on the GH Actions builds (not the releases) as Ton needed to change stuff due to changes in how GH Actions operates.
I personally like the single large download for uploading it to the webflasher, but I really dislike the download speeds offered by GH. (takes 5 - 10 minutes to download the full zip)

Not sure if Ton even could change it as how it is done now, but it is nice to know if there are some likes/dislikes about the current situation.

fly74, 2024-02-12:
I'll have a look.

TD-er, 2024-02-12:
I can also upload the same zip to my server and link it on the webflasher page.
Then it is probably a lot easier to download.

fly74, 2024-02-12:
> I can also upload the same zip to my server and link it on the webflasher page. Then it is probably a lot easier to download.

Need a little more description.

iz8mbw, 2024-02-12:
Ciao.
Will test later or tomorrow.

fly74, 2024-02-12:
@TD-er But there's not ESP_Easy_mega_20240212_neopixel_ESP32_4M316k_LittleFS in the zip. Is littleFS still in pre tests?

TD-er, 2024-02-12:
Well I'm not 100% sure yet as converting from SPIFFS to LittleFS does require some user action as the file system will be formatted.
So I want to let the users do this only when really needed.
And since I want to create new partition layouts so you can use the MAX builds on any 4MB units, it will require another change/convert action from the user.
So I'm not sure whether I will create LittleFS builds for all current builds, or maybe it is better to wait till this "safeboot" option is ready as this also reduces the number of builds significantly.

fly74, 2024-02-12:
As I remember you said that LittleFS is needed in the future as default?

TD-er, 2024-02-12:
Yep, but since it requires users to perform a significant update step, I want to ask this only once from the users.

So maybe better to combine the update to a LittleFS build with partition resize which is required for "safeboot" build layouts as this allows for MAX builds for all ESP32 builds.

fly74, 2024-02-12:
No issues with:

´´´
./ESP8266_neopixel_ESP8266_4M1M/ESP_Easy_mega_20240212_neopixel_ESP8266_4M1M.bin
./ESP8266_normal_ESP8266_4M1M/ESP_Easy_mega_20240212_normal_ESP8266_4M1M.bin
./ESP8266_climate_ESP8266_4M1M/ESP_Easy_mega_20240212_climate_ESP8266_4M1M.bin
./ESP32-D0WDQ6_normal_ESP32_4M316k/ESP_Easy_mega_20240212_normal_ESP32_4M316k.bin
./ESP32-D0WDQ6_neopixel_ESP32_4M316k/ESP_Easy_mega_20240212_neopixel_ESP32_4M316k.bin
./ESP32-D0WDQ6_normal_ESP32_4M316k_LittleFS/ESP_Easy_mega_20240212_normal_ESP32_4M316k_LittleFS.bin
´´´
until now.

TD-er, 2024-02-12:
Could you test with the LittleFS build to see what happens if you give the command `wifidisconnect` after it has been connected to WiFi for > 5 minutes?
Or check how long it takes to get a WiFi connection after a power cycle?

At least on an ESP32-C3 I sometimes see a crash-reboot when calling wifidisconnect (at least on a "custom" build with IPv6 enabled) and also on any ESP32-platform (LittleFS builds) that it seems to do nothing after performing a WiFi scan until this times out after 2 minutes.

fly74, 2024-02-12:
```
<html>
<body>
<!--StartFragment-->
ESP Board
--
ESP Chip ID: | 13294316 (0xCADAEC)
ESP Chip Frequency: | 240 MHz
ESP Crystal Frequency: | 40 MHz
ESP APB Frequency: | 80 MHz
ESP Chip Model: | ESP32-D0WDQ6
ESP Chip Features: | Wi-Fi bgn / BLE
ESP Chip Revision: | 1.0
ESP Chip Cores: | 2
ESP Board Name: | Espressif Generic ESP32 4M Flash ESPEasy 1810k Code/OTA 316k FS

<!--EndFragment-->
</body>
</html>
```

It reconnect's immediately:

```
00:02:02.634 : (224612) Info   : HTTP: wifidisconnect
00:02:02.640 : (224504) Info   : WiFi : WifiDisconnect()
00:02:02.751 : (232164) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
00:02:02.754 : (232068) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:02:02.875 : (255784) Info   : WiFi : Best AP candidate: #Hidden# xxxxxxxxxxxxxx Ch:6 (-53dBm)WPA2/PSK (bgn)
00:02:02.881 : (255808) Info   : WIFI : Connecting xxxxxxxxxxxxx#Hidden# xxxxxxxxxxxxxxx Ch:6 (-53dBm)WPA2/PSK (bgn) attempt #0
00:02:03.734 : (234036) Info   : WIFI : Connected! AP: xxxxxxxxxxxxxx(xxxxxxxxxxxxxxxx) Ch: 6 Duration: 748 ms
00:02:03.740 : (233988) Info   : WIFI : DHCP IP: xxxxxxxxxxxxxxxxxxxxxxxxx   duration: 25 ms
00:02:03.760 : (234416) Info   : Webserver: stop
00:02:03.765 : (234084) Info   : Webserver: start
00:02:05.987 : (233988) Info   : WD   : Uptime 2  ConnectFailures 0 FreeMem 234132 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
00:02:35.987 : (233680) Info   : WD   : Uptime 3  ConnectFailures 0 FreeMem 233824 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
```

fly74, 2024-02-12:
I don't know if it the release but a button is bouncing:

```
00:10:13.649 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:13.858 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:13.949 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:14.159 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:14.249 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:15.261 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:15.351 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:15.561 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:15.649 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:16.258 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:16.349 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:16.560 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:16.649 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:17.565 : (234776) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:17.649 : (234808) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:17.859 : (234576) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:17.949 : (234608) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:18.059 : (234576) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:18.149 : (234608) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:18.360 : (234576) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:18.449 : (234608) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:19.158 : (234576) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:19.249 : (234608) Info   : SW  : GPIO=23 State=0 Output value=1
00:10:19.462 : (234576) Info   : SW  : GPIO=23 State=1 Output value=0
00:10:19.550 : (234608) Info   : SW  : GPIO=23 State=0 Output value=1

```

TD-er, 2024-02-12:
Do you have anything connected to it? Or configured in ESPEasy (see HW tab, probably the simplest to see whether something is configured to use it)

fly74, 2024-02-12:
Yes, a button with internal pullup.

[image]

TD-er, 2024-02-12:
Internal pull-up isn't that strong.
Does the button have a relatively long cable?
Maybe you can add some small capacitor (e.g. 100 nF) close to the GPIO pin and GND? (or 2 capacitors, one between the pin and 3V3 and one between the pin and GND)

fly74, 2024-02-12:
[image]

Is as short as possible 😎

TD-er, 2024-02-12:
Well adding a small cap over the pins will probably filter out the noise generated by the WiFi activity :)
The breadboard rails are still acting like an antenna.

fly74, 2024-02-12:
Is not that important at the breadboard.

TD-er, 2024-02-12:
Example of what I meant (tested on ESP32-C3, but may also happen on other ESP32 boards)

```

00:36:08.208 : (198164) Info   : WD   : Uptime 36  ConnectFailures 0 FreeMem 198308 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
00:36:23.146 : (194188) Info   : WIFI : Disconnecting...
00:36:23.651 : (198228) Info   : WiFi : WifiDisconnect()
00:36:23.770 : (199344) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
00:36:23.772 : (199244) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:36:23.883 : (199280) Info   : WiFi : Start network scan all channels
00:36:30.098 : (197380) Info   : WiFi : Scan finished, found: 22
00:36:30.101 : (196876) Info   : WiFi : Best AP candidate: Lurch4 CC:CE:1E:13:B7:58 Ch:1 (-43dBm)WPA2/PSK (bgn)
00:36:30.104 : (197524) Info   : WiFi : Added known candidate, try to connect
00:36:30.106 : (197428) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:36:38.218 : (197444) Info   : WD   : Uptime 37  ConnectFailures 0 FreeMem 197588 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:37:08.226 : (197444) Info   : WD   : Uptime 37  ConnectFailures 0 FreeMem 197588 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:37:38.227 : (197444) Info   : WD   : Uptime 38  ConnectFailures 0 FreeMem 197588 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:37:50.110 : (197396) Info   : WIFI : Connecting Lurch4 CC:CE:1E:13:B7:58 Ch:1 (-43dBm)WPA2/PSK (bgn) attempt #0
00:37:50.119 : (197380) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
00:37:51.146 : (196216) Info   : WIFI : Connected! AP: Lurch4 (CC:CE:1E:13:B7:58) Ch: 1 Duration: 1030 ms
00:37:51.148 : (196680) Info   : WIFI : Arduino wifi status: WL_IDLE_STATUS 0 ESPeasy internal wifi status: Conn. 
00:37:51.167 : (195964) Info   : WIFI : DHCP IP: 192.168.10.75 (ESP-Easy) GW: 192.168.10.1 SN: 255.255.255.0 DNS: 192.168.10.1 / 0.0.0.0   duration: 20 ms
00:37:51.183 : (196396) Info   : Webserver: stop
00:37:51.197 : (202208) Info   : Webserver: start
00:37:51.199 : (202208) Info   : mDNS : Starting mDNS...
00:37:51.203 : (195980) Info   : mDNS : Started, with name: ESP-Easy.local
00:37:52.429 : (195044) Info   : WIFI : STA got IP6 fe80::36b4:72ff:fe4c:3708%st4
00:37:52.432 : (195168) Info   : UDP : Start listening on port 8266
00:37:55.428 : (195248) Info   : WIFI : STA got IP6 2a10:3781:218:1:36b4:72ff:fe4c:3708
00:37:55.431 : (195376) Info   : UDP : Start listening on port 8266
```

tonhuisman, 2024-02-12:
Adding a debounce setting of 50 - 100 msec would probably help too...

iz8mbw, 2024-02-13:
The GPIO "flapping" is an old story... see also https://github.com/letscontrolit/ESPEasy/issues/4800

@fly74 what I can suggest to you is to go in "Hardware" page, then scroll down to "GPIO boot states" and set your GPIO PIN (23 in your case) from "Default" to "Input pullup" if when your push button is not pressed and you need voltyage on the PIN or to "Input pulldown" if when your push button is not pressed and you need 0 Volt (ground) on the PIN.
In this way you should enable the ESP32 internall Pullup (or Pulldown) resistor, The internal pull resistor should be enoght to avoid GPIO flapping between 0 and 1.

iz8mbw, 2024-02-13:
> Internal pull-up isn't that strong. Does the button have a relatively long cable? Maybe you can add some small capacitor (e.g. 100 nF) close to the GPIO pin and GND? (or 2 capacitors, one between the pin and 3V3 and one between the pin and GND)

@TD-er Remember that Internal pull-up or Internal pull-down setting works **ONLY** if the "GPIO boot states" (in Hardware page) has been configured to "Input pullup" or "Input pulldwon".

Dickesplumpel, 2024-02-13:
No issues so far on ESP8266-1M-normal and 4M1M-climate. I've sometimes searched and never found "Input pulldown" - Default state?

iz8mbw, 2024-02-13:
go in the Hardware page and there is the possibility to choose the Pin mode:

[image]

Dickesplumpel, 2024-02-13:
Sorry, my mistake, I've searched on ESP8266 device which I often use with motion sensors in ceiling lights and there is no "Input pulldown".

TD-er, 2024-02-13:
ESP8266 does not have multiple pull resistors per pin.
It is pull up except for 1 pin which has a pull down and I think GPIO-16 has neither.

TD-er, 2024-02-13:
> > Internal pull-up isn't that strong. Does the button have a relatively long cable? Maybe you can add some small capacitor (e.g. 100 nF) close to the GPIO pin and GND? (or 2 capacitors, one between the pin and 3V3 and one between the pin and GND)
> 
> @TD-er Remember that Internal pull-up or Internal pull-down setting works **ONLY** if the "GPIO boot states" (in Hardware page) has been configured to "Input pullup" or "Input pulldown".

That might something to look into before next build as it sounds like a bug.

iz8mbw, 2024-02-13:
@TD-er Yes, it since September 2023 (https://github.com/letscontrolit/ESPEasy/issues/4800) that I shared my experience about the relationship that exists between the "Internal PullUp" flag of the "Switch input - Switch" and the ESP32 Internal Pull-up and Pull-down GPIO state.

As I wrote maybe can be enought (at least for ESP32 boards) to add a note in the HTML web page just down the "Internal PullUp" flag of the "Switch input - Switch" that this will works only if the "GPIO boot states" (in Hardware page) has been configured to "Input pullup" or "Input pulldown".

TD-er, 2024-02-13:
Yep (now) I remember.
And also why we parked it for now as it does involve quite some code changes to do it right.

- ESP32 does allow to set a pull strength, which is not yet implemented
- There should not be a hard set relation between boot state and task setting as you should not need to reboot to make something work
- It should add those for each GPIO pin you can setup in any task where now a single "enable pull-up" checkbox is present.

However adding a note should be (for now) a good quick fix.

Dickesplumpel, 2024-02-13:
My ESP32 C3 mini is running since an hour without WiFi disconnections. I've set up a "disconnect counter" so I can see whats happening in the next hours.

TD-er, 2024-02-13:
@Dickesplumpel 
Probably something like this (without the stack decoding)

```
00:06:58.222 : (186348) Info   : WIFI : Disconnecting...
00:06:58.242 : (191120) Info   : WiFi : WifiDisconnect()
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40043d16  RA      : 0x4211b0fe  SP      : 0x3fca0100  GP      : 0x3fc8b600  
=> 0x40043d16: ?? ??:0
=> 0x4211b0fe: ieee80211_output_do at ??:?
TP      : 0x3fc76968  T0      : 0x40386e1a  T1      : 0x40387ce0  T2      : 0x6e617073  
=> 0x40386e1a: vPortEnterCritical at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:323
=> 0x40387ce0: gpspi_flash_ll_set_dummy at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32c3/include/hal/gpspi_flash_ll.h:352
      (inlined by) spi_flash_hal_gpspi_configure_host_io_mode at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/spi_flash_hal_common.inc:140
S0/FP   : 0x3fcdf86c  S1      : 0x3fce0000  A0      : 0x00000000  A1      : 0x3fcb3d9a
A2      : 0x3fca012c  A3      : 0x3fca04b0  A4      : 0x3fc96000  A5      : 0x00000000
A6      : 0x3fc9293c  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x3fc93e88
S4      : 0x00000000  S5      : 0x00000036  S6      : 0x3fcb3d9a  S7      : 0x00000000
S8      : 0x00000006  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fcadb54  T4      : 0x00000006  T5      : 0x00000000  T6      : 0x00000001
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000018
=> 0x40380001: _vector_table at ??:?
MHARTID : 0x00000000

Stack memory:
3fca0100: 0x3fc76968 0x40386e1a 0xffffffe0 0xffffffff 0x00000000 0x3fce0000 0x3fce0000 0x4211b0e2
=> 0x40386e1a: vPortEnterCritical at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:323
=> 0x4211b0e2: ieee80211_output_do at ??:?
3fca0120: 0x40386e4c 0x4211f7e0 0x3fca01a0 0x00000000 0x3fc76968 0x40386e1a 0x4038771e 0x00000000
=> 0x40386e4c: vPortClearInterruptMask at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:385
=> 0x4211f7e0: xQueueGenericSend at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:882 (discriminator 4)
=> 0x40386e1a: vPortEnterCritical at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:323
=> 0x4038771e: i2c_ll_master_disable_tx_it at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/esp32c3/include/hal/i2c_ll.h:889
      (inlined by) i2c_hal_master_handle_tx_event at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/hal/i2c_hal_iram.c:24
3fca0140: 0x3fc96000 0x3fcadb54 0x3fc92960 0x3fca2c28 0x3fca2b50 0x3fcb3d74 0x3fca2aa8 0x420b756e
=> 0x420b756e: esp_netif_dhcps_option at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:2353
3fca0160: 0x00000ffb 0x00000000 0x00000000 0x00000000 0x3fca2b50 0x3fcb3d74 0x00000000 0x420b34b8
=> 0x420b34b8: ethernet_output at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c:272
3fca0180: 0x3fc96000 0x3fcadb54 0x3fca2c28 0x3fcb3d74 0x3fc92954 0x3fc92960 0x3fca2b50 0x420ad364
=> 0x420ad364: etharp_output_to_arp_index at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:763
3fca01a0: 0x3fc96000 0x3fcadb54 0x3fca2b50 0x3fcadb6c 0x3fcb3d74 0x3fcadb6c 0x3fca2b50 0x420ad5b6
=> 0x420ad5b6: etharp_output at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/etharp.c:826 (discriminator 1)
3fca01c0: 0x3fc96440 0x0000005c 0x3fcb3d6c 0x40388e20 0x3fcb3d74 0x0001804f 0x3fcb3da8 0x420ae5a8
=> 0x40388e20: multi_heap_malloc at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:254
      (inlined by) multi_heap_malloc at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:240
=> 0x420ae5a8: ip4_output_if_opt_src at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:1050
3fca01e0: 0x420ae5d8 0x420a9ad2 0x3fca0260 0x3fc8b600 0x00000000 0x00000000 0x00000000 0x00000000
=> 0x420ae5d8: ip4_output_if_opt_src at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:1017
=> 0x420a9ad2: tcp_output_control_segment at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c:1938
3fca0200: 0xe12a2a1e 0xdb3e2be7 0x3fcb3dbc 0x3fcadb54 0x3fcadb54 0x3fcadb6c 0x3fcb3d74 0x420ae5ce
=> 0x420ae5ce: ip4_output_if_opt_src at /Users/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c:1017
[... cut ...]
3fca0460: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca0480: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fca04a0: 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x3fca01e0 0x000661f1 0x3fc9141c 0x3fc9141c
3fca04c0: 0x3fca04b0 0x3fc91414 0x00000007 0x3fc9f008 0x3fc9f008 0x3fca04b0 0x00000000 0x00000012
3fca04e0: 0x3fc9faa0 0x00546974 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca0490 0x00000012

ELF file SHA256: 0000000000000000
```

Dickesplumpel, 2024-02-13:
No, but I have only looked in the log of the ESP (left page open in browser) and there wasn't any error.

Dickesplumpel, 2024-02-13:
Custom variables on top - great :+1:

TD-er, 2024-02-13:
> Custom variables on top - great 👍

Ah had forgotten about it, but it kinda annoyed me too, so I changed it.
Glad you noticed and also agree with the change :)

fly74, 2024-02-13:
> > Custom variables on top - great 👍
> 
> Ah had forgotten about it, but it kinda annoyed me too, so I changed it. Glad you noticed and also agree with the change :)

What is the change ?

fly74, 2024-02-13:
Can you provide a screenshot ?

TD-er, 2024-02-13:
[image]

See sysvars page

fly74, 2024-02-13:
And where can I create one?

TD-er, 2024-02-13:
using the `let` command in rules.
See [rules documentation](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#internal-variables)

fly74, 2024-02-13:
> using the `let` command in rules.
> 
> See [rules documentation](https://espeasy.readthedocs.io/en/latest/Rules/Rules.html#internal-variables)

I didn't know that. That's nice. I use it much too.

fly74, 2024-02-13:
> @fly74 did you test https://github.com/letscontrolit/ESPEasy/issues/4974#issuecomment-1940653377 ?

I'll test it later.

Dickesplumpel, 2024-02-13:
collection_G_ESP8266_4M1M no issues in my case, ESP32-C3 no wifi disconnections on 7 hours run.

fly74, 2024-02-13:
> The GPIO "flapping" is an old story... see also #4800
> 
> @fly74 what I can suggest to you is to go in "Hardware" page, then scroll down to "GPIO boot states" and set your GPIO PIN (23 in your case) from "Default" to "Input pullup" if when your push button is not pressed you need voltage on the PIN or to "Input pulldown" if when your push button is not pressed you need 0 Volt (to ground) on the PIN. Then reboot.
> 
> In this way you should enable the ESP32 internall Pullup (or Pulldown) resistor, The internal pull resistor should be enoght to avoid GPIO flapping between 0 and 1.

Unfortunately this don't work. After pressing ist stays 1

```
04.942 : (236164) Info   : Webserver: start
05.049 : (235656) Info   : SW   : State 0
05.059 : (235568) Info   : SW  : GPIO=23 State=1 Output value=0
06.153 : (235456) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 235600 WiFiStatus: WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
08.888 : (234516) Info   : SW  : GPIO=23 State=0 Output value=1

```

Dickesplumpel, 2024-02-13:
Why not use an external powerdown? 4-5kOhm to ground?

iz8mbw, 2024-02-13:
Do you have still the State flapping between 1 and 0 as you reported `[here](https://github.com/letscontrolit/ESPEasy/issues/4974#issuecomment-1939569290)`

TD-er, 2024-02-13:
> Unfortunately this don't work. After pressing it stays 1 and don't fall to 0 after release.

How is the button wired?

It is best to have some current flowing when the button isn't pressed.
Since the button is over the GPIO pin and GND, I guess it makes sense to have some physical pull-up resistor between GPIO and 3V3 and set the boot state of that GPIO to pull-down.
The internal pull resistors are of quite high resistance, like 50k - 100k
So if you put a physical resistor of a few k (e.g. 4k7 ... 10k, whatever you have laying around, not less than 1k) between the GPIO and 3V3 then it has a strong pull-up and a weak pull down, so the logic level is high.
When pressing the button you have a strong pull-down (stronger than the resistor) and thus a logic low level.

As this will result in some current flowing through the resistor and the internal pull-down resistor in the ESP, you need a much stronger noise signal to change the pin state.

fly74, 2024-02-13:
But isn't the sense of having internal pullup/-down to avoid external resistors?

fly74, 2024-02-13:
> Do you have still the State flapping between 1 and 0 as you reported [here](https://github.com/letscontrolit/ESPEasy/issues/4974#issuecomment-1939569290)?
> 
> More, when you press the button you provide 3V3 to the PIN or you provide GND to the PIN?

No flapping, but only one switch after reboot is possible.

Wiring is switch provide GND to GPIO23.

Config is this:

[image]
[image]

chemmex, 2024-02-13:
Testing ESP8266, ESP32, C3 and C6 (normal LittleFS builds only for 32-xx). Found some unexpected behaviour: on C3 and C6, could not get http server start captive portal. Neither via Android's internal helper nor directly via browser. Tried with both phome and PC
With 8266 and ESP32 everything as expected
Will check further

tonhuisman, 2024-02-13:
> Found some unexpected behaviour:

Using GH Actions builds or self-built?

Dickesplumpel, 2024-02-14:
> But isn't the sense of having internal pullup/-down to avoid external resistors?
As far as I know: No. In most cases when one of my "constructions" has state-flapping a pull-down or pull-up resistor of 4-5k ends it. You never know what "noise" you are catching from wherever  (power supply, radio over cables, ...). When I remember right the motion sensor HC-SR-501 has a 4k7 pull-down build in.

Dickesplumpel, 2024-02-14:
I remembered wrong,motion sensor HC-SR-501 hasn't  a 4k7 pull-down build in, I've done it. :-)

chemmex, 2024-02-14:
> Probably something like this (without the stack decoding)
> 
> ```
> 00:06:58.222 : (186348) Info   : WIFI : Disconnecting...
> 00:06:58.242 : (191120) Info   : WiFi : WifiDisconnect()
> Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
> ```

Got something like this once when tried to reset the C3 from the web interface. 
```
00:01:24.701 : (204296) Info   : NVS  : Load FacDefPref
RESET: Resetting factory defaults... using default settings
RESET: Warm boot, reset count: 1
RESET: formatting...
RESET: formatting done...
00:01:28.910 : (204360) Info   : SaveToFile: ControllerSettings index: 0
00:01:28.913 : (204360) Info   : SaveToFile: free stack: 7208
00:01:31.015 : (204976) Info   : FILE : Saved config.dat offset: 8192 size: 820
00:01:31.017 : (204912) Info   : SaveToFile: free stack after: 7208
00:01:31.020 : (205000) Info   : NVS  : Load UnitName
00:01:31.021 : (204420) Info   : NVS  : Load WIFI_SSID1
00:01:31.023 : (204368) Info   : NVS  : Load WIFI_PASS1
00:01:31.025 : (204320) Info   : NVS  : Load WIFI_AP_PASS
00:01:31.026fo   : NVS  : Load WIFI_Flags
00:01:31.028 : (204224) Info   : NVS  : Load WiFI_IP
00:01:31.030 : (204176) Info   : NVS  : Load ETH_IP
00:01:31.031 : (204128) Info   : NVS  : Load ETH_HW_CONF
00:01:31.032 : (204088) Info   : NVS  : Load Log
00:01:31.034 : (204040) Info   : NVS  : Load Console
00:01:31.038 : (202992) Info   : ESPEasy console using ESPEasySerial
00:01:31.060 : (202928) Info   : ESPEasy console fallback enabled
00:01:31.064 : (202880) Info   : SaveToFile: free stack: 7336
00:01:33.350 : (203328) Info   : FILE : Saved config.dat offset: 0 size: 3028
00:01:33.353 : (203264) Info   : SaveToFile: free stack after: 7336
00:01:33.355 : (203216) Info   : SaveToFile: freeGuru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40043d16  RA      : 0x4210a776  SP      : 0x3fc9f910  GP      : 0x3fc8b400
TP      : 0x3fc79578  T0      : 0x40386bbe  T1      : 0x40387358  T2      : 0x72617764
S0/FP   : 0x3fcdf86c  S1      : 0x3fce0000  A0      : 0x00000000  A1      : 0x3fcacd46
A2      : 0x3fc9f93c  A3      : 0x3fc9fcc0  A4      : 0x3fc96000  A5      : 0x00000000
A6      : 0x3fc9263c  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x3fc93b78
S4      : 0x00000000  S5      : 0x00000036  S6      : 0x3fcacd46  S7      : 0x00000000
S8      : 0x00000006  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fcad488  T4      : 0x00000006  T5      : 0x00000000  T6      : 0x00000006
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000018
MHARTID : 0x00000000
```
And one more observation: resetting (from wherever) doesn't clean the wifi settings, it still connects to the same AP

iz8mbw, 2024-02-14:
> > Do you have still the State flapping between 1 and 0 as you reported [here](https://github.com/letscontrolit/ESPEasy/issues/4974#issuecomment-1939569290)?
> > More, when you press the button you provide 3V3 to the PIN or you provide GND to the PIN?
> 
> No flapping, but only one switch after reboot is possible.
> 
> Wiring is switch provide GND to GPIO23.
> 
> Config is this:
> 
> ![Screenshot 2024-02-13 214315](https://private-user-images.githubusercontent.com/38754941/304543385-0f7fd03a-b89a-48d7-9c3c-3960bb2a11cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5MDA0MTMsIm5iZiI6MTcwNzkwMDExMywicGF0aCI6Ii8zODc1NDk0MS8zMDQ1NDMzODUtMGY3ZmQwM2EtYjg5YS00OGQ3LTljM2MtMzk2MGJiMmExMWNmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE0VDA4NDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdmNGMzZTJjYzEzMGJkMWY1OWZmN2QyMmQ5OGQ2OTZkOGE2MjEyYjFmYTIwOTE3NDQxZTIyMzMwZWNmY2RkOTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.XmsfYl_xwjc5uBHQXvgE36zyMwEZbTsVvqWA4Efa7gc) ![Screenshot 2024-02-13 214347](https://private-user-images.githubusercontent.com/38754941/304543391-f875c44b-ad3c-4005-9599-2d5fdedd8ed0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5MDA0MTMsIm5iZiI6MTcwNzkwMDExMywicGF0aCI6Ii8zODc1NDk0MS8zMDQ1NDMzOTEtZjg3NWM0NGItYWQzYy00MDA1LTk1OTktMmQ1ZmRlZGQ4ZWQwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE0VDA4NDE1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg2NjczZThmMjQ5NmMzNTdjOGFlYWZiZGE0Yjc5NjhiN2JlODEwNjQ1Y2ZmZjVmODJlMTRiYjViMzMzY2M2ZWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.06vkeHJMsHh3pmYPpIzePQO0eojD5N3bZGxy0dzUBGc)

OK, so the GPIO flapping issue is resolved thanks to the ESP32 internal Pullup resistor.
About your GPIO23, did you set "Input pullp" as show here:
[image]
?

When switch is not pressed, did you check (with a meter) if you provide 3V3 to GPIO23?
When switch is pressed, did you check (with a meter) if you provide GND to GPIO23?

iz8mbw, 2024-02-14:
@TD-er tested ESP_Easy_mega_20240214_custom_ESP32_4M316k_LittleFS Feb 14 2024 and ESP_Easy_mega_20240214_max_ESP32_16M8M_LittleFS Feb 14 2024 with no issues.
Thanks!

TD-er, 2024-02-14:
> And one more observation: resetting (from wherever) doesn't clean the wifi settings, it still connects to the same AP

That might be due to a checkbox on the "Factory reset" page?

[image]

Any checkbox here does store the related settings outside the file system in the NVS partition.
This way a complete format (e.g. when changing from SPIFFS to LittleFS or vice verse) will keep those settings.

chemmex, 2024-02-14:
Every "Keep" was unchecked. And reset command from the terminal did not clear wifi settings either

TD-er, 2024-02-14:
OK, I will check if I missed some erasing of the NVS stored settings.

Dickesplumpel, 2024-02-14:
3 already mentioned issues an old sonoff basic with ESP8266-1M: slow wifi connection, mqtt disconnections every 10-20 minutes and state flapping on gpio14 although there is an 5k pull-down resitor at HC-SR-501. Back to 20231225 everything is working well.

fly74, 2024-02-15:
/latest/:

[image]

[98 more replies]

closed 2024-02-29 by TD-er