letscontrolit/ESPEasy #4976: ESP32 loose Wifi and is not able to reconnect for a while to DECO MESH WLAN

stwa2020, 2024-02-16:
Hello,

ESP32c3 lost WIFI to DECO MESH and can't reconnect, it only works after a pause 

I have the following setup and asked on the TP-LINK DECO forum if anyone else has seen this behaviour in the MESH.
I also have an ESP8266 mega-20211105_c79d675 which seems to have similar behaviour, but I haven't tested it in detail, I can only see that it loses the WLAN and reconnects within a certain time.

The ESP8266 and ESP32c3 have 24x Neopixel and OLED SSD1306 connected, via MQTT are every 4s 4 Values send which will be shown on the Display and LEDs will be turned on/off

I've created a rule which turns on some Neopixel LEDs to see if it's MQTT is disconnected and if WIFI is Disconnected.
I don't know why the WIFI connection got lost but it looks like the ESP did a reset, maybe related to MQTT in combination with QoS 1

WLAN MESH DECO 
=============
Firmware Deco X50 - 1.1.0 Build 20230522 Rel. 68968
Fast Roaming: off
Beamforming: off

DECO OG (Master):	MAC: 50-91-E3-DF-23-E4	Backhaul - LAN    IP: 192.168.1.1
DECO Carport: 	MAC: 50-91-E3-DF-25-24	Backhaul - LAN
DECO UG: 	        MAC: 50-91-E3-DF-21-64       Backhaul - LAN
DECO Keller Zubau:	MAC: 3C-52-A1-49-AB-C0 	Backhaul - WLAN
DECO Keller:            MAC: 50-91-E3-DF-25-24       Backhaul - LAN

I can't find out the WLAN MAC in the DECO APP but the name is ghost and the ESP32c3 find it but can't connect.

ESP32c3 ESP EASY mega-20231225
=======================
STA MAC:	48:31:B7:3D:CD:34
AP MAC:	48:31:B7:3D:CD:35
IP: 	DHCP	
MQTT QoS: 1

After a while the WLAN connection got lost and it's not possible to reconnect, the ESP32 reported(2) Auth expire and try to reconnect but this is not possible and the reason is (1) Unspecified.
I've already tried different Power Supplies. 

ESP32 LOG
==========
[11:55:32]07.292 : (205004) Info  : WiFi : Best AP candidate: ghost 50:91:E3:DF:23:E6 Ch:4 (-65dBm)WPA2/PSK (bgn)
[11:55:32]07.296 : (204892) Info  : WIFI : Connecting ghost 50:91:E3:DF:23:E6 Ch:4 (-65dBm)WPA2/PSK (bgn) attempt #0
[11:55:32]07.310 : (204704) Info  : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
[11:55:32]07.318 : (199888) Info  : Webserver: start
[11:55:34]09.337 : (200056) Info  : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 02.029
[11:55:34]09.344 : (214700) Info  : WIFI : Arduino wifi status: WL_NO_SHIELD 255 ESPeasy internal wifi status: DISCONNECTED
[11:55:34]09.448 : (217952) Info  : WIFI : Connecting ghost 50:91:E3:DF:23:E6 Ch:4 (-65dBm)WPA2/PSK (bgn) attempt #1
[11:55:35]10.575 : (197788) Info  : WD   : Uptime 0 ConnectFailures 0 FreeMem 197936 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
[11:55:54]29.460 : (197572) Info  : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
[11:55:54]29.464 : (197572) Info  : WiFi : WifiDisconnect()
[11:55:54]29.568 : (197508) Info  : WIFI : Disconnected! Reason: '(1) Unspecified'
[11:55:54]29.671 : (197412) Info  : Reset WiFi.
[11:55:54]29.674 : (197348) Info  : WiFi : Start network scan all channels
[11:56:00]35.688 : (196444) Info  : WiFi : Scan finished, found: 12
[11:56:00]35.693 : (197012) Info  : WiFi : Best AP candidate: ghost 50:91:E3:DF:23:E6 Ch:4 (-66dBm)WPA2/PSK (bgn)
[11:56:00]35.696 : (197124) Info  : WiFi : Added known candidate, try to connect
[11:56:00]35.699 : (197140) Info  : WiFi : WifiDisconnect()
[11:56:00]35.803 : (197172) Info  : WIFI : Disconnected! Reason: '(1) Unspecified'
[11:56:00]35.806 : (197076) Info  : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
...

In the DECO log is the reported for MAC:48:31:B7:3D:CD:34 trying to disassociate, but it is not associated

Fri Feb 16 11:54:57 2024 daemon.err hostapd: hostapd_notif_disassoc num_sta 4
Fri Feb 16 11:54:57 2024 daemon.notice hostapd: Station 48:31:b7:3d:cd:34 trying to disassociate, but it is not associated
Fri Feb 16 11:54:57 2024 daemon.notice nrd[13031]: wlanifLinkEventsCmnGenerateDisassocEvent: Client 48:31:B7:3D:CD:34 disassociated on APId 255 ChanId 4  ESSId 0 
Fri Feb 16 11:54:59 2024 daemon.notice nrd[13031]: ar_pat_calc_cand_datarate[line 882]: patrate for 5091E3DF23E4 is 656, txRate: 713.881836
Fri Feb 16 11:54:59 2024 daemon.notice nrd[13031]: ar_pat_calc_cand_datarate[line 882]: patrate for 5091E3DF2524 is 553, txRate: 608.014160
Fri Feb 16 11:54:59 2024 daemon.notice nrd[13031]: ar_pat_calc_cand_datarate[line 882]: patrate for 5091E3DF2164 is 532, txRate: 584.783081
Fri Feb 16 11:54:59 2024 daemon.notice nrd[13031]: ar_pat_calc_cand_datarate[line 882]: patrate for 3C52A149ABC0 is 391, txRate: 543.751465
Fri Feb 16 11:54:59 2024 daemon.notice nrd[13031]: wlanifLinkEventsCmnGenerateDisassocEvent: Client 48:31:B7:3D:CD:34 disassociated on APId 255 ChanId 4  ESSId 0 
Fri Feb 16 11:54:59 2024 daemon.err hostapd: hostapd_notif_disassoc num_sta 4
Fri Feb 16 11:55:00 2024 daemon.err client_mgmt: can not parse json_obj
Fri Feb 16 11:55:00 2024 daemon.err client_mgmt: update_history_wire_type:exit failed.
Fri Feb 16 11:55:00 2024 daemon.err client_mgmt: update /tmp/client_mgmt/history_wire_type failed, offline client type recognization of link priority will be influenced.
Fri Feb 16 11:56:57 2024 daemon.notice nrd[13031]: wlanifLinkEventsCmnGenerateDisassocEvent: Client 48:31:B7:3D:CD:34 disassociated on APId 255 ChanId 4  ESSId 0 
Fri Feb 16 11:56:57 2024 daemon.err hostapd: hostapd_notif_disassoc num_sta 4

When I disconnect the ESP32 device for around 15min, it can reconnect without any issues and after a while it has the same problem again.

Fri Feb 16 15:13:28 2024 daemon.notice hostapd: ath0: AP-STA-CONNECTED 48:31:b7:3d:cd:34
Fri Feb 16 15:13:28 2024 daemon.notice hostapd: ath0: ubus call access_client access '{"mac":"48-31-b7-3d-cd-34","ifname":"ath0","action":"associate"}' &
Fri Feb 16 15:13:28 2024 daemon.notice hostapd: ath0: ubus call client_mgmt access '{"mac":"48-31-B7-3D-CD-34","ifname":"ath0","action":"associate"}' &
Fri Feb 16 15:13:28 2024 daemon.notice nrd[13031]: wlanifBSteerEventsHandleNodeAssociatedInd: Node 48:31:B7:3D:CD:34 associated on APId 255 ChanId 4  ESSId 0 , Capabilities:  , Max bandwidth: 0, Num of spatial streams: 1, PHY mode: 8, Max MCS: 7. Max TX power: 0
Fri Feb 16 15:13:28 2024 daemon.err client_mgmt: wireless client associate or disassociate.
Fri Feb 16 15:13:28 2024 daemon.err client_mgmt: client_mac:48-31-B7-3D-CD-34, client_ifname:ath0, client_action:associate
Fri Feb 16 15:13:28 2024 daemon.err client_mgmt: client(48-31-B7-3D-CD-34) assocaite by if(ath0).
Fri Feb 16 15:13:28 2024 daemon.err udhcpd[12903]: Received DISCOVER
Fri Feb 16 15:13:28 2024 daemon.err udhcpd[12903]: find_free_or_expired_nip
Fri Feb 16 15:13:28 2024 daemon.err udhcpd[12903]: find a new ip: 192.168.1.126, ready to arp check
Fri Feb 16 15:13:28 2024 daemon.err udhcpd[12903]: send arp packet:192.168.1.126
Fri Feb 16 15:13:29 2024 daemon.err udhcpd[12903]: Received DISCOVER
Fri Feb 16 15:13:29 2024 daemon.err udhcpd[12903]: checked a timeout node!
Fri Feb 16 15:13:29 2024 daemon.err udhcpd[12903]: Sending OFFER of 192.168.1.126
Fri Feb 16 15:13:29 2024 daemon.err udhcpd[12903]: Sending ACK to 192.168.1.126
Fri Feb 16 15:13:29 2024 daemon.err udhcpd[12903]: clear ip 7e01a8c0

Thx and kind regards,

Stefan

stwa2020, 2024-02-16:
I've set now Force WiFi B/G=ON and it looks better, I will monitor it for the next few days.

stwa2020, 2024-02-17:
Regardless of the version installed, I have to set **Force WiFi B/G=ON**, even with the test build. With this setting it has been running for 15 hours now without losing a single WiFi connection and the MQTT connection is now stable as well. 

I am using the settings below and it looks stable so far, just in case anyone has the same problems with DECO MESH.

MQTT Controllers
============
Controller Queue:
-Minimum Send Interval: 250ms
- Max Queue Depth: 20
- Max Retries: 5
- Full Queue Action: Delete Oldest
- Allow Expire: ON
- De-duplicate: ON
- Client Timeout: 250ms 
MQTT:
- Will Retain: ON
- Clean Session: OFF

Tools -> Advanced Settings 
- Force WiFi B/G: ON
- Force WiFi No Sleep: ON
- CPU Eco Mode: OFF

TD-er, 2024-02-17:
Minimum send interval should not matter for MQTT controllers, as you keep a connection open.
I guess the only real connection related setting for controllers is the timeout.
As a rule of thumb: 100 - 300 msec is good enough for local servers you need to connect to and set it to 1000 msec for servers "online".

The other settings like queue depth only relate to memory usage on the ESP when the server is not able to process your requests fast enough (or the minimum send interval is too long for the amount of messages you try to send)

The last ones on the Advanced settings page are very likely the ones that make all the difference here regarding stability with mesh access points.

stwa2020, 2025-07-12:
@TD-er 
After running lots of tests, I found that the problem could only be located in the DECO Mesh Network. I posted the details on the TP-Link forum to see if anyone else had experienced similar behaviour, and the support team contacted me directly to provide me with beta firmware. After updating, I ran my known tests and all the problems had gone.

Link to my post in the tp-link forum (no download link for the beta firmware): 
https://community.tp-link.com/en/home/forum/topic/830824

Thanks for you help,

Stefan

TD-er, 2025-07-12:
Thanks for the headsup