letscontrolit/ESPEasy #4981: [FR] Add support for WiFi roaming (802.11k and 802.11v)

TD-er, 2024-02-18:
See [ESP-IDF example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/roaming)

Maybe this also can help users with some WiFi mesh enabled access points.

iz8mbw, 2024-02-18:
GREAT! WiFi roaming is a really good improvement.
I can test since I have 2 APs in my house managed by the Omada Controller and the WiFi roaming is supported.

TD-er, 2024-02-18:
The main reason for me to look into this is not just for the roaming options, but rather to make sure users won't end up with unstable connections as the mesh tries to kick them off  to nudge them to another AP of which they have no idea it may exist.

If this also allows to actually use the roaming, then that's a nice benefit, but first of all we need to get rid of those situations like this:
AP: "I told you to go to the other AP of which I sent you hidden beacon packets"
ESP: "huh?? I will try to reconnect"

And repeat because the ESP never collected those extra packets to see where it might need to reconnect.

iz8mbw, 2024-02-18:
Sure, stability first.

TD-er, 2024-02-18:
Yep and if it adds new features along the way, who am I to complain :)

Jason2866, 2024-03-04:
You have to use a different framework in future for EspEasy. Tasmota Arduino framework will not support 802.11k and 802.11. The option "eats up" resources and has impact on WiFi performance. It does add more harm than it solves.

Jason2866, 2024-05-29:
espressif Arduino 3.0.0 does not include too. It has serious bugs and the wifi stack is not working anymore.

TD-er, 2024-05-29:
OK, so this can be closed then. Thanks for the headsup!

closed 2024-05-29 by TD-er