letscontrolit/ESPEasy #4982: [FR] Home Assistant Auto Discovery

TD-er, 2024-02-19:
Need to have auto discovery for Home Assistant.
So first need to collect some info.

- [initial discussion on Tweakers.net forum](https://gathering.tweakers.net/forum/list_message/78181404#78181404)
- [Tasmota HA integration](https://www.home-assistant.io/integrations/tasmota/)

TD-er, 2024-02-19:
So this would really benefit from a specific publish command to set the retained state of those messages.

sincze, 2024-02-19:
> So this would really benefit from a specific publish command to set the retained state of those messages.

I think I get the point of retained.
1. Option 1, Publish you config on each ... message (ChatGPT came up with that idea.
2. Option 2, Send it to a topic hand have Node-Red forward the topic with retain flag on.
[image]

3. Use automations in HA to scan for the device topic and do the whole autodiscovery in there.
4. Ideally ... look at the developer of ESPEasy for having this automated ;-)

TD-er, 2024-02-19:
I think I'll go for option 4 as that would be the least amount of work for me.... oh wait.... ;)

sincze, 2024-02-19:
> I think I'll go for option 4 as that would be the least amount of work for me.... oh wait.... ;)

Yes I can imagine, I've read through all the forum and noticed your tag attached to it so I thought, first try to solve myself than ask questions later.

sincze, 2024-02-19:
Ok,  one moment, So a lot of inspiration I found here.
https://github.com/enesbcs/shellyteacher4domo/blob/main/mqtt_templates.txt

TD-er, 2024-02-19:
Yep, the idea of using retained messages for discovery/config topics is that Home Assistant can find those again right after connecting to the MQTT broker again (e.g. after a reboot of HA)

sincze, 2024-02-19:
> Yep, the idea of using retained messages for discovery/config topics is that Home Assistant can find those again right after connecting to the MQTT broker again (e.g. after a reboot of HA)

I started cheating a bit (in case mosquitto thinks about 'restarting') using the following config change as well:

```
persistence true
persistence_location /mosquitto/data/
log_dest file /mosquitto/log/mosquitto.log

# Port to use for the default listener.
listener 1883
listener 9001
protocol websockets

log_timestamp true
allow_anonymous true

# Store info on disk ecery 30 sec
autosave_interval 30
autosave_on_changes false

```
At least storing every 30 sec the last messages.

TD-er, 2024-02-19:
Retained messages don't have a lot to do with Mosquitto MQTT broker restarting.
It is about sending the last message of a topic to anyone subscribing to that topic.
Thus if HomeAssistant reboots, then it subscribes to the topics on the MQTT broker and will receive all retained messages sent to those topics it subscribes to.

tonhuisman, 2024-02-19:
> storing every 30 sec the last messages.

Not sure if that's a good plan when running your RPi on an SD-card... (default interval is 30 minutes, that seems often enough)

sincze, 2024-02-19:
> > storing every 30 sec the last messages.
> 
> Not sure if that's a good plan when running your RPi on an SD-card... (default interval is 30 minutes, that seems often enough)

Ok, Point taken ;-) still learning ;-0

I left SD-Cards back in 2015, fried them like hot chicken. MediaMarkt customer support even had a nickname for me.

So since then Domotica on SSD-ish media minimum with a backup for power outages..
[image]

TD-er, 2024-11-02:
Have been away the last couple of days, so a late reply.

Currently I am working on doing a full re-work of the WiFi part of the code as I promised I would do as soon as the transition to ESP-IDF5.x was finished.
So that takes up all my time for ESPEasy right now.

After this I will have a look at the auto discovery.

I know there is also some auto discovery code implemented in C014, however I have no idea about how compatible this is with HomeAssistant.
Also the code of C014 may need "some" TLC as it is a lot of `<copy><paste><paste><paste><paste><paste>`, which makes it near impossible to maintain and thus I have postponed looking into that controller code for quite some time.

ArjenWols, 2024-11-02:
Good morning TD-er,

Thanks the update and all the effort you pun into this.

Thanks for all the help.

Best regards
Arjen

tonhuisman, 2025-01-22:
Check the Controllers page. Only the Home Assistant (openHAB) Controller currently has this enabled.

tonhuisman, 2025-01-22:
> what to put in "publish topic" and other "topic" lines ?

The Controller Subscribe, Controller Publish and Controller LWT Topic can be set as desired.

This is the suggested AutoDiscovery setup:
Discovery Trigger topic: `homeassistant/status`
Auto Discovery topic: `homeassistant/%devclass%/%sysname%/%tskname%/%valname%`

The Auto Discovery topic is appended with `/config` by default.
At HA the Birth message payload should be set to default value `online`, and  (both should probably be made configurable).

Currently all Discovery messages are sent _without_ Retain flag.

closed 2025-07-02 by TD-er