letscontrolit/ESPEasy #4983: Communication Error on Serial Server

dersparkiie, 2024-02-20:
I'm trying to read Modbus data from Pylontech batteries via Modbus TCP. I configured the Serial Server as mentioned from Pylontech protocoll (115200, 8, N, 1). The answer of the Pylontech should provide 128bit but I only get 48bits sometimes 64bits or 80bits. 
I already gave up, thought the ESP8266 is not fast enough. But lately tried the Tasmota TCP to Serial option and it works perfectly.

What is the difference between ESPEasy Serial Server and Tasmota TCP to Serial?

tonhuisman, 2024-02-20:
It might work (much) better if you set the `RX Receive Timeout` value to 50 mSec, to allow for some small pauses between the sent data without ESPEasy 'giving up' on the receiving end.

dersparkiie, 2024-02-20:
Unfortunately max `RX Receive Timeout` is 20mSec. It seems thats not enough for Software Serial. 
I tried with Hardware Serial that worked for me with 20mSec  :)

TD-er, 2024-02-20:
Please do not use Software Serial unless there is really no other way.
It really is error-prone in receiving (and sending) data.
So better only use it for very short messages and also with proper checksum handling.
For example you can use it for some sensors like CO2 sensors, but when using it for rather "chatty" devices like GPS, it will result in lots of checksum errors and due to the low baudrate of GPS it will also make the ESP a lot slower because of the many interrupts when using SW serial.

tonhuisman, 2024-02-20:
Well, using Software Serial on an ESP8266 may be challenging indeed (you hadn't mentioned that in your OP...).
I didn't remember that the max. Timeout is limited to 20 mSec (it's using the 50/sec interval so 20 mSec is already stretching that), but in fact, 10 mSec should also be enough. It just has to be somewhat higher than 0 😅

closed 2024-02-20 by TD-er