letscontrolit/ESPEasy #4989: [FR] serialsend for the serial proxy plugin

chromoxdor, 2024-02-25:
Since with the help of @tonhuisman  i can process already data coming from my arduino nano it would be nice to also send some data/commands back.

Right now this can only be achieved by following steps:

1. Creating two tasks. "serial proxy" and to use `serialsend` also "serial server "
2. De- and reactivate the "serial proxy" task (without it receiving will not work anymore)

So having the option to use `serialsend` by using only "serial proxy" would be really great. :)

tonhuisman, 2024-02-25:
Hmm, that's what the commands `serialproxy_write` and `serialproxy_writemix` are intended to do... 🤔 [Docs](https://espeasy.readthedocs.io/en/latest/Plugin/P087.html#commands-available)

chromoxdor, 2024-02-25:
Ubbs.. how did i miss that. Strange 😬

closed 2024-02-25 by chromoxdor