letscontrolit/ESPEasy #4992: P135; SCD4x Pugin doesn't seem to save the settings

Sminter4, 2024-02-28:
I like to use the SCD40 module for my CO2 measurements.

I created a custom build for this purpose. With the following settings in “Custom.h”:

#define USES_P135 // SCD4x
#define P135_FEATURE_RESET_COMMANDS 1 // Enable/Disable quite spacious (~950 bytes) 'selftest' and 'factoryreset' subcommands

Build: [ESP_Easy_mega_20240225_custom_312_ESP8266_4M1M Feb 25 2024](https://github.com/letscontrolit/ESPEasy/releases)

On the configuration page there are values that are obviously not saved. This includes:
altitude; Temp offset; Automatic self-calibration

**The check marks in the GUI are still set, but when I read the values using a command, all values are "0".**
In my case, for example, this is "let,76,[SCD40#SerialNumber]"

[<taskname>#GetAltitude]
[<taskname>#GetTempOffset]
[<taskname>#SerialNumber]
[<taskname>#GetSelfCalibration]

https://espeasy.readthedocs.io/en/latest/Plugin/P135.html

Is it wanted like that?

TD-er, 2024-02-28:
```
[#GetAltitude]
[#GetTempOffset]
[#SerialNumber]
[#GetSelfCalibration]
```
These should use the task name before the `#`
Like if you named the task "SCD40":
```
[SCD40#GetAltitude]
[SCD40#GetTempOffset]
[SCD40#SerialNumber]
[SCD40#GetSelfCalibration]
```

Also I wonder why you're building "custom_312_ESP8266" and not the one at the top of the list?
"312" is using the latest Arduino code base and "274" is using version 2.7.4, which is still the better choice at this moment as it uses less memory and there are still some other quirks with the latest SDK version for ESP8266.

I don't really get what you mean by settings not being saved.
The title of this issue seems clear enough, but I can't really imagine what isn't being saved as this plugin mainly uses the standard save functions and thus other plugins/tasks should see the same issue.

TD-er, 2024-02-28:
When your ESP doesn't succesfully update the new build, you can try to disable some tasks and try to update again via the web interface.

When just giving the command `taskdisable,N` with N being the taskindex, it will only be disabled in memory and thus be re-enabled after a reboot. (when you do some kind of save, the disabled state will be saved of course)

tonhuisman, 2024-02-28:
Maybe it's a better plan to install the `Climate` or the `Collection E` build on your Wemos, that's pre-built, and known to work.

Sminter4, 2024-02-28:
The tip helped. By deactivating the plugins, I was able to successfully load the 274 version.
However, this build is unstable. Now when I load bmp photos onto my ST7735, the Wemos d1 mini crashes.
Then I loaded the 312 version again. This does not crash when loading photos.

Both versions behave the same when reading out the values for the SCD40. The result is always "0".
```
[SCD40#GetAltitude]
[SCD40#GetTempOffset]
[SCD40#GetSelfCalibration]
```

Only reading out the serial number works.

`[SCD40#SerialNumber]`

[image]

Loading the `Collection E` version doesn't help me. Otherwise I can no longer use the ST7735...

TD-er, 2024-02-28:
At least you can try to see if the plugin behaves differently.

TD-er, 2024-03-31:
I have been looking at the source code of this plugin, and I think it is also fixed in the upcoming build (now building)

I forgot to ask you for testing before triggering the new build. :(
Can you test if it has now been fixed?
The latest code currently in the `mega` branch is what is now being used to make the new "March" build.

iz8mbw, 2024-04-04:
@Sminter4 can this be closed?

tonhuisman, 2024-04-27:
@Sminter4 I finally identified the bug, and fixed it. Can you test using [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/8861511223) build, and report your findings?

closed 2024-05-01 by TD-er

Sminter4, 2024-11-27:
test result:
https://github.com/letscontrolit/ESPEasy/pull/5042

PR #5042 (2024-05-01): [Bugfix][P135] SCD4x fix reading sensor settings
Resolves #4992 

Bugfix:
- Reading sensor settings/parameters like the altitude and temperature compensation can only be retrieved when the sensor isn't measuring.
- During initialization, only set sensor parameters if needed (resolves a FIXME)

TODO:
- [ ] Testing by bug reporter (tested locally)