letscontrolit/ESPEasy #4993: [FR] Are you considering a matter agreement in the future?

wisonlau, 2024-02-29:
I want to connect the device connected to espeasy to Apple homepod, and I also need to cooperate with HomeAssistant. Will there be a version like tasmota that can access other hubs through matter protocol in the future?

TD-er, 2024-02-29:
Let's start with the most typical Dutch reply one can give:
"Yes, but...."

I currently have added support for IPv6, which is a requirement for Matter.
So all else needed is to implement authenticating, communicating with the blockchain in the Matter controller (HomeAssistant) and adding profiles.
Maybe also add support for Bluetooth for the commisioning.

Now the "but" part.
For true Matter support, one must also be a member to get signed keys.
HomeAssistant does have support for "self signed" keys, but others like Apple and Google et.al. do not.
This means you can only use Home Assistant to add devices with self-signed keys to the blockchain.
Once it has been added, other Matter devices should work with them.

And the other issue is that there aren't that many profiles right now for devices.
So I have to figure out a way to map ESPEasy devices onto the few existing profiles out there.

Jason2866, 2024-03-04:
"Keys" is the critical point. For Tasmota we use the development key(s). Currently all major brands do accept to connect. But there is the risk that this will be not be supported in future.
To be clear this would be the end of every Open Source Implementation of Matter.
Since for getting a key (beside the horrible costs) every device with it specific setup needs to be certified.