letscontrolit/ESPEasy #5001: RFID - Wiegand / Problem with values name

GitCybers, 2024-03-08:
Hello,

I have a problem with the "values name" in the options of my RFID-Wiegand Device. I can't let the values name empty. It is filled with "Tag" and so I get transfered to fhem this: "Tag: 1234"

I want only get "1234" without "Tag:" but I can't leave the value names empty. Is there any change to get only the ID transfered?

tonhuisman, 2024-03-08:
'Transferred' to where/what and how?

tonhuisman, 2024-04-07:
@GitCybers You haven't yet provided any additional information about your question, so it's hard to help you solve this.
A few pointers about the FHEM Controller in ESPEasy:
- The controller sends out JSON formatted data, and that's parsed by FHEM into something usable (I assume, as that's a legitimate way of providing data to FHEM)
- When parsing the JSON data sent from the controller yourself, you're supposed to parse the (complete) JSON part of it, where that "Tag": 1234, is a valid JSON name/value pair.

closed 2025-12-14 by TD-er