letscontrolit/ESPEasy #5008: [FR]

betux69, 2024-03-19:
Hi,

I would like to know if espeasy can work with crowpanel display of electrow.
I think we can go futher than Nextion.

Thanks

https://www.elecrow.com/esp32-display-series-hmi-touch-screen.html?idd=2

TD-er, 2024-03-19:
Not sure what the exact display model is.
If that display and touch screen chip is supported, then I don't see why it can't run ESPEasy.

Just keep in mind that we don't (yet) have very easy to use HMI screen elements  like a slider etc. So it will still require a lot of work to make it look "nice".

betux69, 2024-03-19:
Thanks for the answer, like nextion this display as a designer but more advanced than nextion.

I found lily go screen can we use it with espeasy ?
https://www.lilygo.cc/products/t4-s3

Thanks a lot

TD-er, 2024-03-19:
Eaxctly the same reply as it doesn't seem to list the exact display chip used.
You can see which are supported [here](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins).
However I doubt we already have this LilyGo display supported as it is an OLED screen and I doubt the same chips can be used to drive both LCD and OLED. And we don't yet have support for color OLED displays (as far as I know...)

betux69, 2024-03-19:
Nextion forever ;o)

thanks

closed 2024-03-19 by TD-er