letscontrolit/ESPEasy #5011: [P008] on ESP32-C3 not receive correct data

Meduzjam, 2024-03-24:
hi folks!

direct (no downshift)
gpio 0 = w0 
gpio 1 = w1
GND united

hikvision DS-K1101MK reader in mode 34 wiegand , uses mifare ultralight rfid, always receive 52 bit data (stable same but not correct key code)

[image]

**Used platform (please complete the following information):**
 - ESP type: ESP32-C3
 - Build version: 20240229 - Mega32-c3
 - Build set ESP_Easy_mega_20240229_normal_ESP32c3_4M316k_LittleFS_CDC

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: [e.g. 4M]
 - Brand/Model: Espressif Generic ESP32-C3 USB CDC 4M Flash ESPEasy 1810k Code/OTA 316k FS
 - Browser + OS [e.g. Windows 11/Chrome, IOS/safari]

**Additional context**
Add any other context about the problem here.

tonhuisman, 2024-03-24:
a) This would have better gone into [the Forum](https://www.letscontrolit.com/forum)

b) Can you show the configuration of your ESPEasy Device/task?

Meduzjam, 2024-03-24:
a) ok
b) 
[image]

tonhuisman, 2024-03-24:
Have you installed a level converter between the output of the reader and the inputs of the ESP? As ESP32 chips (all models) don't allow input signals over 3.3V, that will damage the input ports (not visible from the outside).

TD-er, 2024-03-24:
There have been some issues in the past with this plugin, where bit/byte order was swapped compared to reading the same card using commercially available readers.
Can you try the 'compatibility' mode and/or convert the read tag to HEX and see if there is some obvious bit/byte swapping in the read tag ID?

Meduzjam, 2024-03-31:
tested with 2 differents levelshifter - get same result. in rpi4 this schema work fine. 
[image]

TD-er, 2024-03-31:
Hmm that image contains transparency, which makes it impossible to understand when the browser is set to 'dark mode'
Here without transparency
[image]

TD-er, 2024-03-31:
I don't have such a reader, so I can't test to see if it really differs on ESP32-C3 compared to others.

TD-er, 2024-03-31:
Shouldn't the "HU" also be connected to 5V?

Meduzjam, 2024-03-31:
> Hmm that image contains transparency, which makes it impossible to understand when the browser is set to 'dark mode' Here without transparency [...]

thanks for the note, fixed it

Meduzjam, 2024-03-31:
> Shouldn't the "HU" also be connected to 5V?

I checked by supplying 5V to the HU and not supplying it - one hell of a deal

TD-er, 2024-03-31:
> > Shouldn't the "HU" also be connected to 5V?
> 
> I checked by supplying 5V to the HU and not supplying it - one hell of a deal

Not sure what you try to say here.... Is it making a difference or was it a lot of effort and no change? (which I find odd as the level converter doesn't "convert" when the desired output 'high' level isn't set)

Meduzjam, 2024-04-07:
tested with esphome (same wiring) - work fine!

closed 2025-12-14 by TD-er