letscontrolit/ESPEasy #5013: ESP8266 MQTT disconnecting & reboot ESP 1-3 minutes

dareksin78, 2024-03-26:
https://github.com/letscontrolit/ESPEasy/issues/4996

ESP_Easy_mega_20231225_normal_ESP8266_4M1M

I have the same problem.
I was looking for 2 days... and I think I found the solution, or so I think.

If I set value corrections here, there is a problem with mqtt!!!

[image]

Mqtt works fine if I don't set any rules.

Back to ESP_Easy_mega 20231225_normal_ESP8266_4M1M
does not help.

It's ok here: ESP_Easy_mega_20231130_normal_ESP8266_4M1M.

Please check and let me know if this is the reason for you.

tonhuisman, 2024-03-26:
Have you tried with this build? https://td-er.nl/ESPEasy/latest/ (From Chrome or Edge you can flash your ESP directly)

dareksin78, 2024-03-26:
[image]

[image]

TD-er, 2024-03-27:
OK, I think I can reproduce it.
First I thought it might be related to publishing an empty message.
I had a sysinfo task running with some formula and it ran fine for hours.
Then I added a second task (1-wire) without anything connected and then it started crashing when sending the sysinfo task.

I also have a stackdump, so I can hopefully find what's wrong here.

dareksin78, 2024-03-27:
The problem occurs immediately, you can see it within a few minutes, you don't have to wait several hours

TD-er, 2024-03-27:
Yep, it seems there is a bug in the Arduino code when using a very specific string length.

I have a bugfix on my desktop running to work around this problem, but have to check a lot of other code to see where it might run into this same issue.

Also I got some strange topic being sent to the MQTT controller, which might explain why it can get disconnected:
[image]

Have to look into this also as I have no clue yet how the formula can end up in the MQTT topic

TD-er, 2024-03-28:
I uploaded the build which should fix this to the [webflasher](https://td-er.nl/ESPEasy/latest/)
Can you test it, to see if it is also fixing your issue?

dareksin78, 2024-03-28:
It's OK for the first 10 minutes. 

I will write more tomorrow because I have to leave the house.

TD-er, 2024-03-28:
> It's OK for the first 10 minutes.

Ah based on the issue title, it should be considered fixed :)

dareksin78, 2024-03-29:
It's ok .
Thank you for fixing it.

PR #5014 (2024-03-29): [MQTT] Fix crash and disconnect sending to MQTT using formula
Fixes: #5013 
Fixes: #4996
closed 2024-03-29 by TD-er