letscontrolit/ESPEasy #5020: New Release

fly74, 2024-04-01:
Only for me to understand, why the release is called "[mega-20240401]" but the filenames are *0331?

TD-er, 2024-04-01:
I started the build at a few minutes past midnight (NL time, GMT + 1, + DST) and the timestamp used in the build script is UTC.

The tag is set from my own PC, which is set to Dutch time. The tag is used as title/release name.
This build is exactly the same as the previouw build, but now using a fixed build script.

Anyway now I know not to make new releases between 0:00 and 1:00 in Winter or 0:00 and 2:00 when we have summer time active. :)

fly74, 2024-04-01:
I solve things like this with building a datestring at the start of the script and use it for the whole run. 😎

closed 2024-04-01 by fly74

TD-er, 2024-04-01:
> I solve things like this with building a datestring at the start of the script and use it for the whole run. 😎

Yep, but that's for the more experienced  :)
We're just happy things compile ;)

tonhuisman, 2024-04-01:
> I solve things like this with building a datestring at the start of the script and use it for the whole run. 😎

But that still wouldn't take care of this situation. 🤔 
An alternative would be to set the timezone for the build runner jobs to CET (_and_ use that 'globally' defined date string), but not sure if that's worth the hassle...

fly74, 2024-04-01:
> But that still wouldn't take care of this situation. 🤔 An alternative would be to set the timezone for the build runner jobs to CET (_and_ use that 'globally' defined date string), but not sure if [...]

@TD-er @tonhuisman If you don't start a compile before midnight, we don't need it 😎

TD-er, 2024-04-01:
I also see it as a bug, but for now we will just 'work around it' as there are lots of tricky difficulties to deal with with the recently renewed build libraries and we need to use those libraries as others have set rather strict dependencies so you can't go back. :(