letscontrolit/ESPEasy #5021: [BUG] ESP32-C3/-C6 not working well in latest build

TD-er, 2024-04-02:
Seems to be related to recent changes in handling the ESPEasy console on USB-CDC port.

m1k83, 2024-04-02:
Hello.

I confirm the same problem with ESP32-S2. The last working build is 20240229.
There is no chance to get logs from serial, since the tty port is not initialized correctly.

Attached some logs from dmesg:

[ 8272.918239] usb 1-2.3: new full-speed USB device number 14 using xhci_hcd
[ 8273.690097] usb 1-2.3: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 1.00
[ 8273.690104] usb 1-2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 8273.690107] usb 1-2.3: Product: Espressif Generic ESP32-S2 USB CDC 4M Flash ESPEasy 1810k Code/OTA 316k FS
[ 8273.711699] usb 1-2.3: can't set config #\1, error -32
[ 8273.720831] usb 1-2.3: USB disconnect, device number 14
[ 8273.958153] usb 1-2.3: new full-speed USB device number 15 using xhci_hcd
[ 8274.410011] usb 1-2.3: device descriptor read/64, error -32
[ 8274.930136] usb 1-2.3: unable to read config index 0 descriptor/start: -32
[ 8274.930146] usb 1-2.3: chopping to 0 config(s)
[ 8274.930149] usb 1-2.3: can't read configurations, error -32
[ 8275.241969] usb 1-2.3: new full-speed USB device number 16 using xhci_hcd
[ 8275.592324] usb 1-2.3: device descriptor read/all, error -32
[ 8275.597691] usb 1-2-port3: attempt power cycle
[ 8276.221978] usb 1-2.3: new full-speed USB device number 17 using xhci_hcd
[ 8276.254361] usb 1-2.3: Device not responding to setup address.
[ 8276.489379] usb 1-2.3: Device not responding to setup address.
[ 8276.697993] usb 1-2.3: device not accepting address 17, error -71
[ 8277.009974] usb 1-2.3: new full-speed USB device number 18 using xhci_hcd
[....]

Thanks

TD-er, 2024-04-02:
Can you try [this test build](https://github.com/letscontrolit/ESPEasy/actions/runs/8522228162) to see if it is fixed with these changes?

m1k83, 2024-04-02:
Hi @TD-er 

Yes, now it's working as expected. 
Tried from esptool cmd line and from the webgui.

Thanks a lot

TD-er, 2024-04-02:
Thanks for testing.
I had not yet tested the S2 myself.

PR #5022 (2024-04-13): [HWCDC] Fix stability issues and bootloops on C3/C6
Fixes: #5021
closed 2024-04-13 by TD-er