letscontrolit/ESPEasy #5023: [BUG] Plugin 135 - SCD4x  freeze after some times

iz8mbw, 2024-04-02:
Hello.
using a SCD41 sensor on an ESP32 board, yes I use latest firmware of 1st April 2024.

What happens is after some times (some hours) the sensor freeze and it return "0" as value.
See here the values on Home Assistant:
[image]

To recover it I need to restart ESP32 board or do a Power-Off / Power-On.
On the logs I don't see nothing when the sensor freeze.

Polling interval set to 60 seconds.

tonhuisman, 2024-04-02:
Is this a `LittleFS` build (IDF 5.x) or SPIFFS (IDF 4.4)?
Does this also occur with older builds?

iz8mbw, 2024-04-02:
Hi Ton, it's a LittleFS build.
Same issue with older build but I connected the sensor since some weeks so cannot told to you if a build of 2 months ago was working.

TD-er, 2024-04-02:
How is the sensor connected?
Length of the cables, for example.
What I2C clock frequency?

iz8mbw, 2024-04-02:
20 cm, i2c slow as per screenshot

TD-er, 2024-04-02:
Have you also tried with the other I2C device disconnected?
Just to make sure it is only an issue with the SCD4x sensor and not related to something else.

iz8mbw, 2024-04-02:
No.
I'll do more test if needed but cannot understand why no EVENTS are in the log when it is frozen.

TD-er, 2024-04-02:
I have not yet any clue what is happening here.
One of the options is that the sensor gets locked up.
Or maybe the plugin is getting out of sync with the sensor.

iz8mbw, 2024-04-03:
Just an update, I changed the power supply of the sensor from 3V3 to 5V and now it's still working since 28 hours.

iz8mbw, 2024-04-03:
[image]

tonhuisman, 2024-04-03:
> Just an update, I changed the power supply of the sensor from 3V3 to 5V and now it's still working since 28 hours.

That sounds more like a wiring mistake or a faulty voltage regulator on your board (assuming there is one installed) or a broken sensor, as the SCD41 tech specs say the max VDD should be no higher than 3.6V...

iz8mbw, 2024-04-03:
According to Sensirion Data Sheet the maximum VDD is 5.5V
[image]

tonhuisman, 2024-04-03:
Hah! Official Sensirion pdf says different: 🤔 ❓ 
[image]

TD-er, 2024-04-03:
Is the digital level of the data signals related to this voltage?
If so, then other I2C devices may also be exposed to higher voltages

iz8mbw, 2024-04-03:
https://sensirion.com/products/catalog/SCD41/

tonhuisman, 2024-04-03:
Ah, too many Sensirion pdfs in the same folder 😊

iz8mbw, 2024-04-03:
> Hah! Official Sensirion pdf says different: 🤔 ❓ [...]

Sensor in scope is SCD41, not SGP41

tonhuisman, 2024-04-03:
Yes, mixup, sorry.

Still, smells like a broken sensor, as it should still work OK from 2.4V, so it's strange it won't cooperate at 3.3V Maybe adding a 10uF capacitor near the sensor might stabilize it better?

iz8mbw, 2024-04-04:
Guys, for now going to close this issue.
Thanks for suggestions.

Ciao!