letscontrolit/ESPEasy #5028: WIFI Connectio after cold boot

fly74, 2024-04-07:
The connection after power cycle with the latest release is very poor. Sometimes it connects after minutes or not, then another cycle. Is it known?

Test:
-with static IP
-hidden and non hidden wifi

TD-er, 2024-04-07:
Nope, not a known issue

TD-er, 2024-04-08:
What exact build are you using? LittleFS or SPIFFS? (thus ESP-IDF5.1 or 4.4 code)

And can you test [this test build](https://td-er.nl/ESPEasy/latest/) I just uploaded?

fly74, 2024-04-08:
This test is with LittleFS:

```
INIT : Booting version: ESP_Easy_mega_20240407_normal_ESP32_4M316k_LittleFS, (GitHub Actions) HEAD_9a51fa3 (ESP32 SDK 5.1.3.240325)
00.176 : (289772) Info   : INIT : Free RAM:289772
00.188 : (289580) Info   : INIT : Soft Reboot #4 Last Action before Reboot: Intended Reboot: CommandReboot Last systime: 1712572055 - Restart Reason: CPU0: Software reset CPU CPU1: Software reset CPU
00.199 : (289516) Info   : FS   : Mounting...
00.248 : (287524) Info   : FS   : LittleFS mount successful, used 151552 bytes of 323584
00.298 : (287316) Info   : CRC  : Settings CRC...OK
00.323 : (287160) Info   : ESPEasy console using ESPEasySerial
00.355 : (287044) Info   : CRC  : SecuritySettings CRC...OK
00.394 : (286996) Info   : INIT : SPI not enabled
00.395 : (286948) Info   : Set Network mode: WiFi
00.398 : (286772) Info   : NVS  : Load WIFICANDIDATE
00.401 : (286724) Info   : NVS  : Load WIFICANDIDATE
00.503 : (286748) Info   : WIFI : Set WiFi to STA
00.697 : (250308) Info   : WiFi : Start network scan all channels
03.899 : (248668) Info   : WiFi : Scan finished, found: 18
03.902 : (248532) Info   : WiFi : Best AP candidate: #Hidden# XXXXXX:B4 Ch:13 (-53dBm)WPA2/PSK (bgn)
03.905 : (248488) Info   : NVS  : Load WIFICANDIDATE
03.918 : (248440) Info   : NVS  : Load WIFICANDIDATE
03.920 : (248324) Info   : WiFi : Best AP candidate: #Hidden# XXXXXX:B4 Ch:13 (-53dBm)WPA2/PSK (bgn)
03.923 : (248280) Info   : NVS  : Load WIFICANDIDATE
03.936 : (248232) Info   : NVS  : Load WIFICANDIDATE
03.962 : (248244) Info   : ESPEasy console using ESPEasySerial
03.963 : (248196) Info   : INIT : Free RAM:248244
03.987 : (248132) Info   : ESPEasy console using ESPEasySerial
03.988 : (248020) Info   : INFO : Plugins: 48 ["Normal"] (ESP32 SDK 5.1.3.240325)
03.991 : (247932) Info   : WiFi : Best AP candidate: #Hidden# XXXXXX:B4 Ch:13 (-53dBm)WPA2/PSK (bgn)
04.004 : (247892) Info   : NVS  : Load WIFICANDIDATE
04.007 : (247844) Info   : NVS  : Load WIFICANDIDATE
04.019 : (247692) Info   : WIFI : Connecting XXXXXX#Hidden# XXXXXX:B4 Ch:13 (-53dBm)WPA2/PSK (bgn) attempt #0
05.033 : (247628) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
05.039 : (243224) Info   : Webserver: start
05.040 : (243160) Info   : WiFi : WiFiConnected(), start AP
05.051 : (243112) Info   : WiFi : WifiDisconnect()
05.254 : (243048) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
05.354 : (243000) Info   : WIFI : Set WiFi to OFF
05.567 : (266360) Info   : WIFI : Set WiFi to AP
05.687 : (240848) Error  : WIFI : [AP] softAPConfig failed! IP: 192.168.4.1, GW: 192.168.4.1, SN: 255.255.255.0
06.343 : (240780) Info   : WIFI : AP Mode enabled. SSID: ESPXXX IP: 192.168.4.1 ch: 1
06.448 : (238564) Info   : WIFI : Set WiFi to AP+STA
...
```

The 8266 have no problems with the last official, only the 32's. Setting is now with dhcp.

TD-er, 2024-04-08:
This is only a problem with hidden SSID?

fly74, 2024-04-08:
Yes only hidden and ESP32. If I unhide and disable the hidden option it can connect. But the 8266 can connect to the hidden net. Something is different between the two esp variants.

TD-er, 2024-04-11:
I have not yet tested with hidden networks, but in this build, there are some fixes regarding event handling in the Arduino code.
I also fixed entering WiFi credentials via the web flasher page.

Can you try this test build:
https://td-er.nl/ESPEasy/latest/

fly74, 2024-04-11:
Unfortunately still with ESP_Easy_mega_20240411_normal_ESP32_4M316k_LittleFS:

```
04.839 : (247612) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
04.844 : (243180) Info   : Webserver: start
04.846 : (243116) Info   : WiFi : WiFiConnected(), start AP
04.847 : (243068) Info   : WiFi : WifiDisconnect()
05.063 : (243004) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
05.163 : (242956) Info   : WIFI : Set WiFi to OFF
05.377 : (266320) Info   : WIFI : Set WiFi to AP
06.152 : (240832) Info   : WIFI : AP Mode enabled. SSID: ESPRING1 IP: 192.168.4.1 ch: 1
```

tonhuisman, 2024-04-11:
Is that ESP cleared before flashing the update? As it nearly immediately seems to go in AP mode...

fly74, 2024-04-11:
No. If I unhide the ssid it works immediately.

TD-er, 2024-04-11:
And just to be sure...
"Include Hidden SSID" and "Hidden SSID Slow Connect" are both checked?

fly74, 2024-04-11:
Yes.

TD-er, 2024-04-11:
I seem to be able to reproduce it here on my Mikrotik AP 

Looks like it keeps on attempting to the same BSSID when it fails.

TD-er, 2024-04-11:
Ah when I set the hidden SSID AP to channel 1 it connects almost immediately

TD-er, 2024-04-12:
OK, seems to be related to the AP also starting.
I unchecked "Do Not Start AP" on the Config tab and the ESP followed the AP when I changed channels almost immediately.

[image]

fly74, 2024-04-12:
not for me:

```
30.303 : (242252) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
30.304 : (242156) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
30.415 : (242268) Info   : WIFI : Set WiFi to OFF
30.737 : (265932) Info   : WIFI : Set WiFi to STA
30.855 : (242108) Info   : WIFI : Connecting XXXXXXX#Hidden# XXXXXXXX:B4 Ch:13 (-41dBm)WPA2/PSK (bgn) attempt #2
31.868 : (242124) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
36.013 : (242188) Info   : WD   : Uptime 1  ConnectFailures 0 FreeMem 242332 WiFiStatus: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
00:01:06.014 : (242188) Info   : WD   : Uptime 1  ConnectFailures 0 FreeMem 242332 WiFiStatus: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
00:01:36.018 : (242188) Info   : WD   : Uptime 2  ConnectFailures 0 FreeMem 242332 WiFiStatus: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
00:02:06.011 : (242188) Info   : WD   : Uptime 2  ConnectFailures 0 FreeMem 242332 WiFiStatus: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
00:02:36.015 : (242188) Info   : WD   : Uptime 3  ConnectFailures 0 FreeMem 242332 WiFiStatus: WL_NO_SSID_AVAIL 1 ESPeasy internal wifi status: DISCONNECTED
```

Do Not Start AP: off

TD-er, 2024-04-12:
I meant "checked" (changed it in my post)
So to force not starting AP when it fails to connect.

fly74, 2024-04-12:
Oh I mean on too :)

fly74, 2024-04-13:
I think I have located it. If I set the Wifi 2.4 channels from automatic to dedicated it works witch channel 1-11. 12 and 13 don't work wich is selected by automatic in my environment. So the hidden seems to ignore 12,13 on reconnect with hidden SSID. I hope that helps. Go back to ESP_Easy_mega_20231130_normal_ESP32_4M316k_LittleFS makes no difference. Maybe the channel changed from changes in channel use by other wifi. Can we extend the channel range if it is the cause?

fly74, 2024-04-16:
But remember it is only affected to the ESP32 builds.

TD-er, 2024-04-16:
Yep, I also discussed this with an Espressif dev.
I have to look into this a bit more to see what may be causing this as it seems to me like this is a bug in ESP-IDF code.

If it was just a matter of having the "wrong" country code set (not sure it is "wrong" or just "global default") then the ESP should also not be able to connect to the AP on that channel when the SSID was not set to be hidden.

I still have to try to see if actually setting the country code does make both behave the same (thus both can connect to AP on channel > 11 or both can't)

TD-er, 2024-04-16:
Just one more question...
Do you know if ESPEasy already has started its own AP since connecting to a hidden SSID may take longer if there are many hidden APs?

TD-er, 2024-04-16:
Hmm found some interesting info here: https://docs.espressif.com/projects/esp-idf/en/v5.1.3/esp32/api-guides/wifi.html#wi-fi-country-code

> If the connected AP has country IE in its beacon, the country info equals to the country info in beacon. Otherwise, use the default country info.
> 
> For scan:
> 
> Use active scan from 1 to 11 and use passive scan from 12 to 14.
> 
> Always keep in mind that if an AP with hidden SSID and station is set to a passive scan channel, the passive scan will not find it. In other words, if the application hopes to find the AP with hidden SSID in every channel, the policy of country info should be configured to WIFI_COUNTRY_POLICY_MANUAL.

fly74, 2024-04-16:
> Just one more question... Do you know if ESPEasy already has started its own AP since connecting to a hidden SSID may take longer if there are many hidden APs?

Not really recognized.

fly74, 2024-04-16:
> if the application hopes to find the AP with hidden SSID in every channel, the policy of country info should be configured to WIFI_COUNTRY_POLICY_MANUAL.

Can we do this ?

TD-er, 2024-04-16:
> > if the application hopes to find the AP with hidden SSID in every channel, the policy of country info should be configured to WIFI_COUNTRY_POLICY_MANUAL.
> 
> Can we do this ?

Not sure yet as I still have to test it.

TD-er, 2024-04-17:
I have it working and it can now connect to hidden SSID on channels > 11
Will make a PR for it, but I'm not yet really happy about it.
It just feels like we might open up a can of worms, so I do have to make some more tests before it will be merged.

TD-er, 2024-04-17:
[GitHub Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/8722483457) (still running)

fly74, 2024-04-17:
But the question is, why it is working on 8266?

TD-er, 2024-04-17:
I don't know and that's also what doesn't make me happy about this fix.

At least I do now understand why other firmwares like Tasmota decided to simply not support hidden SSID.
It really doesn't add any extra security and it is for sure some edge case which probably hasn't been tested very well.

fly74, 2024-04-17:
I know the never-ending discussion about hiding the SSID. It not used as a security reason for me. But all thing that using the wifi can do it very well in our infra.

TD-er, 2024-04-18:
Anyway does it work on your setup too?
If so, then I will try to look into this a bit more to see if I can turn it into settings like max. channel, country code and indoor/outdoor use etc.

Right now it just feels really sketchy what I added because I have no idea what I may have changed by not explicitly setting those parameters I mentioned or what may have changed by actually setting the start and end channel to 1 ... 13

fly74, 2024-04-18:
Yeah work!

```
INIT : Booting version: ESP_Easy_mega_20240417_normal_ESP32_4M316k_LittleFS, (GitHub Actions) HEAD_e55007b (ESP32 SDK 5.1.3.240325)
00.170 : (289812) Info   : INIT : Free RAM:289812
00.182 : (289684) Info   : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reset by PRO CPU
00.193 : (289572) Info   : FS   : Mounting...
00.253 : (287580) Info   : FS   : LittleFS mount successful, used 151552 bytes of 323584

Build changed!
00.341 : (287276) Info   : ESPEasy console using ESPEasySerial
00.392 : (287160) Info   : CRC  : SecuritySettings CRC...OK
00.443 : (287112) Info   : INIT : SPI not enabled
00.444 : (287064) Info   : Set Network mode: WiFi
00.547 : (286960) Info   : WIFI : Set WiFi to STA
00.739 : (250504) Info   : WiFi : Start network scan all channels
03.064 : (249040) Info   : WiFi : Scan finished, found: 16
03.067 : (248896) Info   : WiFi : Best AP candidate: #Hidden# xxxxxxxxxxxx:B4 Ch:13 (-48dBm)WPA2/PSK (bgn)
03.070 : (248784) Info   : WiFi : Best AP candidate: #Hidden# xxxxxxxxxxxx:B4 Ch:13 (-48dBm)WPA2/PSK (bgn)
03.106 : (248792) Info   : ESPEasy console using ESPEasySerial
03.107 : (248744) Info   : INIT : Free RAM:248792
03.131 : (248680) Info   : ESPEasy console using ESPEasySerial
03.132 : (248568) Info   : INFO : Plugins: 48 ["Normal"] (ESP32 SDK 5.1.3.240325)
03.135 : (248480) Info   : WiFi : Best AP candidate: #Hidden# xxxxxxxxxxxx:B4 Ch:13 (-48dBm)WPA2/PSK (bgn)
03.147 : (248328) Info   : WIFI : Connecting xxxxxxxxxxxx#Hidden# xxxxxxxxxxxx:B4 Ch:13 (-48dBm)WPA2/PSK (bgn) attempt #0
03.961 : (247532) Info   : WIFI : Arduino wifi status: WL_CONNECTED 3 ESPeasy internal wifi status: DISCONNECTED
03.967 : (243084) Info   : Webserver: start
03.994 : (242788) Info   : WIFI : Connected! AP: xxxxxxxxxxxx(xxxxxxxxxxxx:B4) Ch: 13 Duration: 727 ms
03.996 : (242644) Info   : WIFI : DHCP IP: xxxxxxxxxxxx (xxxxxxxxxxxx) GW: xxxxxxxxxxxx SN: 255.255.255.0 DNS: xxxxxxxxxxxx / 0.0.0.0   duration: 15 ms
```

TD-er, 2024-04-18:
OK, I did find some entries in the WiFi scan result from which I might get some country info.
Problem is that by not setting things to 'auto', all country related stuff will be used from whatever I set in the config instead of what the AP advertises.

But maybe I can deduct some info from the APs seen during the scan.
Still I might need to switch to 'slow connect to hidden AP' to actually use the country settings of that specific AP before connecting.

Or maybe do it a bit more intelligently to only set to specific country settings if there are possible candidates found with channels > 11.

fly74, 2024-04-18:
Maybe It is a special thing for the Config tab. 99% of all users not really need it. So it would be sufficient set on the tab an option 'use ch>11 for hidden SSID' e.g. without doing things to analyze the APs.

TD-er, 2024-04-18:
As a test to see what info I can get...
[image]

The 'g' network doesn't show any of this info.
Meaning I can't use this to connect to hidden SSID 'g' networks.

My APs are nicely set to "NL" but the APs of my neighbor are not properly setup and announce themselves as "DE".
They do announce the max. channel they allow.
None of these have been set to "Outdoor" or "Indoor" or "X" (non-country specific settings)

I know I can set my Mikrotik APs to outdoor/indoor, so I will try to see if it does show up.

TD-er, 2024-04-18:
Hmm this mAP Lite Mikrotik AP is now set to "2 GHz-only-N" and the ESP still sees it as a "g/n" capable AP and it never tells what regulatory domain is used even though I have set it to "Netherlands" and also tried switching indoor/outdoor.
It just doesn't show it. (even when it is not set to hidden SSID and I also set the channel to a lower channel to be sure)

fly74, 2024-04-18:
As I said, don't care about the APs. There are to many in the universe. Just let anybody configure higher channels, if they know what they are doing.

fly74, 2024-04-18:
Make it as a 'your own risk' option :)

TD-er, 2024-04-18:
> Make it as a 'your own risk' option :)

Like all other settings ;)

closed 2024-05-01 by TD-er

fly74, 2024-05-01:
@TD-er What is the solution? An option or wifi property reaction in code?

PR #5036 (2024-05-01): [Hidden SSID] Fix ESP32 connecting to hidden SSID with channel > 11
See:  https://docs.espressif.com/projects/esp-idf/en/v5.1.3/esp32/api-guides/wifi.html#wi-fi-country-code

If the connected AP has country IE in its beacon, the country info equals to the country info in beacon. Otherwise, use the default country info.

For scan:

Use active scan from 1 to 11 and use passive scan from 12 to 14.

Always keep in mind that if an AP with hidden SSID and station is set to a passive scan channel, the passive scan will not find it. In other words, if the application hopes to find the AP with hidden S [... cut]
commit c41d3457 (2024-05-01): [Hidden SSID] Fix ESP32 connecting to hidden SSID with channel > 11

Fixes: #5028