letscontrolit/ESPEasy #5033: lost in build ;o)

betux69, 2024-04-16:
Hi,

It's been a while since I've done a build. I see custom 274 or custom 312 esp8266_4M_1M. 
What is the difference ? before there was only one choice it was simpler for for beginners ;o)

Thanks

TD-er, 2024-04-16:
For ESP8266 there is SDK version 2.7.4 and 3.1.2
For a long time the SDK code in the 3.x branch was unstable and resulted in larger build sizes.
So I only had one or two "Custom" builds just to see when a build may fail due to code changes in the 3.x branch.

A few months ago I did spend quite a lot of time to look into the 3.x branch of the SDK to see if it is now working more stable or even can be considered "usable".
To my surprise it is more usable and the build sizes are for some builds even smaller than the 2.7.4 SDK builds.
However SDK 3.x still does use quite a bit more RAM, which does make it less practical in some use cases.

I did decide to keep both in the build scripts as for some setups they do work better and more stable than 2.7.4 based builds.
And for some other use cases the 2.7.4 SDK based build is more stable.

So I do agree with "less choice was simpler" and I wish there was just a simple "one build fits all".

tonhuisman, 2024-04-16:
On the main ESPEasy Github page (readme.md) there is an overview of how the binary names are 'assembled': https://github.com/letscontrolit/ESPEasy#binary-releases

betux69, 2024-04-16:
So i will try both :o)

thanks

betux69, 2024-04-16:
close

closed 2024-04-16 by betux69