letscontrolit/ESPEasy #5034: no module named 'recommonmark'

betux69, 2024-04-16:
hi,

I made a fresh install of vs studio on windows 11, and i try to make a build.
i've an error : no module named 'recommonmark'.
Something is missing in my python modules but i don't know how to install it.

Thanks

TD-er, 2024-04-16:
That one is only mentioned in the requirements.txt file for the docs.

Have you followed these steps? 
https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#platformio-with-vscode

And you mention "VS Studio", but I assume you mean "VS code" since Visual Studio is not free to use.

betux69, 2024-04-16:
yes it's vs code sorry, 20 years ago i used visual studio ;o)
now it's good bi i did nothing and yes i followed steps.

thanks

closed 2024-04-16 by betux69