letscontrolit/ESPEasy #5035: the revenge of issue with rs232

betux69, 2024-04-16:
Hi,

I posted last year an issue with rs232 command and it was solved.
ticket reference :
RS232 command how it works ? #4558 

I try to make a second esp for an other epson video projector.
So i flashed _Easy_mega_20240414_normal_ESP8266_4M1M.bin but it didn't work.
when i send my command (serialsendmix,"PWR OFF",0x0D) the return is ok but nothing work.
So i thought maybe  i need to make a build with P020 to be sure it is in the build.
Same thing it didn't work.
I did the same schema RX with RX and TX with TX, link to 5V.

Have the integrations you made been integrated into the project?

thanks

tonhuisman, 2024-04-16:
> when i send my command (serialsendmix,"PWR OFF",0x0D) the return is ok but nothing work.
> So i thought maybe i need to make a build with P020 to be sure it is in the build.

If the command reports Ok, then the code is included, and executed. 🤔

betux69, 2024-04-17:
it's me i'am to stupid ;o)

thanks and sorry

closed 2024-04-17 by betux69