letscontrolit/ESPEasy #5040: [FR] Add Direction value

navneev, 2024-04-25:
**Is your feature request related to a problem? Please describe.**
Not problem related, just a nice to have

**Describe the solution you'd like**
When the distance value changes, determine the direction of the change.
Direction can have 3 values;

- -1 -> distance gets smaller
- 0 -> no change in distance
- 1 -> distance gets bigger

**Describe alternatives you've considered**
I tried to add this logic into flows, but that gets pretty complex

tonhuisman, 2024-04-25:
You haven't mentioned what plugin you are talking about (I want to avoid guesses 😉), please add that info to the title and description.

tonhuisman, 2024-04-27:
@navneev I've added the requested values in the PR linked above this comment, and also did some code optimizations, and added async processing for VL53L0x. Can you test once the build is completed, available at this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/8859742691), and report your findings here?

tonhuisman, 2024-05-28:
@navneev Any news on testing the new functionality for VL53L0X and/or VL53L1X?

closed 2024-05-30 by TD-er

navneev, 2024-06-04:
> completed via #5041 <https://github.com/letscontrolit/ESPEasy/pull/5041>.

Hi Ton,

Sorry dat ik nu pas reageer, maar ik heb kort na mijn verzoek een ongeval
gehad (aangereden door een fatbike - oa zware hersenschudding) en ben pas
gisteren uit het ziekenhuis ontslagen.
Mijn oog/hand coordinatie is nog steeds niet in orde en ik weet niet
wanneer ik weer in staat ben om te testen en bouwen.

MvG, Hans van Veen

PS. Dit bericht is door zijn zoon Marco geschreven. Mijn vader is nog niet
in staat om beeldschermwerk te doen

_______________________________________________________________
The cure for boredom is curiosity. There is no cure for curiosity.

tonhuisman, 2024-06-04:
> Hi Ton, Sorry dat ik nu pas reageer, maar ik heb kort na mijn verzoek een ongeval gehad (aangereden door een fatbike - oa zware hersenschudding) en ben pas gisteren uit het ziekenhuis ontslagen. [...]

@navneev 
Hans,
Oef, dat is geen mooi bericht. Veel sterkte met het herstel.
Ik heb de wijzigingen zelf al kunnen testen, maar een tweede blik is altijd goed natuurlijk. Ik hoop en verwacht dat je in de nabije toekomst weer met de electronika hobby aan de slag kunt, en evt. mijn resultaten bevestigt.
Succes,
Ton

PR #5041 (2024-05-30): [P110]/[P113] Add Direction value, optimizations
Resolves #5040 

Features:

[P110] VL53L0x:
- Add Direction value
- Add `Send event when value unchanged` and `Trigger delta` options (changed behavior!)
- Process sensor reading asynchronously

[P113] VL53L1x:
- Add Direction value
- Code optimizations

TODO:
- [ ] Testing by requester (both sensors tested locally)
- [x] Update documentation
  - [x] VL53L0x
  - [x] VL53L1x