letscontrolit/ESPEasy #5044: Information of which set contains which plugin

Popyllol, 2024-04-29:
Hi,
I'm using ESPEasy for different purposes and really love it. The only thing what I sometimes need and haven't found so far is the information which plugin is in which collection.
So I made a little awk script which parses
  src/src/CustomBuild/define_plugin_sets.h
and prints the results.
```
#!/usr/bin/env -S awk -f
BEGIN{
  printf("List by sets:\n-------------\n");
}
/^#ifdef PLUGIN_SET_/,/^#endif/ {
  if(/^#ifdef PLUGIN_SET_/) {
    set=$NF;
    gsub(/^PLUGIN_SET_/,"",set);
    first=0;
  }
  if(/^\s+#define USES_/) {
    if(first++==0){
      printf "%s\n  Plugins:\n",set;
    }
    plugin=$2;
    gsub(/^USES_/,"",plugin);
    description[plugin]=$0;
    gsub("^.*//","//",description[plugin]);
    printf ("\t%s %s\n", plugin, description[plugin]);
    sets[plugin]=sets[plugin]","set;
  }
}
END {
  printf("\nList by plugins:\n---------------\n");
  asorti(sets,sorted_plugins);
  for(plugin in sorted_plugins) {
    printf("%s: %s\n  Sets:\n",sorted_plugins[plugin],description[sorted_plugins[plugin]]);
    split(gensub("^,","","g",sets[sorted_plugins[plugin]]),sets_array,",");
    asort(sets_array);
    for(set in sets_array) {
      printf("\t%s\n",sets_array[set]);
    }
    printf("\n");
  }
}
```

Something like this or a matrix of it would be nice.
Greetings,
   Lars

tonhuisman, 2024-04-29:
Hm, we don't currently have a matrix, but isn't the [list as found in the documentation](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins) what you'd want?

Popyllol, 2024-04-29:
Not the information I needed lately for example: I needed to know where the Controller RN2483 (C018) is supported in and all I can see is in Collection, but not which one.
And my script shows it is found only in MAX... which might be wron but the Collections I installed to check hasn't this controller available.
The script just is a approach to get more information about in which build I can find a feature.

tonhuisman, 2024-04-29:
> all I can see is in Collection, but not which one.

That means it's included in _all_ Collection builds...

I'll find a way (by python script, as that's what we use for other scripting stuff) to generate that overview from the documentation source-files, where it is all available (as long as the plugin is properly documented, and most are).

tonhuisman, 2024-04-29:
> Controller RN2483 (C018)

For technical reasons, most likely build-size limits, this plugin has been disabled from all builds except MAX, but you can manually include it in a Custom build.
Maybe we should re-enable it for ESP32, as that has less build-size restrictions then ESP8266. (I'll do some tests later today)

Popyllol, 2024-04-29:
Thank you a lot!
Have a nice day!
   Lars

tonhuisman, 2024-05-04:
@Popyllol I've created a (python) script that generates the plugins per build set, including Normal and/or Collection where needed, and adding and excluding some stuff where needed, etc.

The result looks like this:
[image]

And when expanded:
[image]

Controllers are also included, at the end of the list per build set:
[image]

closed 2024-05-04 by TD-er

Popyllol, 2024-05-05:
@tonhuisman Pretty cool!
Thank you very much!

PR #5047 (2024-05-04): [Docs] Add plugin overview per build set (generated)
Resolves #5044 

Features:
- Adds a list of plugins per build set to the Plugins page (collapsed by default to avoid getting a really long page)
- Includes Controllers and sorts Controllers _after_ Plugins
- Generated from existing documentation files
- Re-generated when built in GH Actions

NB: There are some differences between ESP8266 and ESP32 builds, but that information is not available in the documentation, so isn't reflected in the overview... A little of that knowledge is implemented in the python script, though.