letscontrolit/ESPEasy #5061: [BUG] mDNS unavailable in ESP32-C3

TobiasPSP, 2024-05-25:
Firstly, I am new to ESPEasy so in case I overlooked something I apologize. Thanks for your awesome work!

I seem to be unable to configure or use mDNS in the latest "Display" build targeting S2/S3/C3 on a C3 Mini.
I searched for hours in the documentation but couldn't find a setting or even mention.

It appears to me that one target audience of ESPEasy are makers that are looking for a lean and generic management platform. For this audience I feel mDNS is crucial. I wouldn't be able to keep track of changing IP addresses when trying to contact devices.

Apparently, mDNS is included in ESPEasy. What am I doing wrong/missing to get it configured?

Many thanks!

TD-er, 2024-05-25:
I have to check, but I think it might have been disabled by me out of pure frustration as it is a continuous source of 'strange issues'.

TD-er, 2024-05-25:
At least I found this commit to explicitly disable mDNS for the ESP32-S2:
https://github.com/letscontrolit/ESPEasy/pull/4672/commits/9aed4b033617c3df19c665f493291f8c1f29abb2

Will look further to see why/when it was disabled for the C3 (and others???)

TD-er, 2024-05-25:
Also found this commit: https://github.com/letscontrolit/ESPEasy/pull/4979/commits/bfe5926589cb009b7dc5ed4b68f841946c5f2008#diff-0e7c5ae1aa6edf7e0f181cf8990739e1ef2d9354826fc56dbd241d540e6ae507R2763
Apparently I did disable it for C3/C6 as it did cause some long stalls where the ESP became unresponsive for quite a long time (seconds)

TobiasPSP, 2024-05-25:
Thank you for looking into this. I am sorry that mDNS in ESPEasy is causing so much trouble. 

Is there a chance to have an option in config to control mDNS by the user?

TD-er, 2024-05-25:
I did look a bit more in some commits for the recent ESP-IDF5.1 and the latest Arduino 3.0.0 code for ESP32
I guess we could try enabling it.
I can add it to a branch I'm now working on for a new plugin, so you could later test a test build.

TD-er, 2024-05-25:
> Thank you for looking into this. I am sorry that mDNS in ESPEasy is causing so much trouble.

Not only ESPEasy.
I just asked the people from Tasmota whether they are using mDNS and this was the reply:

> We do use only for Matter. For this restricted use case it does work
> It is not activated for the "classic" use

Anyway I will include it in a test build, however I'm not sure it will now work with less issues compared to earlier builds where it was still enabled.

TD-er, 2024-05-25:
As soom as this [GitHub Actions test build](https://github.com/letscontrolit/ESPEasy/actions/runs/9238303245) is ready, you can try any ESP32-xx build with "LittleFS" in the name as those have `FEATURE_MDNS` enabled again.

N.B. if you had a build running without "LittleFS" in the name, you need to backup your settings and rules file as the file system will be erased when moving from SPIFFS to LittleFS.

TD-er, 2024-05-25:
I also uploaded the build to the webflasher: https://td-er.nl/ESPEasy/latest/

TobiasPSP, 2024-05-26:
Greatly appreciated! I‘ll test and report back. 

Just to clarify: will there be a setting in config showing up in mDNS-enabled builds where I would specify the mDNS name, or is this automatically using the node name? 

Sorry if this sounds naive, I did wrap my head around the documentation quite extensively but mDNS doesn’t seem to be explicitly mentioned, or I possibly missed it.

Anyway, looking forward testing it. Thank you for your response and willingness to look into this.

TD-er, 2024-05-26:
It just uses the set hostname as mDNS
You can see the <hostname>.local name on the root page

 For example:
[image]

TD-er, 2024-05-29:
I will disable it again as just having mDNS enabled will cause the ESP32-C3 to have a load of 35 - 50% without doing anything else...
And it will also interfere significantly with just about anything as the background task handling this will sometimes take 200 - 500 msec which is blocking the entire system.

So it will be disabled unless you explicitly enable it for a custom build.

TobiasPSP, 2024-05-29:
Ok, I understand. I am not as quick as you are ;-) family and job taking its toll. 

I was about to test tonight but if the side effects are so severe then I agree it doesn’t make sense. 

So this issue is isolated to newer mc like c3/s2,3? Would a build for a classic esp32 have mDNS enabled?

TD-er, 2024-05-29:
I have it now disabled for all builds if it hasn't been explicitly enabled (so only for custom builds where you enable it using a define)

Later I can try to figure out whether it is related to ESP32-C3/C6 only or also for the others. (I assume they all suffer from it)

I had it enabled for a pending PR I'm working on to implement support for a lightning detector.
This is somewhat time-critical as you need to fetch the info from the sensor ASAP or else you will be missing strikes and that was exactly what was happening.

thomastech, 2026-09-13:
Two years later.... 
Have there been any recent magic that allows ESPEasy on an ESP32-C3 to work reliably with mDNS?

TD-er, 2026-09-13:
Yep, I have spent almost a full week a few months ago  to make it work and the conclusion again was -> Disabled! as it caused lots and lots of issues, even BSoD on my PC!
I have no idea what is happening with mDNS on ESP's but it is for sure causing lots and lots of issues.

You can enable it in a custom build, but on your own risk.

TD-er, 2026-09-13:
I did update the ESP-IDF and enabled mDNS for ESP32xx_MAX builds for this build: https://github.com/letscontrolit/ESPEasy/actions/runs/34783853085
There have been changes/fixes in the ESP-IDF repo (and still more not included in this build), so maybe things have changed since my last tests.

TobiasPSP, 2026-09-13:
I am using mDNS without issues in my own firmwares with esp32, so likely it is not a core esp32 issue but possibly related to other parts in your firmware. Thanks for all your efforts to pinpoint it.

There are occasionally eerie cross-effects. Did you ever try i.e. codex and let AI have a look at your code? It is really good at this. I was able to solve a few awkward issues lately that way that I wouldn’t have discovered manually in a million of years. 

That said, imho mDNS can have issues with certain network designs, not so much related to esp32.

TD-er, 2026-09-13:
Nope, have not let AI have a look at the code... or at least not intentional (see forum issues).
I have tried once to use the free AI token stuff you get as OpenSource developer from GitHub, but I never managed to get it to work and already burned a significant portion of the monthly free tokens on just installing the plugins for VScode, without doing anything else.
So then I just gave up on AI tools.

But if you are able to find what I'm likely doing wrong with mDNS, it would be highly appreciated as I've spent quite a lot of time on it already and have honestly no idea what I'm doing wrong.

thomastech, 2026-09-13:
The mDNS problems have reportedly been device specific. For example, the ESP32-C3 was one such bad boy. And it just so happens my ESPEasy device is using a C3 and .local would be nice. 

The mDNS issues surfaced a couple years ago and maybe it's fixed by now? Or maybe not. I was hoping someone had already validated mDNS on a C3 in recent months.

BTW, I'm using mDNS on a ESP32-S3 for both .local URL use, as well as device advertising. Works perfectly. :)

And for the record, the paid version of ChatGPT is very useful for coding. I recently used it to create a custom ESPEasy plugin and it saved me a lot of painful hours. Oh sure, it can be a brat and time waster, but overall I am satisfied.

Please don't hate me for using AI, because, well, it is evil. I am convinced it will attempt kill all humans in my lifetime. So there's that.

TobiasPSP, 2026-09-14:
My ba

TD-er, 2026-09-14:
The main differences between C3 and S3 are:
- single core vs. dual core
- Completely different CPU architecture

The single core vs. dual core may be an issue here as it seems to be also a timing issue.
However, the completely different architecture also may impose very nasty programming issues where certain types are not defined the same and more importantly there is quite a distinct difference in how the data in uninitialized structs is initialized. (all 0 vs. random data)
This last one alone did take me quite a lot of work when adding SSL support some years ago as mbed-tls does not initialize the structs they use, causing non-existing pointers to be destructed and thus leading to crashes.