letscontrolit/ESPEasy #5072: Problem with PCF8574, plugin P019

tomatos-st, 2024-06-13:
I built an ESP32 controlled device.
I was to control several relays using the PCF8574  port expander and ULN2003 (7-ch Darlington transistor array). It turned out that the expander control was not working properly.
It seems that if it only drives the LED, it is ok. But if I want to turn on the transistor, the port sets the state to 1 for a short moment and then resets to zero. I quickly built a test circuit and sent it to the command "PCFGPIOrange,1,8,0" and "PCFGPIOrange,1,8,1" and only the LED on port 1 works, the others only blinked for a moment.
[image]

You can see it 10 seconds into the video

https://github.com/letscontrolit/ESPEasy/assets/56383991/4eb967f2-77e8-4c8d-b300-ea0214a77f56

My ideas after analyzing the ULN2003 diagrams: PCF8574 issues signal 1 on the port and then reads the status at the input. Due to the fact that inside the ULN 2003, in addition to the transistors, there are resistors connecting the input to ground, the system reads the 0V state (pulling the input to ground) and switches the port output back to 0.
With the MCP23017 expander there are no such problems, everything works fine.

Tomorrow I will check the waveforms on the oscilloscope.

TD-er, 2024-06-13:
Shouldn't there be pull-up resistors for the ULN2003 inputs?
The PCF8574 can handle a bit more current to GND compared to deliver current on its output pins.

tomatos-st, 2024-06-14:
There is no need for pull-up resistors.
The PCF8574 has a load of 25mA per pin (according to specifications). The output is full-range - transistors on the GND and VCC branches.
[image]

I measured the load, above about 2mA there is a problem

uzi18, 2024-06-14:
Check out typical application in datasheet of pcf8574 and output current for low and high state. 
You will figure out that pullups are necessary here and as it is very old design. 
This chip can only connect pin to ground or detect low state on pin. 
Please add resistor ladder and check one more time.

TD-er, 2024-06-14:
Yep, there is a big difference in max. current between delivering a high signal and connecting it to GND.

closed 2024-06-14 by tomatos-st

uzi18, 2024-06-14:
Could you please prove it is a solution here for future issues like this one?

tomatos-st, 2024-06-15:
No problem, on Monday I will try to modify the test and target boards with pull-up resistors and let you know

uzi18, 2024-06-15:
This will prove solution for other users.

tomatos-st, 2024-06-18:
Ok, I checked
 pull-up resistors help

https://github.com/letscontrolit/ESPEasy/assets/56383991/8b5a26a3-c069-4dd2-a9ed-9eba8bf289ea
Commands sent:
PCFGPIOrange,1,8,1
PCFGPIOrange,1,8,0
PCFGPIOtoggle,1
PCFGPIOtoggle,2
PCFGPIOtoggle,3
PCFGPIOtoggle,4
PCFGPIOtoggle,5
PCFGPIOtoggle,6
PCFGPIOtoggle,7
PCFGPIOtoggle,8

I only have a strange thing in the target board, but I suspect PCF8574 is dead
I will exchange it for another one and possibly reopen the issue

uzi18, 2024-06-19:
Will suggest you to connect outputs of pcf8574 to individual transistors like IRLML2244 because of default High state of PCF after power on/reset.

TD-er, 2024-06-19:
What voltage levels are present on the input pins of the ULN2003?
And what levels are required?
As what @uzi18 also suggests, maybe you need some level converters as either the PCF can't reach high enough voltages or the PCF is not really happy with higher pull-up voltage levels.

uzi18, 2024-06-19:
What I mean - uln is inverting signal - High state on input gives max 0,5A connection to ground on output - Low state. With pullups on PCF outputs you got default state on for connected relays. 
So it is better to drive mosfet-p with pcf output and from there energize coil directly - remember to add fast diode to protect pcf/transistor.

tomatos-st, 2024-06-19:
To control relays, I prefer transistor control in the OC system. I usually use LUM2003 (ULN2803) or BCR503 transistor.
[image]

When using PCF8574, pulling up with resistors solves the problem.
You just need to watch out for one problem... related to the way the port is handled by the P019 plugin.
I had a situation where the _PCFGPIO,1,1_ and _PCFGPIO,1,0_ port controls worked without any problems, but _PCFGPIOtoggle,1_ did not.
Cause:
- too large a resistor pulling up to 3.3V (4.7k) caused that in the high state on the PCF port, together with the built-in resistor in the BCR503 transistor (just like ULN2003), a voltage divider of 1.2V was created, which the PCF constantly read as state 0  ( after PCFGPIO command,1,1)

- with the PCFGPIOtoggle,1 command, ESP checks the input status (reading 0 at a voltage of 1.2V) and sets it to 1, subsequent _PCFGPIOtoggle,1_ commands do nothing because ESP still reads 0
Another effect is that issuing the _PCFGPIOtoggle_ command to any of the ports reads the entire port (and since the resulting voltage divider for PCF8574 is constantly at 0), it causes all other outputs to be reset to zero.

Solution: reducing the pull-up resistor to 1k, the voltage is then 2V, which is easily interpreted as 1

With 1K pull-up resistors, everything is OK and the problem is solved.

PS. In the modified version, I used MCP23017, which solves the problem with pull-up resistors (they are not needed) and after switching on, the output is set to 0.

uzi18, 2024-06-20:
ok but for BCR503 default is relay on, with IRLML2244 you will get default off behavior.