letscontrolit/ESPEasy #5078: [BUG] P087 Serial Proxy plugin not correct handle 0x00 sequence in the middle of hex string

tabunslonopotamov, 2024-06-29:
Steps to reproduce the behavior:
1. Go to 'Tools - Command'
2. entry command: serialproxy_writemix,0x1122334455000000FFFFFF
3. Show log
4. See info message HTTP: serialproxy_writemix,0x1122334455000000FFFFFF
1915868: Proxy: Sending 5 bytes.

Bytes 000000FFFFFF is stop processing and no send to serial.

TD-er, 2024-06-29:
Maybe since that's considered a 0-terminator value in a string?
Have to check the source code, but I think it will be something like that.

tabunslonopotamov, 2024-06-29:
veroyatneye vsego. No ya ne nashel gde.
Zato nashel v shapke plagina upominaniye pro "".

No vse ravno ne rabotayet.

Most likely. But I didn't find where.
But I found a mention of “2023-03-25 tonhuisman: Change serialproxy_writemix to handle 0x00 also, by implementing parseHexTextData()” in the plugin header.

But it still doesn't work.

tonhuisman, 2024-06-29:
@tabunslonopotamov What exact build do you have installed on your ESP? The build name and date is on the lower right of nearly all pages of the Web UI.

tabunslonopotamov, 2024-06-29:
I used the following workaround
it worked. 
but it looks terrible.

#sp_whex,01101870000204460006004863
serialproxy_writemix,'0x01','0x10','0x18','0x70','0x00','0x02','0x04','0x46','0x00','0x06','0x00','0x48','0x63'

#sp_whex,01101870000204080000005CEB
serialproxy_writemix,'0x01','0x10','0x18','0x70','0x00','0x02','0x04','0x08','0x00','0x00','0x00','0x5C','0xEB'

#sp_whex,011018700002044400100047BB
serialproxy_writemix,'0x01','0x10','0x18','0x70','0x00','0x02','0x04','0x44','0x00','0x10','0x00','0x47','0xBB'

#sp_whex,01101870000204008000005F63
serialproxy_writemix,'0x01','0x10','0x18','0x70','0x00','0x02','0x04','0x00','0x80','0x00','0x00','0x5F','0x63'

#sp_whex,01101870000204440029FF15AB
serialproxy_writemix,'0x01','0x10','0x18','0x70','0x00','0x02','0x04','0x44','0x00','0x29','0xFF','0x15','0xAB'

#sp_whex,01101870000204460010004603
serialproxy_writemix,'0x01','0x10','0x18','0x70','0x00','0x02','0x04','0x46','0x00','0x10','0x00','0x46','0x03'

tabunslonopotamov, 2024-06-29:
> @tabunslonopotamov What exact build do you have installed on your ESP? The build name and date is on the lower right of nearly all pages of the Web UI.

Build: [ESP_Easy_mega_20240629_custom_ESP32_4M316k_LittleFS_ETH Jun 29 2024](https://github.com/letscontrolit/ESPEasy/releases)

I am use Custom.h

tonhuisman, 2024-06-29:
You don't need the quotes.

I've found the culprit, and am adding it to PR #5077 

What version of the source are you using? Cloned and up to date from `mega` or downloaded and extracted Releases source zip?

tabunslonopotamov, 2024-06-29:
> What version of the source are you using? Cloned and up to date from `mega` or downloaded and extracted Releases source zip?

I used the "git pull" command in the Visual Studio Code console.

Command "git branch" report * mega

tonhuisman, 2024-06-29:
You can get the fix by issuing these commands in the console (will pull the changes into a new branch so you can later return to the `mega` branch once it's merged)

```
git checkout -b pulls/5077
git fetch origin pull/5077/head
git pull origin pull/5077/head
```

To update to later commits I might add there, you can repeat the last 2 commands when having that `pulls/5077` branch active.
To return to the original `mega` branch use `git checkout mega`.

tabunslonopotamov, 2024-06-29:
I can do it a few hours later

PR #5077 (2024-07-10): [2024-07] Improvements and corrections
Resolves #5078 

Features:
- [Build] Add P095 to `Collection` builds (ESP32 only, except `IRExt` builds) to match documentation
- [P047] Fix compiler argument warnings on I2C calls
- [P087] Fix `parseHexTextData()` unwanted increment during parsing
- [Docs] GPIO pins > 16 allowed for ESP32 corrections, README.md updates
- ?
closed 2024-07-10 by TD-er