letscontrolit/ESPEasy #5092: SHTC3 digital humidity sensor

ilsicca, 2024-07-29:
Hi! 
I need to add a SHTC3 I2C sensor on ESP32 C6. ESPEasy dont' have a plugin for this sensor.
[DFRobot SHTC3 Sensor](https://wiki.dfrobot.com/SHTC3_Humidity_and_Temperature_Sensor_SKU_SEN0414)
How I can add it?
 
I2C address: 0x70
Board: FireBeetle 2 ESP32-C6
Build: ESP_Easy_mega_20240729_normal_ESP32c6_4M316k_LittleFS_CDC Jul 29 2024 with ENVIRONMENT SENSORS

uzi18, 2024-07-30:
i2c address is different, commands are not bytes but words : 
https://github.com/cdjq/DFRobot_SHTC3/blob/master/DFRobot_SHTC3.h
https://github.com/cdjq/DFRobot_SHTC3/blob/master/DFRobot_SHTC3.cpp

tonhuisman, 2024-07-30:
I've reserved Plugin ID `P173` for for this plugin.
Probably won't need the library, as it's quite straightforward and can use already available I2C and CRC functions, available in ESPEasy core.

ilsicca, 2024-08-01:
@tonhuisman tranks!

tonhuisman, 2024-08-02:
Well, it still has to be implemented, usually we keep issues open and close them automatically when the feature/fix is merged, so you will be notified 😸

tonhuisman, 2024-08-27:
@ilsicca I've created a PR to support this SHTC3 sensor, are you available to test?

Builds are available from [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10584507484)

ilsicca, 2024-08-27:
Hi Ton! Perfect, I will test it.

ilsicca, 2024-08-27:
@tonhuisman , Can you add P173 in a ESP_Easy_mega_20240827_normal_**_ESP32c6_**_**_4M_**316k_LittleFS_CDC_ETH.factory.bin release? because I'm using a FireBeetle 2 ESP32-C6 and it has only 4MB. 
Thanks!

tonhuisman, 2024-08-28:
Changing the `NORMAL` build configuration isn't in the plans, but the next best thing is a `Custom` build with this plugin included.

[ESP_Easy_mega_20240828_custom_ESP32c6_4M316k_LittleFS_CDC_ETH.zip](https://github.com/user-attachments/files/16778473/ESP_Easy_mega_20240828_custom_ESP32c6_4M316k_LittleFS_CDC_ETH.zip)
(Both the OTA-update .bin and the .factory.bin file are included)

If there is anything from `NORMAL` that you need but is not in this build, please reply, and I'll add it.

tonhuisman, 2024-08-30:
@ilsicca I've added the documentation for the plugin, and also added it to the default Custom build for ESP32-C6, so you can download the latest once [this GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10629265465) is completed.

ilsicca, 2024-08-30:
Amazing work @tonhuisman ! Thank you very much!

tonhuisman, 2024-08-30:
There are 2 possible causes here:
- The plugin doesn't respond immediately after power-on to the I2C check, done by ESPEasy for (nearly) all I2C plugins when starting the plugin
- The plugin doesn't reply with the correct Device ID during initialization of the plugin, as that's checked by the code.

The first cause can be verified by disabling the Tools/Advanced setting `Check I2C devices when enabled`, and replaying your scenario.
The second cause is not much I can do about, I think, adding delays 'somewhere' doesn't seem like a solid solution.

@ilsicca Can you try by un-checking that setting and replay your scenario? (The initial I2C check can also be disabled on a per-plugin basis)

ilsicca, 2024-08-30:
@tonhuisman I disabled the option: _Check I2C devices when enabled_ but the problem continue. 
[image]

TD-er, 2024-08-30:
How do you enable the task?
If you just enable it using the command `taskenable`, then the settings are not saved with this checkbox set to enabled.
So after a reboot the settings you load from flash then still have this task set to disabled.

tonhuisman, 2024-09-04:
@ilsicca Any update on this?

ilsicca, 2024-09-07:
@tonhuisman This is my task configuration for the sensor, I enable it with task settings:
[image]
I disable the option: "Check I2C devices when enabled" but the problem continue.
I enable the option: "Check I2C devices when enabled" but the problem continue.

tonhuisman, 2024-09-07:
I received a SHTC3 this week, and wired it to an ESP32-C6 (4MB), installed a fresh Custom build of this PR (only difference it that it's using a somewhat newer version of the platform), and tested your scenario: SDA on GPIO19, SCL on GPIO20, default sensor pull-up resistors (10k), enabled task, Interval 5 sec, I2C device-check enabled (Tools/Advanced), let it run for some time (~10 minutes), unplug the ESP, wait a few seconds, re-plug the ESP, and watch the logging. It's starting just fine, right at boot, and keeps running fine, so far, even after multiple power off/on tests.

This is my setup (sensor-cable is ca. 1 m):
[image]

I assume you're running your sensor from 3.3V and not 5V? AFAICS, it's not really 5V tolerant (data sheet: VDD max 4.0V), and the GPIO pins of the ESP also don't work well/long when being pulled to 5V.

The actual build I'm running is this one:
[ESP_Easy_mega_20240907_custom_ESP32c6_4M316k_LittleFS_CDC_ETH.zip](https://github.com/user-attachments/files/16917384/ESP_Easy_mega_20240907_custom_ESP32c6_4M316k_LittleFS_CDC_ETH.zip)

can you please verify if this makes any difference?

ilsicca, 2024-09-18:
hi @tonhuisman I tried with your build and it works fine!! Thanks you!

PR #5114 (2024-09-28): [P173] Add plugin Environment - SHTC3
Resolves #5092 

Features:
- [P173] Add plugin Environment - SHTC3 (Temperature/Humidity) (Similar to P153 - SHT4x, with less configuration options)
- [I2C] Add functions `I2C_write16()` and `I2C_write16_LE()`
- Available in `CLIMATE` builds (ESP8266 for as long as it fits and ESP32), ESP32 `COLLECTION G` and `MAX` builds.
- Added to default ESP32-C6 Custom build.

TODO:
- [x] Documentation
- [x] Testing (~no device available, so help is welcome~) ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/5092#issuecomment-2318159957) [confirmed 2](https [... cut]
closed 2024-09-28 by TD-er