letscontrolit/ESPEasy #5096: [BUG] Pulse counter issue

themanfrommoon, 2024-08-06:
**Describe the bug**
On the website https://espeasy.readthedocs.io/en/latest/Plugin/P003.html the pulse counter is described.
There ist stated: "Counter Type: (practical use of this parameter is unclear, but kept for compatibility reasons)"
[image]
Looks like somebody who originally developed this has left the team and did no documentation?!
I think it is pretty clear.
This option is choosing which values are processed.
1. "delta" = only delta value = 1 value
2. "delta, total and time" = all 3 values
3. "total" = only total value = 1 value
4. "delta, total" = delta and total value = 2 values
Like this two options are missing
5. "time"
6. "total, time"
7. "time, delta"

**To Reproduce**
On the website https://espeasy.readthedocs.io/en/latest/Plugin/P003.html the pulse counter is described.
There ist stated: "Counter Type: (practical use of this parameter is unclear, but kept for compatibility reasons)"
[image]
Looks like somebody who originally developed this has left the team and did no documentation?!
I think it is pretty clear.
This option is choosing which values are processed.
1. "delta" = only delta value = 1 value
2. "delta, total and time" = all 3 values
3. "total" = only total value = 1 value
4. "delta, total" = delta and total value = 2 values
Like this two options are missing
5. "time"
6. "total, time"
7. "time, delta"

**Expected behavior**
On the website https://espeasy.readthedocs.io/en/latest/Plugin/P003.html the pulse counter is described.
There ist stated: "Counter Type: (practical use of this parameter is unclear, but kept for compatibility reasons)"
[image]
Looks like somebody who originally developed this has left the team and did no documentation?!
I think it is pretty clear.
This option is choosing which values are processed.
1. "delta" = only delta value = 1 value
2. "delta, total and time" = all 3 values
3. "total" = only total value = 1 value
4. "delta, total" = delta and total value = 2 values
Like this two options are missing
5. "time"
6. "total, time"
7. "time, delta"

**Screenshots**
My problem in this case is the following (for me this is enough if this can be solved):

I just need the "delta" value.
But no matter what I choose always all three values are send to the controller on the same channel and this causes a big mess.
So how is it possible to only send the "delta" value to the controller?
[image]

I have following firmware:
[image]

**Used platform (please complete the following information):**
 - ESP type: [e.g. ESP8266, ESP32, ESP32-S2, ESP32-S3, ESP32-C3]
 - Build version: [e.g. self built, mega-YYYYMMDD]
 - Build set [e.g. normal, energy, display, climate, collection-X, max]

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: [e.g. 4M]
 - Brand/Model: [e.g. Sonoff Basic r2, NodeMCU, Wemos D1]
 - Browser + OS [e.g. Windows 11/Chrome, IOS/safari]

**Additional context**
Add any other context about the problem here.

themanfrommoon, 2024-08-06:
this is shown in ESPeasy -> Devices:
[image]

and this is the result in my database:
[image]

...I only want the first value ("count"), but not the other two values

tonhuisman, 2024-08-06:
Ah, this plugin uses an outdated method to change the intended number of output values. I'll fix that soon.

Please keep this issue open, it will be closed automatically once the pull request for the fixes is merged.

tonhuisman, 2024-08-06:
@themanfrommoon I have already created the PR, and an GH Actions run build is [running here](https://github.com/letscontrolit/ESPEasy/actions/runs/10274086995). Can you test if this fixes your issue?

I'll update, and extend where applicable, the documentation later this week.

themanfrommoon, 2024-08-07:
Dear tonhuisman,

thank you very much.
I uploaded the new firmware and it looks correct at the moment
[image]
But what is the new option "Data source"?
[image]

If it works now I need to wait until the gas heater is working again, I will come back and report

Thank you and best regards,
Christian

tonhuisman, 2024-08-07:
> what is the new option "Data source"?

That shows the unit for a remote device this task can represent, when configured via the [P2P Controller](https://espeasy.readthedocs.io/en/latest/Controller/C013.html#c013-page). This has been made visible for all tasks recently, as work is in progress to simplify the configuration for that feature.

tonhuisman, 2024-08-07:
@themanfrommoon I've added the extra possible Counter Type options, can you test these too? They won't be available in most ESP8266 builds, as the feature is only included for builds without the `LIMIT_BUILD_SIZE` compile-time flag, and most ESP8266 builds have that...
New [GH Actions run here](https://github.com/letscontrolit/ESPEasy/actions/runs/10289710551)

themanfrommoon, 2024-08-07:
@tonhuisman Sorry, I can not test it, because I'm running the counter in a productive environment, writing monitoring data in a database, which seems to work now, and additionally the firmware I use is a special ESP8266 "climate" build (so I guess my hardware is not able to do it)

themanfrommoon, 2024-08-07:
Another additional thing is the following.
Maybe I need another configuration of the counter?!
In the database it looks like this:
[image]
The configuration looks like this:
[image]

I set the interval to 600 seconds (10minutes) to avoid too many "0" values in the database because they make no sense, but on the other hand for a precise interpretation of the monitored values it would be much better to set the interval 20 or 60 seconds.

I think the best option would be that the interval is maybe 20 seconds after the first count of something bigger then "0" (so >0) until the first "0" within the interval. But when the last value was already a "0" then no additionally "0" needs to be send to the database (...because for what reason? It only fills the database with unneccessary data).

Is this possible and what do I need to configure?

Thank you and best regards,
Chris

tonhuisman, 2024-08-07:
Maybe an optional check on the Delta value would be enough? So if 0 since last Interval no event is to be generated.

tonhuisman, 2024-08-08:
I've added a new setting `Ignore Delta = 0` for this plugin, available in this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10303962051).

themanfrommoon, 2024-08-08:
Thank you so much, I update the firmware and test it and come back to you how it works (but testing will take some days because the gas heater is not running so much in the summer)

themanfrommoon, 2024-08-08:
...hmm, it seems to be that I cannot activate the new option "Ignore Delta = 0".
If I check it and submit, it confirms "submitted" and loads the page new. But then it is unchecked again!

tonhuisman, 2024-08-08:
> I cannot activate the new option "Ignore Delta = 0"

I made a typo/copy-paste error, fixed it in a new [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10310207655)

themanfrommoon, 2024-08-09:
seems to work now, thank you very much, I test it now and come back to you later

themanfrommoon, 2024-08-10:
@tonhuisman Now I understand what do did with the option "Ignore Delta = 0". It ignores all 0 values. But this does not help.
Meanwhile I am not sure if the concept of the impulse counter ist the right concept, or if I did not understand it correct.

I try to explain what the goal is:
There is a gasmeter which gives an impuls ("1" in the database) for every used 0,01m³ gas.
With this information and the timestamp it is possible to calculate the consumption per hour (m³/h).
So I need the information "0" for no gas consumption.
But when the gas is flowing I need the "1" and the timestamp to calculate the flowrate.
For my understanding this means:
I only need one "0" if no gas is flowing (...and not fill up the database with many "0" when no gas flows)
And I need all "1" and the timestamp when the gas flows.
This means when the gas stops to flow, there must be a "0" in the database.
And before the gas starts to flow there must also be a "0" in the database.

Here is the original behaviour when all "0" are written in the database:
[image]

At the moment only the "1" impulses are written in the database:
[image]

I think it must look like this.
Only "0" before the "1" and after the "1" impulses needs to be written in the database like this:
[image]

Additionally I have another question:
If I understand it correct, the counter counts every impulse until the interval time and then sends the summed up counts.
But for a gas flowrate I need the exact period between two impulses, which means that the impulse needs to be send to the database exactly when it happens. Does it mean I have to set the interval to 0? Or does the ESP8266 then it totally busy by sending data to the database without any pause? (I expect the impulses maybe every 16-20 seconds at highest flowrate). I think when you have higher impulserates like some thousands or tenthousands per minute the counter and database can come into troubles.

Thank you and best regards,
Chris

tonhuisman, 2024-08-10:
> I only need one "0" if no gas is flowing

Ah, yes, that sounds logical, the database/report can't know when the interval is over, so doesn't insert the required 0, I'll fix that.

> Does it mean I have to set the interval to 0?

By default you can't set the Interval to 0 for this plugin, but that is a compile-time option.
This is also a quite logical request, but a bit more tricky. I'm working on a fix/feature for this, to immediately send out the data when Interval is set to 0.
Only 'restriction' would be that the actual pulse-frequency shouldn't be too high (> 25 Hz I would think, the plugin does support quite high frequencies > 10 **k**Hz), or the ESP and the Controller backend (database) could become too busy processing the constant stream of data, as this is checked at 50 Hz (every 20 msec.). Though the Controller Minimum Send Interval and subsequent buffering should flatten that out a bit.

I'll put a big warning for this situation in the documentation, and also set the default Interval at the current default (usually 60 seconds), because when the 0-Interval option is enabled, the default for a new task will be 0...

tonhuisman, 2024-08-10:
@themanfrommoon Improvements are brewing in this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10331692241)

themanfrommoon, 2024-08-12:
I'm not so sure, if it works or not (up til now no new pulses are written in the database)
Did I setup everything correct?
"Ignore multiple delta = 0:" = enabled
"Interval:" = 0

[image]

tonhuisman, 2024-08-12:
> I'm not so sure, if it works or not (up til now no new pulses are written in the database)

I had a few mistakes in my code, but I've hooked up a pulse source (a push button 😅) and did some more testing, so was able to correct that. A new [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/10351440774) is started.

PR #5098 (2024-08-21): [P003] Countertype to determine the number of values
Resolves #5096 

- Bugfix for `Pulse Counter` to set the correct number of output values (and value), so Controllers will send out the data as intended.
- Add extra value combinations `Time`, `Time/Total` and `Time/Delta` (only for non-limited builds).
- Add extra setting 'Ignore Delta = 0' to suppress sending out events and data to controllers if Delta (Count) is 0 since the last read action.

TODO:
- [x] Testing by reporter. ([confirmed for part 1](https://github.com/letscontrolit/ESPEasy/issues/5096#issuecomment-2272565799)).
- [x] Testin [... cut]
closed 2024-08-21 by TD-er