letscontrolit/ESPEasy #5102: [BUG] Shelly Plus 1 (blue case) boot loop

stwa2020, 2024-08-11:
Shelly Plus 1 (blue case) is in a boot loop

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32-U4WDH (revision 3)
 - Number of Cores: 1
 - Max CPU Frequency: 80MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: YES
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 0C:DC:7E:F7:15:54

Actions done:
1. Used ESP32 Flash download tool 3.9.7 and wrote blank_4MB.bin to starting address 0x0
2. Flashed with ESP_Easy_mega_20240414_climate_ESP32solo1_4M316k_LittleFS.factory.bin starting address 0x0
3. Reboot
4. Check with Serial Console, boot loop / CPU Software reset

Settings
[image]

[image]

**ESP32 Flash download tool 3.9.7 output** 
_E:\ESPeasy\ESPEasy_mega_20240414_ESP32solo1_binaries\bin\ESP_Easy_mega_20240414_climate_ESP32solo1_4M316k_LittleFS.factory.bin
test offset :  0 0x0
case ok
.
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 1152000
Changed.
FLASH_CRYPT_CNT 0
ABS_DONE_0 False
Compressed 1517296 bytes to 996481...

 is stub and send flash finish_

**Serial Console output right after reboot**
_Using 'COM5' as serial port.
Showing logs:
[16:19:16]04.651 : (333844) Info   : SaveToFile: ControllerSettings index: 0
[16:19:16]04.654 : (333844) Info   : SaveToFile: free stack: 7280
[16:19:18]06.819 : (333828) Info   : FILE : Saved config.dat offset: 8192 size: 820
[16:19:18]06.820 : (333764) Info   : SaveToFile: free stack after: 7280
[16:19:18][16:19:18]06.847 : (333624) Info   : SaveToFile: free stack: 7424
[16:19:21]09.160 : (333544) Info   : FILE : Saved config.dat offset: 0 size: 3028
[16:19:21]09.161 : (333472) Info   : SaveToFile: free stack after: 7424
[16:19:21]09.163 : (333424) Info   : SaveToFile: free stack: 7344
[16:19:21]09.270 : (333344) Info   : FILE : Saved security.dat offset: 0 size: 593
[16:19:21]09.271 : (333280) Info   : SaveToFile: free stack after: 7344
[16:19:21]09.278 : (332044) Info   : SaveToFile: ExtendedControllerCredentials index: 0
[16:19:21]09.280 : (332060) Info   : SaveToFile: free stack: 7088
[16:19:21]09.371 : (332032) Info   : FILE : Saved security.dat offset: 1024 size: 6
[16:19:21]09.372 : (331968) Info   : SaveToFile: free stack after: 7088
[16:19:21]RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)
[16:19:22]10.384 : (332956) Info   : WiFi : WifiDisconnect()
[16:19:22]ets Jul 29 2019 12:21:46
[16:19:22]
[16:19:22]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:19:22]configsip: 188777542, SPIWP:0xee
[16:19:22]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:19:22]mode:DIO, clock div:2
[16:19:22]load:0x3fff0030,len:184
[16:19:22]load:0x40078000,len:13120
[16:19:22]load:0x40080400,len:4
[16:19:22]ho 8 tail 4 room 4
[16:19:22]load:0x40080404,len:2736
[16:19:22]entry 0x40080544
[... 110 lines cut ...]
[16:40:38]00.319 : (334036) Info   : NVS  : Save Console
[16:40:38]00.332 : (334060) Info   : INIT : SPI not enabled
[16:40:38]00.333 : (334012) Info   : Set Network mode: WiFi
[16:40:38]00.438 : (333964) Info   : WIFI : Set WiFi to STA
[16:40:38]ets Jul 29 2019 12:21:46
[16:40:38]
[16:40:38]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:40:38]configsip: 188777542, SPIWP:0xee
[16:40:38]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:40:38]mode:DIO, clock div:2
[16:40:38]load:0x3fff0018,len:4
[16:40:38]load:0x3fff001c,len:1044
[16:40:38]load:0x40078000,len:8896
[16:40:38]load:0x40080400,len:5828
[16:40:38]entry 0x400806ac
[16:40:39][16:40:39]
[16:40:39]INIT : Booting version: ESP_Easy_mega_20240414_climate_ESP32solo1_4M316k_LittleFS, (GitHub Actions) mega-20240414 (ESP32 SDK 5.1.3.240325)
[16:40:39]00.166 : (337360) Info   : INIT : Free RAM:337360
[16:40:39]00.178 : (337216) Info   : INIT : Soft Reboot #2 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU_

Please let me know if I should test other settings or Firmware version 

Thx and best regards,

Stefan

TD-er, 2024-08-11:
Can you also try to flash the solo1 build from here: https://td-er.nl/ESPEasy/latest/ 
?

stwa2020, 2024-08-12:
Still in a boot loop, 

_Showing logs:
[08:13:52]04.578 : (342336) Info   : SaveToFile: ControllerSettings file: config.dat task: 1
[08:13:52]04.580 : (342372) Info   : SaveToFile: free stack: 7284
[08:13:54]06.701 : (342360) Info   : FILE : Saved config.dat offset: 8192 size: 820
[08:13:54]06.701 : (342308) Info   : SaveToFile: free stack after: 7284
[08:13:54][08:13:54]06.728 : (342164) Info   : SaveToFile: free stack: 7444
[08:13:56]09.009 : (342112) Info   : FILE : Saved config.dat offset: 0 size: 3028
[08:13:56]09.009 : (342060) Info   : SaveToFile: free stack after: 7444
[08:13:56]09.011 : (342024) Info   : SaveToFile: free stack: 7364
[08:13:56]09.107 : (341972) Info   : FILE : Saved security.dat offset: 0 size: 593
[08:13:56]09.107 : (341920) Info   : SaveToFile: free stack after: 7364
[08:13:56]09.114 : (340680) Info   : SaveToFile: ExtendedControllerCredentials file: security.dat task: 1
[08:13:56]09.126 : (340732) Info   : SaveToFile: free stack: 7092
[08:13:56]09.206 : (340720) Info   : FILE : Saved security.dat offset: 1024 size: 6
[08:13:56]09.206 : (340668) Info   : SaveToFile: free stack after: 7092
[08:13:56]RESET: Successful, rebooting. (you might need to press the reset button if you've just flashed the firmware)
[08:13:57]10.218 : (341656) Info   : WiFi : WifiDisconnect()
[08:13:57]ets Jul 29 2019 12:21:46
[08:13:57]
[08:13:57]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:13:57]configsip: 188777542, SPIWP:0xee
[08:13:57]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:13:57]mode:DIO, clock div:2
[08:13:57]load:0x3fff0030,len:184
[08:13:57]load:0x40078000,len:13120
[08:13:57]load:0x40080400,len:4
[08:13:57]ho 8 tail 4 room 4
[08:13:57]load:0x40080404,len:2736
[08:13:57]entry 0x40080544
[08:13:58][08:13:58]
[08:13:58]INIT : Booting version: ESP_Easy_mega_20240715_normal_ESP32solo1_4M316k_LittleFS_ETH, (GitHub Actions) HEAD_6d9bdb0 (ESP32 SDK 5.1.4.240625)
[08:13:58]00.153 : (345544) Info   : INIT : Free RAM:345544
[08:13:58]00.166 : (345412) Info   : INIT : Soft Reboot #1 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU
[08:13:58]00.167 : (345376) Info   : FS   : Mounting...
[08:13:58]00.201 : (343520) Info   : FS   : LittleFS mount successful, used 151552 bytes of 323584
[08:13:58]00.234 : (343356) Info   : CRC  : Settings CRC...OK
[08:13:58][08:13:58]00.275 : (343168) Info   : CRC  : SecuritySettings CRC...OK 
[08:13:58]00.287 : (343072) Info   : NVS  : Save FacDefPref
[08:13:58]00.289 : (343024) Info   : NVS  : Save UnitName
[08:13:58]00.290 : (342988) Info   : NVS  : Save WIFI_Flags
[08:13:58]00.291 : (342952) Info   : NVS  : Save WIFI_SSID1
[08:13:58]00.302 : (342916) Info   : NVS  : Save WIFI_PASS1
[08:13:58]00.304 : (342880) Info   : NVS  : Save WIFI_AP_PASS
[08:13:58]00.305 : (342844) Info   : NVS  : Save WiFI_IP
[08:13:58]00.317 : (342808) Info   : NVS  : Save ETH_IP
[08:13:58]00.317 : (342772) Info   : NVS  : Save ETH_HW_CONF
[08:13:58]00.318 : (342752) Info   : NVS  : Save Log
[08:13:58]00.329 : (342716) Info   : NVS  : Save Console
[08:13:58]00.331 : (342692) Info   : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
[08:13:58]00.343 : (342656) Info   : INIT : SPI not enabled
[08:13:58]00.448 : (342620) Info   : WIFI : Set WiFi to STA
[08:13:58]ets Jul 29 2019 12:21:46
[08:13:58]
[08:13:58]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[08:13:58]configsip: 188777542, SPIWP:0xee
[08:13:58]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[08:13:58]mode:DIO, clock div:2
[08:13:58]load:0x3fff0030,len:184
[08:13:58]load:0x40078000,len:13120
[08:13:58]load:0x40080400,len:4
[08:13:58]ho 8 tail 4 room 4
[08:13:58]load:0x40080404,len:2736
[08:13:58]entry 0x40080544
[08:13:58][08:13:58]
[08:13:58]INIT : Booting version: ESP_Easy_mega_20240715_normal_ESP32solo1_4M316k_LittleFS_ETH, (GitHub Actions) HEAD_6d9bdb0 (ESP32 SDK 5.1.4.240625)
[08:13:58]00.155 : (345544) Info   : INIT : Free RAM:345544
[08:13:58]00.168 : (345412) Info   : INIT : Soft Reboot #2 Last Action before Reboot: Background Task Last systime: 0 - Restart Reason: CPU0: Software reset CPU_
....

TD-er, 2024-08-12:
Hmm the max. frequency of 80 MHz you mention seems a bit odd.
I have to check if ESPEasy still might try to switch to 160 MHz for the solo1.

TD-er, 2024-08-12:
According to [this document](https://www.espressif.com/sites/default/files/pcn_downloads/PCN-2021-021%20ESP32-U4WDH%20升级为双核处理器产品.pdf), the chip you have should be capable of running at 160 MHz if it is still the single-core revision.
If it is another revision, then [this document](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) (page 11) states it is a dual core.

tonhuisman, 2024-08-12:
Is this Shelly powered from an USB-Serial adapter? As that might be the cause of the reboot when starting WiFi, because that adapter probably can't supply the WiFi-power-peak requested by the ESP.

stwa2020, 2024-08-12:
Shelly Plus 1 is powered by a USB-serial adapter connected to the USB 3.0 port, I'll try it again without USB and connect it directly to the power source, of course I won't be able to check the boot process in this setup.

I've opened the Shelly Plus 1, 

[image]

[image]

TD-er, 2024-08-12:
In theory, the solo1 builds should work on any ESP32-classic (thus non C2/C3/C6/S2/S3...)
It is 'just' a stripped down version.
However I am not 100% sure it is still using the same revision of the SDK builds as I need to update that line by hand when I update to the later SDK.

Given when the unit reboots, I think Ton's argument about power consumption may be a valid one as the ESP shortly takes upto 500 mA when starting WiFi.
So if the header doesn't make proper connection, or you're using too thin wiring (or the voltage regulator isn't able to keep up) then the ESP may reboot.

stwa2020, 2024-08-12:
The power source was the problem, I connected to 240V as recommended by @tonhuisman and the Shelly Plus 1 started with the ESPSolo1 image.
I organised a new Shelly Plus 1, connected it to the console via USB with the original firmware and it was the same, it was in a boot loop.

closed 2024-08-12 by stwa2020

tonhuisman, 2024-08-12:
Ah, great you found the cause, and thanks for reporting back.

As an alternative, you could have used a 3.3V external power supply with shared gnd to power the Shelly while having a serial connection with the computer. That's my go-to solution for devices that don't properly run from such USB-serial adapter.

TD-er, 2024-08-12:
Just make sure the external power does like to have the ground connected to the GND of the PC.
If the external power supply is also connected to other devices, then you may run into issues when connecting the USB cable to your PC as there may be significant currents flowing through GND and/or the USB cable.

I've seen USB cables light up literally and just by luck the connected computer wasn't damaged.

I always try to use a 60Watt powered USB hub made by Anker (so no shady Chinese brand) and have some modified USB to serial adapter which can handle upto 1A.
Or when connecting stuff to my PC which I know might not be isolated from mains, I use an isolation transformer (those can be expensive) and an USB isolator.
Such an USB isolator can often not handle more than 100-ish mA delivering power, but you can use it to isolate the USB data lines. Only supports 12 Mbps USB1.1, but that's enough for programming and debugging.