letscontrolit/ESPEasy #5103: [Test build] Testers for new upcoming build (August '24)

TD-er, 2024-08-12:
I've just uploaded the [latest GH Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/10349331884)  to the [web flasher](https://td-er.nl/ESPEasy/latest/)

So I would like to summon the very appreciated testing crew :)
@fly74 @ghtester @chemmex @alex-kiev @iz8mbw @Wookbert @Dickesplumpel

Just as a reminder:
- ESP32-xx LittleFS builds are based on ESP-IDF 5.x code
- ESP32-xx SPIFFS builds are based on ESP-IDF 4.4

There are lots and lots of changes, but the main tricky one is that it might be impossible to OTA update larger builds from the April release as there was a bug in the Arduino code which did not properly use timeouts when handling HTTP requests.

If this is an issue, to update any "max" builds, I will add intermediate builds for those typically 16M and 8M builds which do not contain any plugin or controller to make them as small as possible.
This way it is more likely you can successfully OTA update such units without the need to hook them up via serial.

So please let me know if this is indeed an issue (and maybe also for "4M builds LittleFS").
N.B. this does only apply to "LittleFS" builds when OTA updating from the April release. (or test builds from around that time)

tonhuisman, 2024-08-12:
That's being succeeded by `ESP_Easy_mega_20240812_normal_ESP32_4M316k_LittleFS_ETH`, to reduce the number of builds (and reduce the GH Actions runs processing time)

fly74, 2024-08-12:
What means ETH?

tonhuisman, 2024-08-12:
Ethernet support, now also supporting SPI Ethernet modules, like W5500 and a few others. List is in the UI on the Hardware tab.

TD-er, 2024-08-12:
Yep, Ethernet support is now enabled for (nearly) all ESP32-xx LittleFS builds as it does hardly make the builds larger and thus it makes no sense to have separate builds with and without Ethernet support.

For the ESP32-C2 I have not added Ethernet support as nearly all modules out there with a C2 are embedded devices like of-the-shelve modules which do not have Ethernet and the C2 already is quite low on resources to begin with.

fly74, 2024-08-13:
But when it is the only normel LittleFS build for ESP32 then ETH could be removed? Then I wouldn't had asked.

fly74, 2024-08-13:
I'm a little confused, as I understand there is only one build from now for "ESP32 normal"
"ESP_Easy_mega_20240812_normal_ESP32_4M316k_LittleFS_ETH" and it replaces also "ESP_Easy_mega_20240812_normal_ESP32_4M316k_LittleFS" ?!?

TD-er, 2024-08-13:
Yep, we simply enabled Ethernet on all ESP32-variant LittleFS builds, all except for the C2.

It just has some extra feature included which doesn't prevent using WiFi when you don't specifically setup Ethernet configuration.

Dickesplumpel, 2024-08-13:
This time I'm not taking part, weather drags me to outside jobs and motorcycling, have fun!

tonhuisman, 2024-08-13:
> This time I'm not taking part, weather drags me to outside jobs and motorcycling, have fun!

That are the perks of making a (high-)summer release 😄. Have fun! 🌞

TommoT1, 2024-08-13:
Hi there!
Are the 1M builds for 8266 missing on purpose?

tonhuisman, 2024-08-13:
> Are the 1M builds for 8266 missing on purpose?

No, not on purpose, and a few specific 1M builds are included, but not the `Normal` builds it seems, they have fallen off (again) because of a few small additions to some plugins.
What exact 1M (standard) build do you use/need?

TommoT1, 2024-08-13:
The "normal" ESP_Easy_mega_xxxxxxxx_normal_ESP8266_1M_VCC and the one without VCC.

By the way, do you guys notice some stability differences between 8266, 32 and 32c3?
I have some issues from time to time mostly with 32c3 devices not beeing in the WiFi anymore....

tonhuisman, 2024-08-13:
What is the current build you are using, 2024-04-14 ?

TommoT1, 2024-08-13:
20240617 (a build with again working fhem plugin and no OTA issues anymore)

tonhuisman, 2024-08-13:
> A few minutes ago I reverted back to Ton's Jul 30 PR build because it has his VL53L1X ROI feature.

I rebased that PR earlier today to again be on par with the latest `mega`, so you _can_ get the latest P113 build [from here](https://github.com/letscontrolit/ESPEasy/actions/runs/10365761997) 😃

tonhuisman, 2024-08-13:
> 20240617 (a build with again working fhem plugin and no OTA issues anymore)

Are there specific plugins that you absolutely can't live without? As I'm considering to leave 1 plugin out of the 1M builds only: P011 (ProMini Extender, it requires quite specific hardware, an Arduino Nano with the 'extender' software on it)...

tonhuisman, 2024-08-13:
> I have some issues from time to time mostly with 32c3 devices not beeing in the WiFi anymore....

The freshly offered pre-release has a newer version of the platform libraries compared to what you're using, so testing with the latest builds might show improved response/working.

TommoT1, 2024-08-13:
> > 20240617 (a build with again working fhem plugin and no OTA issues anymore)
> 
> @TommoT1 Are there specific plugins that you absolutely can't live without? As I'm considering to leave 1 plugin out of the 1M builds only: P011 (ProMini Extender, it requires quite specific hardware, an Arduino Nano with the 'extender' software on it)...

That sounds good, I anyway moved away from the ProMiniExtender to the MCP23017.

TommoT1, 2024-08-13:
> > I have some issues from time to time mostly with 32c3 devices not beeing in the WiFi anymore....
> 
> The freshly offered pre-release has a newer version of the platform libraries compared to what you're using, so testing with the latest builds might show improved response/working.

Thanks, I'll monitor those

TD-er, 2024-08-13:
> The "normal" ESP_Easy_mega_xxxxxxxx_normal_ESP8266_1M_VCC and the one without VCC.
> 
> By the way, do you guys notice some stability differences between 8266, 32 and 32c3? I have some issues from time to time mostly with 32c3 devices not beeing in the WiFi anymore....

I know there are some C3 boards out there which are extremely cheap. (< 2 euro) 
I have a few of those myself and when they work, they work just fine.
However during manufacturing there was clearly absolutely no quality control.
Some even have the solder balls residue on them which you also see when soldering using expired solder paste (and don't clean the board after you're done)
I also have a board which is missing a few resistors.

So given those have not seen any quality control, I wouldn't be surprised when those may show differences in stability.

Other boards I have with the C3 (more like NodeMCU form factor and 2 USB ports) are just as stable as can be.

There is however an issue which I have not yet been able to fix, where the ESP may loose its WiFi and for whatever reason is not able to reconnect.
If you're seeing such behavior, you might want to check the checkbox "Restart WiFi Lost Conn" on Tools->Advanced page.

I want to make a complete redesign of the network stack in ESPEasy as there is clearly some logic error in my code.
But I would like to do this after this release as it has already been 4 months since the last one due to various reasons...

First I wanted to make sure the OTA update was working again, which took some PR (which got merged) in the Arduino code and then I was extremely busy with other issues regarding how ESPEasy is being built.
So a lot of behind-the-scenes work, which may look like not much is happening here.

TommoT1, 2024-08-13:
Thanks for explaining!
I use ESP8685-wroom-03 and 06 units. They came in reel packaging and look quite clean cause everything ist hidden behind the metal shielding...

The "Restart WiFi Lost Conn" ist checked but that doesn't help. I also addes rules where after 10min with WiFi down a wifidisconnect is done and if no help after three times a reboot shall be done.
Looks it stops those timers when it hangs.
But on the other hand, when in such a condition, hardware attached switches and relays still work.
So luckily this doesn't Happen too often
Let's see what the new build adds to stability.

tonhuisman, 2024-08-13:
@TommoT1 Not sure if you found it already, but I created a PR to fix some ESP8266 1M builds, the Actions run can be [found here](https://github.com/letscontrolit/ESPEasy/actions/runs/10372470739)

TommoT1, 2024-08-13:
Thanks! Got it!

TD-er, 2024-08-13:
@TommoT1 Those Espressif modules should be the perfect choice as they really don't use the cheapest parts available and also have proper QC.
... assuming those are original ones.

So this leaves the usual suspects:
- power supply quality (or connections to it)
- Presence of metal near the antenna.

Just checked 2 ESP32-C3 units I have here (running the new AS3935 Lightning Detector).
These are running on those (too) cheap boards I mentioned and their uptimes are over 39 days and that's also the 'connected time'. (number of reconnects: 0)

- Connected: 43T20:53:14.725
- Connected: 39T21:39:36.556

TommoT1, 2024-08-13:
I put those in exchange for the original Chip in those Meross 4port +USB Power Strips.
Nothing I can do about powersupply and Position.
And I also geht those running times, but obce in a while suddenly one ist Not reachae anymore.
While it still reacts on the Push Button build in this Meross...
So only WiFi is once in a while not playing as it should...
Thanks for your work!

TD-er, 2024-08-13:
It might be related to the 'logic error' I mentioned before in my WiFi code.
That's for the next release.

iz8mbw, 2024-08-14:
Hello @TD-er sorry but cannot test this time.

TD-er, 2024-08-22:
Thanks for testing y'all
New build has just been uploaded to the webflasher and on the releases page.

And you all know it... no critical bugreports in the next 24h... please ;)

closed 2024-08-22 by TD-er

thomastech, 2024-08-22:
> And you all know it... no critical bugreports in the next 24h... please ;)

Challenge accepted. :)

fly74, 2024-08-23:
> And you all know it... no critical bugreports in the next 24h... please ;)

I'll try my best but cannot promise. 😎

chemmex, 2024-08-30:
@TD-er don't you have plans to switch to pioarduino?

uzi18, 2024-08-30:
News: https://www.cnx-software.com/2024/08/27/platform-espressif32-fork-platformio-arduino-esp32-c6-esp32-c5-esp32-h2-esp32-p4/

TD-er, 2024-08-30:
> @TD-er don't you have plans to switch to pioarduino?

LOL....

OK let me explain it.... I am one of the few people behind pioarduino :)

I have been working with Jason2866 for way over a year on getting the platform packages working and ready for ESP-IDF 5.x and Arduino 3.0.x
This because there was no official platformIO support for it. 
It was then already clear to us that there was something wrong and that there might be a real possibility we would never see PlatformIO support for those newer SDK versions unless we do it ourselves.

At the end of '23 it became obvious others would be left in the dark too, so Jason did all the hard work in finding ways to make the platform packages less specific tailored to just what ESPEasy and Tasmota is using and more "vanilla Arduino"
This has taken a lot of work (glad some invaluable help from others was offered behind the scenes) and as you may have seen in the ESPEasy changelogs I have been running a lot of builds of those platform packages in the past 18 months.

Now that Arduino 3.0.0 is released and most of the libraries have fixes present to make them work with Arduino 3.0.x and ESP-IDF5.x pioarduino is made publicly available.

N.B. the pioarduino platform and the platform builds Jason and I have been using/working on do mainly differ in the amount of libs stripped from it to make the builds smaller.

This is not yet present in pioarduino but work is in progress behind the scenes.

Also this one will be merged soon: https://github.com/letscontrolit/ESPEasy/pull/5111
It will move all ESP32-variation builds to ESP-IDF5.1 for SPIFFS and ESP-IDF5.3 for LittleFS builds.

So I hope the "don't you have plans to switch to pioarduino?" question may be answered with this? ;)

chemmex, 2024-08-30:
Yes, I'm fully aware of the story, have been watching it for months. The question only was if you are still using Jason2866's Tasmota packages or there will be a shift towards pioarduino as a complete platform

TD-er, 2024-08-30:
Right now we're still using the platform builds Jason makes as those result in quite a lot smaller builds for us.
And he even makes a build specific for ESPEasy with SPIFFS.
The first 'plain' Arduino 3.0.x platform builds (like pioarduino 0.0.1-pre-alpha style) I made of ESPEasy on my PC (about a year ago) were about 1 MB larger than the ones with Jason's builds.
So there already has been a lot of progress since.

When it is made possible to also make smaller platform builds without a lot of hassle, I will for sure switch to vanilla pioarduino :)
Right now there is not a lot of differences anymore between those pioarduino builds and the one Jason makes for us.
Even some specific code changes and patches have been removed 2 days ago.
It mainly differs in SDK config and support for JL1101 Ethernet which has not yet been merged by Espressif, however those changes still require a full platform build, properly setup build environment, etc.

chemmex, 2024-08-30:
Thanks for the explanation, it becomes clear, finally.
Today, at last, I found the time to test recent release, and when I tried to custom build  my config for ESP32-C3, after pulling latest code, I run into the same problem we discussed earlier [here](https://www.letscontrolit.com/forum/viewtopic.php?t=10242),  so I created a new env with vanilla pioarduino as a platform, and was able to finish the build.
Thanks again guys for your efforts and for keeping us from falling back into the stone age.

TD-er, 2024-08-30:
For Windows you may still run into issues, until you change it to use this PR: https://github.com/letscontrolit/ESPEasy/pull/5111
Or at the very least change the folders as they are in this PR in platformio.ini.
Also make sure you don't use too deep paths.