letscontrolit/ESPEasy #5117: ESP32-C3 problem with connecting to WiFi

tomatos-st, 2024-09-01:
I have an unusual problem with ESP32-C3. I bought 2 batches of modules, 1 with a red antenna the other with a black antenna.

[image], [image] 

The first module does not cause any problems,
while the second one... is programmed ok (https://td-er.nl/ESPEasy/latest/) 20240828 Normal 4M LittleFS ETH [ESP32,ESP32-S2/C3/C6]
I have problems with connecting to the WiFi network.
The WiFi network is available, the signal strength is ok (the router is 1m away), but it cannot establish a connection.
Once in a while it connects (100:1), then I can go to the configure page and everything is supposedly ok, until I turn off the power, then the problem appears again
Interestingly, I cannot even connect to the AP: it does not accept the password.

I've tested 10 modules already, they all behave the same. I changed the antenna, the range improves (the black one is very weak) and I don't know what to do anymore.

[esp-web-tools-logs.txt](https://github.com/user-attachments/files/16828646/esp-web-tools-logs.txt)

 LOG:

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x403817b8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x98
load:0x403cc710,len:0x86c
load:0x403ce710,len:0x212c
entry 0x403cc710
00.163 : (265136) Info   : INIT : Free RAM:265136
00.174 : (264976) Info   : INIT : Soft Reboot #1 Last Action before Reboot: Intended Reboot: ResetFactory Last systime: 13 - Restart Reason: CPU0: Software reset CPU
00.186 : (264944) Info   : FS   : Mounting...
00.206 : (262960) Info   : FS   : LittleFS mount successful, used 151552 bytes of 323584
00.227 : (262748) Info   : CRC  : Settings CRC...OK
00.241 : (262580) Info   : ESPEasy console using ESPEasySerial
00.262 : (262516) Info   : ESPEasy console fallback enabled
00.273 : (262400) Info   : CRC  : SecuritySettings CRC...OK 
00.284 : (262280) Info   : NVS  : Save FacDefPref
00.286 : (262232) Info   : NVS  : Save UnitName
00.287 : (262184) Info   : NVS  : Save WIFI_Flags
00.288 : (262136) Info   : NVS  : Save WIFI_SSID1
00.290 : (262088) Info   : NVS  : Save WIFI_PASS1
00.301 : (262040) Info   : NVS  : Save WIFI_AP_PASS
00.303 : (261992) Info   : NVS  : Save WiFI_IP
00.305 : (261944) Info   : NVS  : Save ETH_IP
00.317 : (261896) Info   : NVS  : Save ETH_HW_CONF
00.318 : (261864) Info   : NVS  : Save Log
00.319 : (261816) Info   : NVS  : Save Console
00.330 : (261792) Info   : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
00.332 : (261744) Info   : INIT : Check for Priority tasks
00.333 : (261696) Info   : INIT : SPI not enabled
00.344 : (261328) Info   : Time : Set Ext. Time Source: No time set time: 13.000 offset: 00:00:12.655
00.447 : (261280) Info   : WIFI : Set WiFi to STA
00.595 : (222540) Info   : WiFi : Start network scan all channels
01.442 : (222340) Info   : WiFi : Scan finished, found: 1
01.444 : (222272) Info   : WiFi : Start network scan all channels
02.290 : (222024) Info   : WiFi : Scan finished, found: 2
02.292 : (222076) Info   : Setup: Scan all channels
02.308 : (222012) Info   : ESPEasy console using ESPEasySerial
02.330 : (221932) Info   : ESPEasy console fallback enabled
02.331 : (221884) Info   : INIT : Free RAM:221932
02.345 : (221820) Info   : ESPEasy console using ESPEasySerial
02.366 : (221760) Info   : ESPEasy console fallback enabled
02.367 : (221648) Info   : INFO : Plugins: 55 ["Normal","NeoPixel"] (ESP32 SDK 5.3.0.240821)
02.369 : (221700) Info   : WIFI : Set WiFi to OFF
02.588 : (247464) Info   : WIFI : Set WiFi to STA
02.701 : (221432) Error  : WIFI : No valid wifi settings
02.703 : (221368) Info   : WiFi : WiFiConnected(), start AP
02.704 : (221320) Info   : WIFI : Set WiFi to OFF
02.923 : (247248) Info   : WIFI : Set WiFi to AP
04.002 : (217372) Info   : WIFI : AP Mode enabled. SSID: NeoPixel IP: 192.168.4.1 ch: 1
04.004 : (217260) Info   : WIFI : Arduino wifi status: WL_STOPPED 254 ESPeasy internal wifi status: DISCONNECTED
[... 84 lines cut ...]
00.301 : (262208) Info   : INIT : SPI not enabled
00.302 : (261848) Info   : Time : Set Ext. Time Source: No time set time: 181.000 offset: 00:03:00.697
00.405 : (261700) Info   : WIFI : Set WiFi to STA
00.552 : (222952) Info   : WiFi : Start network scan all channels
01.398 : (222664) Info   : WiFi : Scan finished, found: 2
01.400 : (222296) Info   : WiFi : Best AP candidate: Hardware-X 18:D6:C7:CE:2A:DA Ch:7 (-69dBm) WPA/WPA2/PSK (bgn)
01.413 : (222356) Info   : WiFi : Added known candidate, try to connect
01.414 : (222308) Info   : WIFI : Set WiFi to OFF
01.623 : (248084) Info   : WIFI : Set WiFi to STA
01.736 : (221772) Info   : WIFI : Connecting Hardware-X 18:D6:C7:CE:2A:DA Ch:7 (-69dBm) WPA/WPA2/PSK (bgn) attempt #0
02.838 : (221644) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
02.840 : (221444) Info   : WiFi : Best AP candidate: Hardware-X 18:D6:C7:CE:2A:DA Ch:7 (-69dBm) WPA/WPA2/PSK (bgn)
02.864 : (221508) Info   : ESPEasy console using ESPEasySerial
02.885 : (221424) Info   : ESPEasy console fallback enabled
02.886 : (221376) Info   : INIT : Free RAM:221424
02.901 : (221304) Info   : ESPEasy console using ESPEasySerial
02.923 : (221240) Info   : ESPEasy console fallback enabled
02.924 : (221128) Info   : INFO : Plugins: 55 ["Normal","NeoPixel"] (ESP32 SDK 5.3.0.240821)
02.928 : (215828) Info   : Webserver: start
03.745 : (238172) Info   : WIFI : Disconnected! Reason: '(2) Auth expire' Connected for 02.002
03.846 : (241452) Info   : WIFI : Arduino wifi status: WL_CONNECTION_LOST 5 ESPeasy internal wifi status: DISCONNECTED
03.847 : (241148) Info   : WiFi : Best AP candidate: Hardware-X 18:D6:C7:CE:2A:DA Ch:7 (-69dBm) WPA/WPA2/PSK (bgn)
03.859 : (241132) Info   : WIFI : Connecting Hardware-X 18:D6:C7:CE:2A:DA Ch:7 (-69dBm) WPA/WPA2/PSK (bgn) attempt #1
04.887 : (214860) Info   : WIFI : Arduino wifi status: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED
04.992 : (214920) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 215064 WiFiStatus: WL_DISCONNECTED 6 ESPeasy internal wifi status: DISCONNECTED

tonhuisman, 2024-09-01:
Can you try with the regular 20240822 release, availablr from https://td-er.nl/ESPEasy?  The 'Latest' uses a newer beta of IDF 5.x and Arduino 3.x, that may be a possible cause for this issue.

TD-er, 2024-09-01:
Or also test a "SPIFFS" build on the /latest url?
The SPIFFS builds there use ESP-IDF5.1 and the LittleFS builds use ESP-IDF5.3

tomatos-st, 2024-09-01:
Loading the **4M Flash SPIFFS** version from https://td-er.nl/ESPEasy/latest/ gave little effect.
only when trying to install:
[image]
a message appeared
[image]

with other versions (Energy, Climate) it was ok.
I achieved that out of 6 modules 2 connected to the network but the stability of this connection raises great doubts.
Sometimes the page loads quickly, and sometimes it takes 1-3 minutes and does not always end successfully.

I have already ordered modules with a red antenna, I will have to wait a bit until they arrive ...:(

tonhuisman, 2024-09-01:
The failing board(s) could have some electrical issues, the image is a bit blurry, but I wonder if there's a tiny drop of solder on the ESP pins, and it's also possible the antenna isn't connected to the ESP, as there might be a 0-ohm resistor missing. You can use a multimeter to validate if they are connected, and if either side is connected to the antenna:

[image]

tomatos-st, 2024-09-01:
The shorted fields are 2 and 3 (VDD3P3) and 31 and 32 (VDDA)

[image]

in both systems they are connected (including the black antenna less elegantly ;)
Pin 1 (LNA_IN) with the antenna and in both cases connects UR15, and the missing elements are probably capacitors UC8 and UC9.

From the measurements on both boards UR15 is probably not a resistor but rather a capacitor.

[image]
[image]

TD-er, 2024-09-01:
Can you also test flashing a "SPIFFS" build using the [latest webflasher](https://td-er.nl/ESPEasy/latest/) ?
It looks like the ESP-IDF5.3 builds (the LittleFS builds on that link) behave differently from the ESP-IDF 5.1 builds (the "SPIFFS" ones, lower in the list of options)

TD-er, 2024-09-02:
OK, I have been testing quite a bit using ESP32-C3 and C6.
Using clone boards and the orginal Espressif dev boards.
I can reproduce this and Ton also can reproduce it when using ESP-IDF5.3.
As soon as I make a build using IDF5.1 there is no problem anymore.

I also have had some talks with someone from Espressif and @Jason2866 from Tasmota.
Both have tested (Jason even tried installing the same build from the web flasher) and could not reproduce this behavior.

So my guess is it might have something to do with some combination of access points.

My current theory (which has neither been proven nor disproven) is that the ESP now stops scanning on a channel as soon as some AP replied.
On ESP-IDF 5.3 builds I never see >2 AP's found on the same channel even though there should be several on the same channel.

tomatos-st, 2024-09-02:
Ok, I've done a lot of tests.:
A "SPIFFS" build using the [latest webflashe](https://td-er.nl/ESPEasy/latest/): result 50:50 on 6 systems 3 connect to the network without a problem.

Uploading from the LittleFS version with the [latest webflashe](https://td-er.nl/ESPEasy/latest/) causes none of them to connect to the network.

Uploading the "SPIFFS" build using the latest webflasher version and then changing to LittleFS from the Update level is successful. :) the system works without a problem.

Another test: I resoldered the processor from the "black" board to the "red" one,
I chose one of those that could not connect and it turned out that it works without problems on the "red" one, both in the SPIFFS and LittleFS versions

The samples that do not connect, after programming **flash_download_tool_3.9.6** also cannot connect

Conclusions: There is a design error in the "black" version that hinders WiFi communication, and the SPIFFS version is more resistant to communication errors

I just wonder why programming via the LittleFS website does not work, but with the update it does...

I will give up on the "black" version. I found a new PRO version with a larger antenna and an antenna socket
[image]

Jason2866, 2024-09-02:
@TD-er imho this is a different issue, not related to an framework. I have such a bad C3 mini too. It is just garbage. Does not work with every framework. If lucky a connection is established. A other C3 mini (which has a red antenna) just works.

tomatos-st, 2024-09-02:
It's good that 1 piece works ok after transplantation.
My last "red" fried after a STEP-DOWN converter failure. I'm waiting for a new delivery.
We'll see how it goes with the PRO version.

https://www.aliexpress.com/item/1005007329486741.html?spm=a2g0o.order_list.order_list_main.5.f4831c24CMl2TE&gatewayAdapt=glo2pol

TD-er, 2024-09-02:
> @TD-er imho this is a different issue, not related to an framework. I have such a bad C3 mini too. It is just garbage. Does not work with every framework. If lucky a connection is established. A other C3 mini (which has a red antenna) just works.

Well the WiFi issue is indeed something completely different from the does flash/doesn't flash issue.
But for the WiFi issue see your personal Discord chat ;)

tomatos-st, 2024-09-02:
Any suggestions on where I should look? I rarely check Discord, I generally don't keep up with the time lately and I work on my projects less and less often.

They say it's old age. :) I feel less and less like doing things.

TD-er, 2024-09-02:
The suggestion for now for you is to use the builds which are not on /latest but on https://td-er.nl/ESPEasy/
Those do not use IDF 5.3 but rather IDF 5.1 which seems not to have these issues.

tomatos-st, 2024-09-03:
So far, there have been no issues with any build version. But I've run out of the "red" ones, and as you can see, the "black" ones are defective.
My joy with the working ESP soldered to the red board was too early, after 10-15 minutes the device starts to freeze and have problems with refreshing the website.

TD-er, 2024-09-03:
Found a work-around for the IDF5.3 WiFi issue.
When I use async mode scanning I do see all APs.
However this may require some extra code.

In the mean time I will upload a new build using only IDF5.1 for both SPIFFS and LittleFS builds. to the web flasher.

closed 2024-09-15 by tomatos-st