letscontrolit/ESPEasy #5118: Fixes for upcoming release (okt 2024)

TD-er, 2024-09-03:
There is a number of issues which for sure need to be fixed for the next release (probably Oktober 2024)

- [ ] Time Sync for devices using deep-sleep [See Forum](https://www.letscontrolit.com/forum/viewtopic.php?t=10405)
- [ ] Connecting to hidden SSID (ESP8266) [See forum](https://www.letscontrolit.com/forum/viewtopic.php?t=10434)
- [ ] Use 'async scan' for ESP32 to scan/connect to WiFi. (#5117)
- [ ] AP not started when no known AP is found (many reports, among which #5037)
- [ ] WiFi not reconnecting (many issues, among which #4976 )
- [ ] Switch to ESP-IDF5.1 (maybe 5.3 ?)

Those will be the main issues for me (TD-er) to focus on for the next release.

chemmex, 2024-09-09:
Ready to test the AP not started problem

TD-er, 2024-09-09:
> Ready to test the AP not started problem

OK, but it has not yet been fixed though... ;(

TD-er, 2024-11-05:
Just as a short update...
I am working hard on a complete rewrite of all network/WiFi related code which was years overdue.

Making quite a lot of progress and it is nice to see quite a lot of code can be thrown away, making it all much easier to read/understand and hopefully also resulting in smaller bin files and better stability.