letscontrolit/ESPEasy #5119: [BUG] Time drift with P2P enabled

DittelHome, 2024-09-04:
With  P2P enabled, I get time drift. (at least 15 seconds)
NTP only works as expected.
Only disabling P2P fixes the problem.

Regards Klaus

TD-er, 2024-09-04:
Thanks for the issue. I had already included it in the todo for the next release: https://github.com/letscontrolit/ESPEasy/issues/5118
But it is good to also have an issue where others may also provide input and/or see what's going on.

DittelHome, 2024-09-04:
Hi, I know about the deepsleep problem and the time drift, I wrote the thread in the forum.
It seems to be a general problem with p2p.
As soon as p2p is activated, the time becomes inaccurate!!
Of course p2p were  already deactivated on Deepsleep devices...

DittelHome, 2024-10-06:
Maybe it helps ..
Today I discovered that an ESP gets the time from another via p2p and vice versa, even though NTP is activated on both. Both have an uptime of over thirty days.
As a result, both are 1 minute and 40 seconds behind, and the trend is increasing

TD-er, 2025-04-24:
I just merged a commit which should fix this issue.
I'm running it locally on my nodes and before my fix some nodes which were set to use NTP themselves got a time sync from another p2p node.
This is obviously wrong.

So hopefully tomorrow morning those  2 nodes will still only be synced via NTP and others only sync to these 2 nodes.

DittelHome, 2025-04-25:
It sounds good ..