letscontrolit/ESPEasy #5123: Plugin P025 (ADS1115) problem with decimal places in formula calculation.

tomatos-st, 2024-09-15:
I measure voltage using ADS1115. Due to the fact that the voltage range is above 30V I have used a resistive voltage divider 1:10

I select the **Convert to Volt** option:
Therefore, the voltage of 12.7V measured by ADS1115 is as 1.27V.

I want ESP to display the original voltage of 12.6V so I multiply:
`%value%*10`

[image]

and I should get 12.7V with **Decimals** set to 1
unfortunately I get 13.0V

the images show the version with multiplied * 10 and without the formula:
[image]

It looks like the **Decimals** parameter is used for the value of the input parameter _%value%_, not for the result of the entire formula.
So
1.27 -> 1.3
1.3*10=13.0V

it looks even worse after multiplying *100

ESP type: ESP32-C3
Build version: ESP_Easy_mega_20240902_normal_ESP32c3_4M316k_CDC

TD-er, 2024-09-15:
Yep, seems like the decimals are applied before the formula is applied.

Will have a look at it.

TD-er, 2024-09-15:
I added a quick fix, can you test it when this GH Actions build is ready?
https://github.com/letscontrolit/ESPEasy/actions/runs/10872194895

tomatos-st, 2024-09-15:
Almost ok :)
only...
for the first 1-2 seconds it displays the value with a precision of 5 decimal places, .
[image]
then it's ok as selected in the Decimals option
[image]

but mqtt to domoticz constantly sends the value with a precision of 4 decimal places
[image]

TD-er, 2024-09-15:
OK, so maybe I should spend more than half an hour on it ;)

TD-er, 2024-09-15:
Should be fixed with this build: https://github.com/letscontrolit/ESPEasy/actions/runs/10874920584  (when ready)

tomatos-st, 2024-09-16:
It seems to be OK now.
:)

closed 2024-09-16 by tomatos-st

tomatos-st, 2024-09-18:
hey,
Today I noticed that I can't give a name to any new device.

When creating a device it deletes the name and throws me the message "Warning: Task Device Name is empty. It is advised to give tasks a unique name"
[image]

TD-er, 2024-09-18:
Which build is this?

tomatos-st, 2024-09-19:
Update to the version you provided in the link: 
https://github.com/letscontrolit/ESPEasy/actions/runs/10874920584
after fixing with decimal places in the formula

TD-er, 2024-09-19:
ESP32 or ESP8266?
LittleFS or SPIFFS?

tomatos-st, 2024-09-19:
still the same ESP32-C3 SPIFFS normal

TD-er, 2024-09-19:
Can you try with this build I made last night?
https://github.com/letscontrolit/ESPEasy/actions/runs/10930088141

This does contain some fixes I submitted to the Arduino code last week regarding flushing of stream objects, some issues in Webserver and some other minor fixes.

I just tested it on an ESP32-classic running this build and there I can just save the settings (tested on LittleFS builds, will now also flash a SPIFFS build to test)

tomatos-st, 2024-09-19:
Still the same..."Warning: Task Device Name is empty. It is advised to give tasks a unique name"
I went back to version 20240822 for testing, and it works fine.

Interestingly, after this message, a working task is created which uses an empty name.
[image]

Another thing, is it just me who has problems saving Rules? It's like the Save button doesn't work, it lights up when you hover over it but doesn't save, sometimes after a few attempts it saves. Tested on several browsers)

TD-er, 2024-09-19:
Hmm this sounds like your device has issues writing to the flash.
Can you maken a TAR export and then do a factory reset?
This will then format the file system.
You can then re-upload the TAR.

TD-er, 2024-09-19:
How full is your file system?

tomatos-st, 2024-09-19:
I exported the settings
I did a factory reset: creating tasks works fine
I imported the settings: the problem returned

Now I will reset again and manually configure it again

is this it:
Free RAM:	188036 [byte] (174344 - )
Free Stack:	7400 [byte] (4788 - ) 
[image]

tonhuisman, 2024-09-19:
2 options worth investigating:
- Have you deleted the task that doesn't want to store the taskname, and recreated it?
- You could try the LittleFS build for the C3 to see if that works, now that you have a .tar backup, it's easy to switch, and LittleFS is the file system of the future. (SPIFFS has been deprecated for years)

tomatos-st, 2024-09-19:
bingo :)

I started manually configuring from the beginning and the problem occurred again....after enabling **Rules** ... :)

[image]

if I disable Rules - the problem disappears

tomatos-st, 2024-09-19:
- yes I deleted tasks that had errors,
- ok, I also tried switching to LittleFS and there is the same problem

I am working on SPIFFS after previous problems with WiFI that occurred on LittleFS and ESP32-C3

tonhuisman, 2024-09-19:
A few more things to check:
- What cable/wire is being used for powering the ESP? Poor power supply can cause really strange issues like these
- It's possible the ESP/flash is broken or worn out in some way. Do you have another ESP32 (any variant will do) to test?

tomatos-st, 2024-09-19:
Currently, I am testing on 2 different ESP32s, one is inserted into the working system, powered by 5V via a pin, 

[image]

the other is connected on the desk to a 5V 2A power supply from Rassbery Pi4
[image]

I remind you that in version 20240822 everything works OK

I can still unpack the new third ESP and test

TD-er, 2024-09-19:
OK, it looks like the ESP somehow can't save the settings.
I know that SPIFFS has some serious issues with fragmentation if the filesystem is above some percentage and frequently has to save things.
So that's why I asked about how full your filesystem is (see sysinfo page)
What you showed is the partition layout chart.

What do you have stored on the file system?
Can you try restoring the TAR file on a build with LittleFS?
The build with SPIFFS is using an older version of ESP-IDF.

tomatos-st, 2024-09-19:
Ok test of the third, new esp without any configuration.

test: enabling Rules and attempting to create a task (Switch input - Switch/ GPIO2)

installation from 20240822 -
Binary Filename: ESP_Easy_mega_20240822_normal_ESP32c3_4M316k_LittleFS_CDC_ETH: works without a problem
Installation from https://github.com/letscontrolit/ESPEasy/actions/runs/10930088141
Binary Filename: ESP_Easy_mega_20240918_normal_ESP32c3_4M316k_LittleFS_CDC_ETH: problem exists

TD-er, 2024-09-19:
OK, so there is something wrong here...
Will look into this.

TD-er, 2024-09-19:
I think I may have found the issue...
https://github.com/espressif/arduino-esp32/commit/0ab2c58b6c14f6dbc8b9ab0e61d776cd3ac5de66

And since we're using this `postion()` function when parsing the rules, it for sure matches exactly your findings
It also explains why it is failing both the SPIFFS and LittleFS builds as the helper class to index the rules is creating a data structure based on position in the file where those rules blocks are located to speed up rules parsing.

tomatos-st, 2024-09-19:
I'm glad you managed to figure something out.

I may not be the best programmer, but lately I've been coming across and helping to find a lot of bugs in various projects I work with.

TD-er, 2024-09-19:
> I may not be the best programmer, but lately I've been coming across and helping to find a lot of bugs in various projects I work with.

That's also really important.

tonhuisman, 2024-09-19:
> helping to find a lot of bugs in various projects I work with.

That's very helpful, real world usage is a great way to test stuff, although it can be quite frustrating if something doesn't work as intended.
Thanks for helping us find these issues 👍

tomatos-st, 2024-09-19:
You're welcome.

I'll remind myself if https://github.com/letscontrolit/ESPEasy/issues/4969 there's a chance that this will be created ? Tasmota also already supports BLE sensors.

TD-er, 2024-09-19:
Right now there are just too many open issues which need to be dealt with.
When that's finished, I can have a look at BLE.
Also there is a lot of activity going on lately in the BLE code for ESP-IDF5.3 and Arduino, so that's another reason to let it rest for now.

uzi18, 2024-09-19:
@TD-er position() method is also used on available() and peek()

TD-er, 2024-09-20:
> @TD-er position() method is also used on available() and peek()

Thanks, I was wondering why it wasn't fixed with my local changes

TD-er, 2024-09-25:
OK, took quite some effort to get this bug and the rules editing one fixed using IDF5.3
I uploaded the latest build here: https://td-er.nl/ESPEasy/latest/

tomatos-st, 2024-09-26:
I tested it, it looks OK