letscontrolit/ESPEasy #5128: Temperature with Gases - ENS16x TVOC/eCO2 and SCD30

betux69, 2024-09-24:
Hello,

First of all, a big thank you for your work !

I made some test with ens16X and scd30 for CO2 and température.
I use climate build.
Neither of them displays the right temperature, it remains stuck at 26/27 degrees.
With an instantaneous precision thermometer I am 21.
I'm wondering if I shouldn't change the i2c settings. Apart from that I don't see ;o)

Thanks

tonhuisman, 2024-09-24:
Hm, last time I tested the SCD30 it worked as expected, but that was some time ago. The plugin hasn't been touched for a while, AFAICS, so no obvious reason for it being stuck. ENS16x I don't have available, so can't check that.

What Interval do you have set for these sensors?

TD-er, 2024-09-24:
How/where are the sensors mounted?
Also the internal temperature sensors of some devices which primary goal is not measuring temperature is often only used internally to compensate the sensor readings.
For example the SenseAir CO2 sensors do also have an internal temperature sensor which can be read.
However this is the internal chip temperature and used in the SenseAir firmware to compensate its readings.

Also if you have your sensor mounted inside some enclosure along with other heat generating parts like a voltage regulator, ESP, etc. then it is normal the temperature will have a somewhat constant offset.

N.B. these kind of questions are not about bugs or change/feature requests, so better asked on the forum.

chromoxdor, 2024-12-22:
> Neither of them displays the right temperature, it remains stuck at 26/27 degrees.

This seems to be normal at least for the ENS16x boards with a temperature sensor due to nature of the sensor (it uses a heating element) and the proximity of the temperature sensor. Therefor i added a temperature offset field for the ATH2x plugin. It works great since the temperature deviation is stable...