letscontrolit/ESPEasy #5132: Update error: ERROR[4]: Not Enough Space

fly74, 2024-10-02:
I've an old one with:

<html>
<body>

Firmware
--
Build:⋄ | 20112 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ |  
Plugin Count:⋄ | 3 [Normal] [Minimal 1M OTA]
Build Origin: | Travis
Build Time:⋄ | Feb 23 2021 12:23:25
Binary Filename:⋄ | ESP_Easy_mega_20210223_minimal_core_274_ESP8266_1M_OTA_FHEM_HA
Build Platform:⋄ | Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄ | mega-20210223_9054eb3
System Status
Syslog Log Level: | None
Serial Log Level: | None
Web Log Level: | None
ESP Board
ESP Chip ID: | 15839117 (0xF1AF8D)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | PLATFORMIO_ESP01_1M
Storage
Flash Chip ID: | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 599 kB (276 kB free)
Max. OTA Sketch Size: | 599 kB (614384 bytes)
OTA possible: | true
OTA 2-step Needed: | true
SPIFFS Size: | 110 kB (27 kB free)

</body>
</html>

and want to flash with:

614.384 Byte ESP_Easy_mega_20240822_minimal_core_274_ESP8266_1M_OTA_Domoticz_MQTT.bin
or
612.624 Byte ESP_Easy_mega_20240822_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT.bin

these 2 are below the OTA limit above, but I'm getting the error in the title.

TD-er, 2024-10-02:
I assume you're flashing those using the 2-step OTA?

fly74, 2024-10-02:
Sorry long ago. Have to use 2 Step.

closed 2024-10-02 by fly74

TD-er, 2024-10-02:
If you really have an ESP-01, then you can also get a form-factor compatible ESP32-C3 version.
Or you could simply replace the flash chip of the module for a 4M version.

fly74, 2024-10-02:
2 step worked. It's an embedded ESP in a GOSUND plug. No replace possible.