letscontrolit/ESPEasy #5136: [FR] Notification Email plugin that supports TLS and STARTTLS protocol

tonhuisman, 2024-10-06:
We already have N001 Email notifications plugin, and with #5135 we get support for sending email via an SSL encrypted connection.

Feature requests:
- [ ] Add email notification support with the more modern TLS encryption protocol and STARTTLS (upgrade unencrypted connection to encrypted connection) support.
- [ ] Send email notifications with attachments, like the data from the Cache controller, to recipient(s).

The features to support all this are available in [this library](https://github.com/mobizt/ESP-Mail-Client), but as it adds quite substantial to the .bin size, it can not be included in all ESP32 builds, like the SSL encryption for N001, so it should be made available as a separate Notification plugin for user-defined Custom builds (and in MAX builds).

TD-er, 2024-10-06:
Other ideas, some of which are also implemented in devices like FritzBox modems:

- Before upgrade email the settings as a ZIP file.
- Allow to send some summary of collected data periodically (e.g. daily)
- Allow to send some logs when something strange is happening (e.g. login from a different IP)
- Send email with IP-settings when these change

These can benefit from the attachment feature, like sending some JSON with task data when the 'stats' are enabled.
And maybe even be able to generate some graphics, or send out captured pictures when we support camera sensors.

TD-er, 2024-10-06:
Some info about the differences between STARTTLS / SSL / TLS:  https://mailtrap.io/blog/starttls-ssl-tls/