letscontrolit/ESPEasy #5138: [FR] Certificate based connecting to MQTT broker using TLS on ESP32

OmarTheEngineer, 2024-10-09:
I am trying to set up a secure MQTT connection with different brokers that require certificates like HiveMQ however this is not available in the current ESP-Easy build which only supports username and password authentication.

Since I'm working on the project that will be in an open network it is a good idea to use MQTTS to encrypt the messages I am sending while using ESP-Easy

TD-er, 2024-10-09:
Just to be sure we're talking about the same....
You need to have support for user certificates purely for authentication, right?
The SSL/TLS connectivity itself has (very) recently been added.

Another still open issue is to validate the server certificate, to prevent man-in-the-middle attacks.

OmarTheEngineer, 2024-10-09:
What I'm trying to do is basically what I would do in script using the WifiClientSecure Library, For connecting with HiveMQ I would assign the Root CA server certificate in the esp32 and that would work.

TD-er, 2024-10-09:
But a root CA server certificate is only needed to validate the server certificate.

I thought you needed to have support for not (just) using username/password but also just allow to have some client certificate and/or client key.

The current implementation as can be found in the latest builds ([here](https://td-er.nl/ESPEasy/latest/)) does not check the server certificate and just allows all presented certificates by the server (even self-signed ones).

Have you checked if that allows you to connect to the MQTT broker of your choice?

OmarTheEngineer, 2024-10-09:
Hey @TD-er , Sorry for the late reply.
I was trying to connect it to HiveMQ, but since i was using port 8883 I had to set the Root CA server certificate to able to authenticate and connect. with one of the builds i tried to connect using only username and password however that did not connect with HiveMQ hence why I was asking where I would define such Root CA server certificate in the code using platformio

TD-er, 2024-10-10:
What error do you get for the webflasher page?
I do have some filter active, since I do get a lot (!!) of hack-attempts on my server.
Incomming traffic which gets blocked is between 2 and 3 TByte a month.

So if you try again and then tell me the timestamp, I can find it in the logs and remove your IP or IP-subnet from the block-list.
Or you can download this latest GH-Actions build: https://github.com/letscontrolit/ESPEasy/actions/runs/11225036186

OmarTheEngineer, 2024-10-10:
Link works, thank.
I will be testing MQTT on one of the builds however I don't think it will work since the the certificate is not defined

OmarTheEngineer, 2024-10-10:
Excuse my technical knowledge here, I'm not sure how the connection gets authenticated by HiveMQ, but what I know is that I download the Root CA Server certificate and I would plug that into my script using the WifiClientSecure
If you could enlighten me more about these facts It would also be appreicated.
I hope there is a way to get this working as ESP-Easy is a really great platform!

TD-er, 2024-10-10:
Right now, I do not actually use root CA certificates.
Those are typically used to validate a server certificate.

Simply put, if you create a certificate for a domain, you typically generate a public/private key pair, create a certificate and sign it yourself with your private key.
Then you hand the self-signed certificate over to a CA which does append some info about their public key and sign it with their private key.
To make sure you can trust that CA (e.g. Let's Encrypt, GoDaddy, etc) you can see in the extra info the CA added that their public key was signed by a root CA.

So to verify a full certificate chain, you have to do the reverse using the public key of the root CA and then verify that each of those certificates in your chain are signed by the keys they claim to have used.

Since I have not yet added all these checks right now, there is no validation of those certificates and it is just assumed they are all who they claim they are and we just use accept the certificate and continue negotiating the connection.

I do have 2 root CA's included in the code, among which is Let's encrypt and Google. (or maybe 3, have to check)
However as far as I know those are not yet used as I have explicitly set the BearSSL client to 'insecure' which should just ignore all these checks.

Right now I don't know why your own test adding a root CA certificate makes it work, as it should only make sense to validate the server certificate.
It is not like you suddenly can generate your own certificate which is signed by some root CA in your Arduino code.

So as I understand it, you probably are also using some client certificate to login to HiveMQ, but this has not much to do with a root CA....

TD-er, 2024-10-10:
Maybe you should also set your controller timeout to the highest allowed value?
The default client timeout in Arduino code is longer than the default timeout used in ESPEasy.