letscontrolit/ESPEasy #5145: [BUG] Plugin 102 (PZEM) Energy reset will reset energy one time and at each board reboot

iz8mbw, 2024-10-22:
Hello.
I noticed that if I reset the Energy of the PZEM:
[image]

it works OK since it reset the Energy value.
But if I reboot the board it will go reset the Energy value, if reboot again it will reset the Energy value again and so on.

To avoid it, after the first Energy reset, I must click on "Submit" to _force_ again the "PZEM Mode" to Read_value.
[image]

In other words, when you reset the Energy value the Energy value is resetted and the web page ruterned (after the reset) is automatically changed to "Read_value." but if you reboot the board the Energy value is resetted again.

TD-er, 2024-10-22:
Isn't there a command to reset the value?

To be honest, I do think this is a design oversight as it makes no sense to reset it each time the plugin is started.

iz8mbw, 2024-10-22:
> Isn't there a command to reset the value?
> 
Yes, you can reset via editing the Device (so from the Web page) or you can reset via HTTP GET to http://*espeasyip*/control?cmd=resetenergy,*PZEM address* 
> To be honest, I do think this is a design oversight as it makes no sense to reset it each time the plugin is started.
>
Indeed, this is the bug.

tonhuisman, 2024-10-22:
Feels like a bug, looking at the code, as during `PLUGIN_INIT` the mode is reset to Reading mode after executing the Reset, but it's never saved...
But there are other things to observe(fix?) there, as the 'multi-instance' implementation isn't very multi-instance, AFAICS.

TD-er, 2024-10-22:
> But there are other things to observe(fix?) there, as the 'multi-instance' implementation isn't very multi-instance, AFAICS.

That's because it is planned for a huge overhaul when I implement the Modbus handler/manager to share all modbus calls from all modbus plugins

tonhuisman, 2025-09-14:
I've updated the code to show a message to Submit the page once more after a Reset Energy action, where the mode is restored to Read energy, but not yet saved. Also updated the documentation.

When reworking the Modbus implementation this plugin will need a big overhaul, so that seems the best time to remove the Reset Energy mode setting, and rely on the `resetenergy,<modbus_address>` command (as that's IMHO simpler and avoids extra/unneeded saving of the settings file).