letscontrolit/ESPEasy #5146: On system#boot gets disregarded

TommoT1, 2024-10-22:
I just loaded the latest action build (ESP_Easy_mega_20241019_normal_ESP8266_4M1M.bin.gz) to a node and noticed that all stuff in the section 
```
on system#boot do
 gpio,12,1
 another command
endon
```
seems to be disregarded....

TommoT1, 2024-10-22:
Not yet, I'm not at home, just flashed a wifiplug from Sonos via OTA and noticed that ist does'nt switch on anymore when pluged in like it did before. Switched back to 20240822 and normal ops again.

TD-er, 2024-10-22:
I just tested on a Sonoff POW r1.
Had build from August running on it, tried to make sure the system#boot event was working and then updated to `ESP_Easy_mega_20241007_hard_SONOFF_POW_4M1M`
It is still working.

Used rules:
```
On System#Boot Do
  if [relais1#state]=0
    gpio,12,1
  else
    gpio,12,0
  endif
  Let,1,123
Endon

on Button1#state do
  if %eventvalue1%=0
    gpio,12,1
  else
    gpio,12,0
  endif
endon
```

I also set a value to a variable, so I can be sure the event was triggered as can be seen on the system variables page.

TommoT1, 2024-10-23:
Good morning!
Sorry, I was wrong....
I have the following Rules:
```
on System#Boot do gpio,12,1
on Taster#Switch=1 do AsyncEvent,toggle

on toggle do
  if [Relay#State]=0
    gpio,12,1
  else
    gpio,12,0
  endif
endon
```

"Taster" is a switch which monitors GPIO 0,
with the latest action  build the  "on Taster#Switch=1 do AsyncEvent,toggle"
gets activated once, there seems to be a spike in GPIO 0 wich had no effect on older FW (20240822 does fine)
So I will add a timer to enable the operation via the switch some seconds after boot.

TD-er, 2024-10-23:
Please do not use those oneliners as they are always tricky.
Also there is no 'endon' so not sure if the rules cache will work well on those.

TommoT1, 2024-10-23:
OK, but also without oneliners it doesn't work anymore like before,
The device is a SONOFF S20 and the rules where like:
```
on System#Boot do 
 gpio,12,1
endon

on Taster#Switch=1 do
  AsyncEvent,toggle
endon

on toggle do
  if [Relay#State]=0
    gpio,12,1
  timerset,1,1
  else
    gpio,12,0
  endif
  TimerSet,1,1
endon
```
The deviced booted, switched on the relay and I was able to toggle the relay via a button wired to GPIO 0
With firmware version later than 20240822 it boots up,
the "on system#boot" section get processed (the relay switches on) and then there must be a spike (or what else?) and the "on Taster#Switch=1" gets processed too, so the relay is switched off again.

So I had to add a timer which inhibits the switch for a while:
```
on System#Boot do 
 gpio,12,1
 TimerSet,1,5
endon

on Taster#Switch=1 do
 if %v1%=1
  AsyncEvent,toggle
 endif
endon

on toggle do
  if [Relay#State]=0
    gpio,12,1
  timerset,1,1
  else
    gpio,12,0
  endif
  TimerSet,1,1
endon

on Rules#Timer=1 do 
 Let,1,1
endon
```
So the older firmwares where slower during boot (the spike in GPIO 0 had already passed)? Something has changed....
anyway, is there a more elegant / shorter way to have a pause after boot before the unit reacts to GPIO changes?

tonhuisman, 2024-10-23:
Do you have 'Send boot state' enabled on the `Taster` switch task?

TommoT1, 2024-10-23:
No

TD-er, 2024-10-23:
Do you have a debounce configured for GPIO-0 in the switch task?

TommoT1, 2024-10-23:
Yes, I tried this up to 100ms but no help, I also tried a "delay,10000" in the "system#boot" section but then the relay switched on, than they 10s freeze occurs and then the "in Taster#Switch=1" switched it off again

TD-er, 2024-10-23:
You should not use `delay` in rules as it indeed blocks everything.
better set a timer.

TommoT1, 2024-10-23:
I know that, I just tried it and was surprised that the "spike" or whatever it is, gets some how "recorded" althoug the whole unit was frozen

TD-er, 2024-10-23:
There may have been some optimizations in the code, which may cause things to load faster.
So maybe there always was such a spike, but not seen.

TommoT1, 2024-10-23:
Is there a more elegant way than setting a timer, and then setting a variable and then, each time the switch is pressed, also checking for the variable?

tonhuisman, 2024-10-23:
Maybe it's worth investigating what is causing the 'pulse' on the `Taster` GPIO pin?
TD-er couldn't reproduce the issue you're seeing, so there may be something specific to your setup, causing this.

TommoT1, 2024-10-24:
Yes maybe, but I'm afraid there is not much I can do about it, it is a SONOFF S20, the only thing I could change was the flash chip to 4MB. 
Next I will see If the SONOFF S26 also have this issue. In those units there was nothing to change because they use the ESP8285.

TD-er, 2024-10-24:
Can you try to open the logs via the web log viewer as soon as it boots, as some of the logs may then still be buffered from boot.
Or maybe try to keep track of how often the event is triggered by storing a count in a variable.
Maybe this can help narrow it down to see what may be happening here.
Connecting something to it to read the serial port is a bit tricky as there is likely no isolation from mains power, so you may fry your computer (or yourself) if you connect to the serial port while plugged into mains power.

TommoT1, 2024-10-24:
I think, one problem is that SONOFF Choose GPIO0 for the switch, this ist easy for flashing but as some other pins GPIO0 does some stuff during boot (this German page says GPIO makes some LOW Pulses during boot (http://stefanfrings.de/esp8266/)).
Is it possible that as espeasy gets quicker there might grow some issues?
I can open up the S20 later and write a log, if it also has those spikes when running on DC5V

tonhuisman, 2024-10-24:
> when running on DC5V

Better do the testing with 3.3V, or you may damage the S20... 👀

TommoT1, 2024-10-24:
On the sonoffs it is easy to go in front of the 3.3V voltage regulator and take a pin of the 6V ELKO.
then you have 3.3V more stable as via a dupont cable.

I now ran the device without the timer, like it did the years before.
at 07.095 it switches the relay on (system#boot)
at 08.173 the switch seems pressed and relay gets shut off

Here is my log:
```
21.832 : (16288) Info   :      : Rebooting...
23.702 : (16552) Error  : MQTT : Connection lost, state: Disconnected
23.907 : (17656) Info   : WIFI : Disconnected! Reason: '(8) Assoc leave'
24.109 : (17592) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'
24.210 : (17560) Info   : Reset WiFi.
24.312 : (17544) Info   : WIFI : Set WiFi to OFF

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
   �U00.458 : (29152) Info   : 

INIT : Booting version: ESP_Easy_mega_20241019_normal_ESP8266_4M1M, (GitHub Act)
00.459 : (29016) Info   : INIT : Free RAM:29016
00.462 : (28944) Info   : INIT : Soft Reboot #6 - Restart Reason: Software/Systt
00.464 : (28904) Info   : FS   : Mounting...
00.491 : (26656) Info   : FS   : SPIFFS mount successful, used 76555 bytes of 94
00.516 : (26464) Info   : ESPEasy console using ESPEasySerial
00.523 : (26264) Info   : Current Time Zone:  DST time start: 1970-03-29 02:00:n
00.537 : (26264) Info   : INIT : I2C                                            
00.538 : (26224) Info   : INIT : SPI not enabled                                
00.541 : (26152) Info   : Set Network mode: WiFi                                
00.647 : (26832) Info   : WIFI : Set WiFi to STA                                
03.039 : (25768) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'        
03.240 : (25648) Info   : WIFI : Disconnected! Reason: '(1) Unspecified'        
03.341 : (25560) Info   : Reset WiFi.                                           
05.531 : (25336) Info   : WIFI : Connecting XXXX xx:xx:xx:xx:xx:xx Ch:x (-29dBm0
06.484 : (25464) Info   : WIFI : Set WiFi to OFF                                
06.701 : (25624) Info   : WIFI : Set WiFi to STA                                
�06.815 : (25448) Info   : ESPEasy console using ESPEasySerial9:87 Ch:1 (-29dBm1
06.817 : (25464) Info   : INIT : Free RAM:25504                                 
06.851 : (24832) Info   : ESPEasy console using ESPEasySerial                   
06.853 : (24752) Info   : INFO : Plugins: 47 ["Normal","No Debug Log"] (ESP82xx)
06.854 : (25096) Info   : EVENT: System#Wake                                    
07.887 : (19488) Info   : Webserver: start                                      
07.900 : (19816) Info   : EVENT: System#Boot                                    
07.905 : (19816) Info   : ACT  : gpio,12,1                                      
07.908 : (19816) Info   : GPIO: port#12: set to 1                               
07.910 : (19704) Info   : ACT  : TimerSet,3,5                                   
07.916 : (19592) Info   : WIFI : Connected! AP: XXXXX (xx:xx:xx:xx:xx:xx) Ch: x s
07.918 : (19416) Info   : WIFI : DHCP IP: 192.xxx.xxx.xxx (S20-3) GW: 192.xxx.xxx.s
07.943 : (19464) Info   : EVENT: WiFi#APmodeDisabled                            
07.949 : (19560) Info   : Webserver: stop                                       
07.963 : (19504) Info   : Webserver: start                                      
07.989 : (19312) Info   : EVENT: WiFi#Disconnected                              
08.008 : (19280) Info   : EVENT: WiFi#Disconnected                              
08.015 : (19344) Info   : EVENT: TaskInit#Taster=1,1                            
08.074 : (17376) Error  : MQTT : Intentional reconnect                          
08.109 : (17064) Info   : MQTT : Connected to broker with client ID: xxxxxxxxxxxx
08.112 : (15448) Info   : Subscribed to: /S20_3/cmd                   
08.117 : (14672) Info   : UDP : Start listening on port 65531                   
08.118 : (14688) Info   : firstLoopConnectionsEstablished                       
08.127 : (15872) Info   : EVENT: TaskInit#Relay=2,1                             
08.146 : (17712) Info   : EVENT: WiFi#ChangedAccesspoint                        
08.153 : (17776) Info   : EVENT: WiFi#ChangedWiFichannel                        
08.165 : (17840) Info   : EVENT: WiFi#Connected                                 
[... 1 lines cut ...]
08.179 : (17760) Info   : ACT  : AsyncEvent,toggle                              
08.185 : (17912) Info   : EVENT: Relay#State=0                                  
08.193 : (17960) Info   : EVENT: Taster#Switch=0                                
08.205 : (18008) Info   : EVENT: Relay#State=1                                  
08.213 : (18056) Info   : EVENT: MQTT#Connected                                 
08.217 : (17920) Info   : ACT  : TimerSet,1,5                                   
08.225 : (18088) Info   : EVENT: p2pNode#Connected=10,'S20_3','20241019'        
08.234 : (18200) Info   : EVENT: toggle                                         
08.246 : (18104) Info   : ACT  : gpio,12,0                                      
08.249 : (18104) Info   : GPIO: port#12: set to 0                               
08.252 : (17992) Info   : ACT  : TimerSet,1,1                                   
08.285 : (18080) Info   : EVENT: Relay#State=0                                  
08.823 : (16664) Info   : NTP  : NTP replied: delay 23 ms round-trip delay: 23 0
08.826 : (16840) Info   : Time set to 1729755979.513                            
[... cut ...]
08.847 : (16800) Info   : EVENT: Time#Initialized                               
09.248 : (17240) Info   : WD   : Uptime 0  ConnectFailures 0 FreeMem 17376 WiFit
09.257 : (17392) Info   : EVENT: Rules#Timer=1,1                                
09.264 : (17256) Info   : ACT  : AsyncEvent,pub                                 
09.269 : (17456) Info   : EVENT: pub                                            
12.912 : (17472) Info   : EVENT: Rules#Timer=3,1                                
12.921 : (17400) Info   : ACT  : Let,1,1                                        
```

tonhuisman, 2024-10-24:
Do you have `Taster` also controlled via MQTT. like via the Domoticz Helper or similar?

TommoT1, 2024-10-24:
No, MQTT IS for publishing status only, well....due  to the fact /cmd ist subscribed you could issue a command, but that is not done

TommoT1, 2024-10-24:
By the way....
I'm trying this build on three nodes right now, they all worked flawless, with this build they all seem to reboot after about a day....keep watching this

TD-er, 2024-10-24:
This is really strange....
```
07.916 : (19592) Info   : WIFI : Connected! AP: XXXXX (xx:xx:xx:xx:xx:xx) Ch: x s
07.918 : (19416) Info   : WIFI : DHCP IP: 192.xxx.xxx.xxx (S20-3) GW: 192.xxx.xxx.s
07.943 : (19464) Info   : EVENT: WiFi#APmodeDisabled                            
07.949 : (19560) Info   : Webserver: stop                                       
07.963 : (19504) Info   : Webserver: start                                      
07.989 : (19312) Info   : EVENT: WiFi#Disconnected                              
08.008 : (19280) Info   : EVENT: WiFi#Disconnected                              
08.015 : (19344) Info   : EVENT: TaskInit#Taster=1,1                            
08.074 : (17376) Error  : MQTT : Intentional reconnect                          
08.109 : (17064) Info   : MQTT : Connected to broker with client ID: xxxxxxxxxxxx
08.112 : (15448) Info   : Subscribed to: /S20_3/cmd                   
08.117 : (14672) Info   : UDP : Start listening on port 65531                   
08.118 : (14688) Info   : firstLoopConnectionsEstablished    
```
Why `EVENT: WiFi#APmodeDisabled` ?
And what's up with the disconnected events....

I think this may be something that might cause issues...

Right after the sequence of connected events, I see this:
```
08.173 : (17888) Info   : EVENT: Taster#Switch=1                                
08.179 : (17760) Info   : ACT  : AsyncEvent,toggle                              
08.185 : (17912) Info   : EVENT: Relay#State=0                                  
08.193 : (17960) Info   : EVENT: Taster#Switch=0                                
08.205 : (18008) Info   : EVENT: Relay#State=1       
```
Maybe it is a good idea that we also log where the event is comming from.

TommoT1, 2024-10-24:
OK, what can I do? Loglevel Debug?

TD-er, 2024-10-24:
You can, but the event source is not yet shown in the logs.
I was just wondering what triggered the event, for example is it from the controller or read from the GPIO pin.

TommoT1, 2024-10-24:
Ah OK

closed 2024-11-03 by TommoT1