letscontrolit/ESPEasy #5153: [BUG] Rules evaluation of Let failes with Calculate: Unknown token input

Sprinterfreak, 2024-11-06:
I had to swap my ESP recently due to hardware failure. I'am pretty shure it worked like this on the previous ESP8266, because thats mostly recovered from its flash.
It's a sensor host serving 2 DHT22 for which I cobbled together homeassistant autodiscovery like this:

Rules 1
```
On MQTT#Connected Do
  Let,1,'"device":{"ids":"%sysname%","name":"Wintergarten","sw":"ESPEasy","mdl":"mega","mf":"Letscontrolit","cu":"http:/{substring:0:1:/}%ip%"}'
  Publish homeassistant/sensor/%sysname%/wintergarden_Temperature/config,'{"name":"Wintergarten Temperatur","stat_t":"climate/GF_Wintergarden/env/temperature","unit_of_meas":"C","uniq_id":"wintergarden_temperature","dev_cla":"temperature","stat_cla":"measurement","stat_cla":"measurement","avty_t":"%sysname%/availability","pl_avail":"online","pl_not_avail":"offline",%v1%}'
  Publish homeassistant/sensor/%sysname%/wintergarden_Humidity/config,'{"name":"Wintergarten Luftfeuchte","stat_t":"climate/GF_Wintergarden/env/humidity","unit_of_meas":"%","uniq_id":"wintergarden_humidity","dev_cla":"humidity","stat_cla":"measurement","stat_cla":"measurement","avty_t":"%sysname%/availability","pl_avail":"online","pl_not_avail":"offline",%v1%}'
  Publish homeassistant/sensor/%sysname%/outdoor_Temperature/config,'{"name":"Aussen Temperatur","stat_t":"climate/GF_Outdoor/env/temperature","unit_of_meas":"C","uniq_id":"outdoor_temperature","dev_cla":"temperature","stat_cla":"measurement","stat_cla":"measurement","avty_t":"%sysname%/availability","pl_avail":"online","pl_not_avail":"offline",%v1%}'
  Publish homeassistant/sensor/%sysname%/outdoor_Humidity/config,'{"name":"Aussen Luftfeuchte","stat_t":"climate/GF_Outdoor/env/humidity","unit_of_meas":"%","uniq_id":"outdoor_humidity","dev_cla":"humidity","stat_cla":"measurement","stat_cla":"measurement","avty_t":"%sysname%/availability","pl_avail":"online","pl_not_avail":"offline",%v1%}'
Endon
```

The first `Let` builds some common string, used in every following `Publish` but failes with 

Log output
```
4199: ACT : Let,1,''device':{'ids':'wintergarden-esp','name':'Wintergarten','sw':'ESPEasy','mdl':'mega','mf':'Letscontrolit','cu':'h
4225: Calculate: Unknown token input: 'device':{'ids':'wintergarden-esp','name':'Wintergarten','sw':'ESPEasy','mdl':'mega','mf':'Lets
4244: ACT : Publish homeassistant/sensor/wintergarden-esp/wintergarden_Temperature/config,'{'name':'Wintergarten Temperatur','stat_
4280: ACT : Publish homeassistant/sensor/wintergarden-esp/wintergarden_Humidity/config,'{'name':'Wintergarten Luftfeuchte','stat_t'
4326: ACT : Publish homeassistant/sensor/wintergarden-esp/outdoor_Temperature/config,'{'name':'Aussen Temperatur','stat_t':'climate
4372: ACT : Publish homeassistant/sensor/wintergarden-esp/outdoor_Humidity/config,'{'name':'Aussen Luftfeuchte','stat_t':'climate/G
```

**Expected behavior**
```
Let,1,'String to be reused'
```
should set `%v1%` to `String to be reused`, so it can be used in e.g. Publish statements.

**Used platform (please complete the following information):**
 - ESP type: ESP32
 - Build version: Build: [ESP_Easy_mega_20240822_climate_ESP32_4M316k Aug 22 2024](https://github.com/letscontrolit/ESPEasy/releases)
 - Build set climate

**Platform Specifics (when applicable, please complete the following information):**
 - Flash size: 4M
 - Brand/Model: ESP32Dev
 - Browser + OS: Firefox + Ubuntu

tonhuisman, 2024-11-06:
This is a support question that should have landed in [the Forum](https://letscontrolit.com/forum).

ESPEasy Rules, at this time, does not support string values to be assigned to variables. And hasn't in the past.
That feature is on a TODO list (for some time already).

NB: Preliminary work has started on the MQTT Auto Discovery feature, you seem to be emulating here.

TD-er, 2024-11-06:
Yep, `let` only accepts numericals, or something to calculate which then results in a numerical.

closed 2024-11-06 by TD-er