letscontrolit/ESPEasy #5159: Mini D1 ES32 flash

comtechnik, 2024-11-11:
Hello, I have a problem with a Mini D1 ESP32. I can't flash it?
Is there a guide on how to do this, what settings do I need to use in the flashing tool?
Or could it be that the Mini D1 ESP32 doesn't work with EspEasy?

Thanks for the help

TD-er, 2024-11-11:
It should work just fine.
Have you also tried the web flasher? https://td-er.nl/ESPEasy/latest/
N.B. you may need to use Chrome on Windows for this to work.

Also this isn't really a bug/feature request, but more like a generic question, which is best asked at the [forum](https://www.letscontrolit.com/forum/)

tonhuisman, 2024-11-11:
Ditto.

Some boards need to be forced into flash-mode by press and hold the `boot` button, then press and release `reset`, and after a few seconds release the `boot` button (or keep it pressed until the flash download process is completed).

TD-er, 2024-11-11:
Not sure if the Mini D1 does have a 'boot' button.
If not present, you can try to pull GPIO-0 to GND before power on and keep it till the flashing is complete.

TD-er, 2024-11-11:
You need a 4M version.
The builds on the webflasher page I linked can be split into:
- "LittleFS", using the LittleFS file system
- Without "LittleFS" in the name, using the SPIFFS file system.

It is preferred to use the LittleFS builds on newly installed nodes as that will be the file system used in the future.
When updating later, you should pick a build with the same file system.

There are "Solo1" builds, which you don't need for this one.

And there are variants like "ETH" with Ethernet support (which you don't need, but doesn't harm if you run an "ETH" build).
In your case I would use the "Climate 4M" build as it has a build for ESP32 (classic) you have.

comtechnik, 2024-11-11:
Great, thanks, it worked.
I used the Klima 4M LittleFS ETH Solo1(ESP32).
Flash: GPIO-0 with GND and permanently connected! Plugged in, and then for the flashing process, briefly pressed the reset button and off we went! :-)

TD-er, 2024-11-11:
Good to see you got it working.

Have fun! :)

closed 2024-11-11 by TD-er