letscontrolit/ESPEasy #5160: MQTT  disconnected

TommoT1, 2024-11-11:
Hi @TD-er, 

I just played around with some action builds....
I got the impression, that 
Bin-ESP32C3-normal_ESP32c3_4M316k_LittleFS_CDC_ETH-20241109_PR_5149_11759494503.zip

runs more stable on one of my c3 nodes which rebooted two or three times a day, now it is running for nearly two days without reboot.
Then I tried 

Bin-ESP32C3-normal_ESP32c3_4M316k_LittleFS_CDC_ETH-20241110_PR_5149_11767795625.zip

on another c3 node. 

Now MQTT doesn't connect anymore. It was configured to use Hostname, and the hostname is a dyndns to my home.
When I change it to use IP adress with the local adress to my MQTT Broker it works again.

Back to the 20241109 version it works again with hostname.

Just want to let you know....

TD-er, 2024-11-11:
Thanks for the test.
Keep in mind the current state of that PR is more or less "it compiles"
There is still lots of things to fix, as you currently don't get events on network changes for example.

TommoT1, 2024-11-12:
I clo

TommoT1, 2024-11-12:
I close this issue, was just to let you know

TommoT1, 2024-11-12:
Something seems still wrong with MQTT, now the node with Version from Nov 9th lost MQTT as well, Load raised to 100%
Controllerdisable of the MQTT Controller No Help>
I had to change it to MQTT Broker via ip address, and after reestablishing MQTT connection Load dropped back to about 20%

TD-er, 2024-11-12:
Yep, like I said, still a lot has to be done as there is not yet a proper disconnected event in ESPEasy to act on.
Those events are not only used for rules, but also to act on by the services running in ESPEasy.

TommoT1, 2024-11-12:
OK, I see you are fully aware of it 😃.
I'll close it again

closed 2024-11-12 by TommoT1