letscontrolit/ESPEasy #5163: [FR] Add Bluetooth support in ESPEasy

iz8mbw, 2024-11-15:
Hi!
Ok, maybe this is an hard request and an already "covered" request, but yes I ask for Bluetooth support for ESPEasy.
Personally I'm interested into a Bluetooth gateway/proxy for Bluetooth Low Energy (BLE) sensors/devices in order to pulbish the sensors/devices values via MQTT (so in WiFi). 

In my case I have several [SwitchBot Meter Plus devices](https://eu.switch-bot.com/products/switchbot-meter-plus) and [SwitchBot Indoor/Outdoor Thermo-Hygrometer](https://eu.switch-bot.com/products/switchbot-indoor-outdoor-thermo-hygrometer) (Temperature and Humidity sensor with inside the Sensirion chip, so good products) that are Bluetooth Low Energy (BLE).

Currently I'm using the [Bluetooth Proxy](https://esphome.io/components/bluetooth_proxy.html) feature of ESPHome to interface the BLE devices with Home Assistant.

What it will be nice to have in ESPEasy is the possibility to "listen for" all the Bluetooth Low Energy (BLE) discovered devices and "manage" the data with the power of ESPeasy.
In otrher words, read all the values published by the various Bluetooth Low Energy (BLE) devices/Mac-Address and _play_ with them using ESPEasy.

Any chance to have it in a future?

Thanks!

tonhuisman, 2024-11-15:
> ... and an already "covered" request, ...

Very similar request, just a few days ago...: #5158 
Please see the replies there for more information.

TD-er, 2024-11-15:
> Any chance to have it in a future?

Yes, however there are for sure concerns as mentioned in the other issue linked by Ton.

I've also looked into the ESPHome BT proxy feature, and as a matter of fact, I got a large packet from Ali yesterday with a number of BT sensors.

So it is on my "looking into it" list, however I first need to finish the WiFi/network rewrite.
But since it is affecting WiFi, I do want to keep it in mind while rewriting the WiFi/network code and that's why I also have ordered the sensors and was also looking into the proxy feature used by ESP-Home.

Another 'problem' I have is that we're the only one (next to Tasmota) who are using ESP-IDF5.x, so there is hardly any example code out there and the BT/WiFi code also has changed on IDF5.x, not only for Arduino.

iz8mbw, 2024-11-15:
> > Any chance to have it in a future?
> 
> Yes, however there are for sure concerns as mentioned in the other issue linked by Ton.

Thanks!
> I've also looked into the ESPHome BT proxy feature, and as a matter of fact, I got a large packet from Ali yesterday with a number of BT sensors.

Thanks!

> Another 'problem' I have is that we're the only one (next to Tasmota) who are using ESP-IDF5.x, so there is hardly any example code out there and the BT/WiFi code also has changed on IDF5.x, not only for Arduino.

It seems ESPHome uses esp-idf:
[image]

iz8mbw, 2024-11-15:
Coming back to SwitchBot, here the API: https://github.com/OpenWonderLabs/SwitchBotAPI-BLE/tree/latest

TD-er, 2024-11-15:
> It seems ESPHome uses esp-idf:

We all do, but there has been a massive change between ESP-IDF4.x and ESP-IDF5.x
It took me over a year to make ESPEasy work with ESP-IDF5.x
Also because while I made the transition, there was no official ESP32-Arduino 3.x yet and no PlatformIO support for it.
So I have been working with Jason2866 (from Tasmota) and people from Espressif to make it all work.
This all resulted in PIO-Arduino as there will not be official support for this Arduino 3.x/ESP-IDF5.x from PlatformIO.

Now that there is PIO-Arduino, I expect other firmwares to follow soon so they can also support ESP32-C6 for example.
However they still need to do quite a lot of work to make it all work with the new IDF5.x/Arduino3.x

chromoxdor, 2024-11-16:
> I got a large packet from Ali yesterday with a number of BT sensors.

Out of curiosity, why you want to use Bluetooth over Zigbee for sensors? I always thought Zigbee was better in terms of range and power consumption. (In terms on convenience it sure is better since all is done by one device. For my setup I run a  zigbee2tasmota device with a berry script, that translates and sends commands to ESPEasy.)

TD-er, 2024-11-16:
The other large bag was with Zigbee stuff :)

And only the C6 and H2 support 802.15.4. The H2 doesn't have WiFi and the C6 is really not that stable when running WiFi along with Thread/Zigbee.
Also there are quite a lot of Bluetooth devices out there which are quite a lot less expensive compared to Zigbee.
For example those quite small (square) temp/hum sensors sold on Ali Express for 1,80 euro, do have programming header behind the LCD and a processor you can program yourself. It also has some free pads :)
Also other MCUs do support Bluetooth, not 802.15.4

When you want devices to operate at really low power, you cannot use WiFi, so for future extensions of ESPEasy, I guess there will be builds for those MCU's as well, where you can configure them via a (WiFi/Ethernet) ESPEasy node and those will then run a very lightweight build communicating with ESPEasy as a kind of gateway/frontend/etc.

That's also why I am making plugins as uniform as possible so you can re-use code for other platforms.

There are so many things I would like to do in the future... I'm missing like a week of time every day.

chromoxdor, 2024-11-16:
> I guess there will be builds for those MCU's as well, where you can configure them via a (WiFi/Ethernet) ESPEasy node and those will then run a very lightweight build communicating with ESPEasy as a kind of gateway/frontend/etc.

That would be a dream come true :)

TD-er, 2024-11-16:
I have this plan already for at least 5 years, even before we had to move to the temporary house (demolision and rebuilding of 30 houses in our street due to earth quakes caused by gas extraction) so that's a good reference point for me to keep track of when things happened.

But for this to be done, a number of things have to be separated and restructured or else you can't re-use code on other platforms.

chromoxdor, 2024-11-16:
> But for this to be done, a number of things have to be separated and restructured or else you can't re-use code on other platforms.

This is for sure a lot of time consuming work. But for now i am really happy that ESPEasy exists anyway. It makes a lot of thing a lot easier.. :) Thank you for your effort!