letscontrolit/ESPEasy #5164: [BUG][P097] Touch isn´t working on ESP32-s3

chromoxdor, 2024-11-17:
Testet with "ESP_Easy_mega_20240822_collection_A_ESP32s3_4M316k_CDC.bin" and later on GPIO 7 and others on ADC1

The touch values are always to high (above 10000) and therefore touch is triggering constantly. 
When touched the value goes up to almost 30000.

TD-er, 2024-11-17:
Can you remove `-DBOARD_HAS_PSRAM` from the board definition.
https://github.com/letscontrolit/ESPEasy/blob/35afc25b30a4bc3c60f8604ff48c673c5aacd401/boards/esp32s3cdc-qio_qspi.json#L8

The 4M build is using `esp32s3cdc-qio_qspi` as board definition.

I never tested the touch plugin on ESP32-S3 since upgraded to IDF 5.x
Do these also trigger an event? If so, then the event queue will just fill up as it cannot handle so many events in a short time.

chromoxdor, 2024-11-17:
> Do these also trigger an event?

They do...

chromoxdor, 2024-11-17:
> Can you remove `-DBOARD_HAS_PSRAM` from the board definition.

I will try it...

chromoxdor, 2024-11-17:
> So you're essentially running out of memory...

this could cause the reboots.. but as i wrote in the forum it doesn´t have anything to do with the WiFi issue..

TD-er, 2024-11-17:
Ah OK, now I get it... I think.
Values are too high, so you can't do anything with them.

chromoxdor, 2024-11-17:
> Ah OK, now I get it... I think. Values are too high, so you can't do anything with them.

exactly :)

chromoxdor, 2024-11-17:
> Can you remove `-DBOARD_HAS_PSRAM` from the board definition.

This didn´t do anything...

chromoxdor, 2024-11-17:
> There's a base-pressure, on my S3 board without any wires connected to the pin it's roughly in the 10000 to 20000 range,

So maybe it is my understanding of the plugin that is buggy... 
I thought, the base pressure  (i assume its the same as the` Current Pressure`)must be below the `Touch Threshold`.
I have nothing connected to the pin and all I get is this with default settings:
```
6202: Touch : ADC1 ch6 (T7): 11548
6402: Touch : ADC1 ch6 (T7): 11994
6702: Touch : ADC1 ch6 (T7): 11535
7002: Touch : ADC1 ch6 (T7): 11544
7402: Touch : ADC1 ch6 (T7): 11535
7902: Touch : ADC1 ch6 (T7): 11501
8302: Touch : ADC1 ch6 (T7): 11540
9202: Touch : ADC1 ch6 (T7): 11529
```

So no events i guess. I thought these are the events. Mhmmmm strange

> Edit: I'm on the latest mega code, and using a LittleFS build, so latest IDF 5.3.x framework build.

And you have no WiFi issues?

tonhuisman, 2024-11-17:
> And you have no WiFi issues?

No, not with this unit.

On my S3-zero board (similar to what you showed in the forum) WiFi is working OK, and I haven't checked touch yet, but the board get really hot, not yet sure what is causing that. Still investigating.

chromoxdor, 2024-11-17:
> but the board get really hot, not yet sure what is causing that. Still investigating.

Mine is around 60°C and with ECO mode it´s around 40°C  which is fine i guess...

> So no events i guess. I thought these where the events but i think i am mistaken. Mhmmmm strange

Oh man, i forgot to enable rules again. 🤦‍♂️

So to get a Touch event the "touch  value" need to be higher than `Current Pressure`  + `Touch Threshold`?

chromoxdor, 2024-11-17:
> So to get a Touch event the "touch value" needs to be higher than `Current Pressure` + `Touch Threshold`?

This seems to work for me. I think I was a bit confused because I think this contradicts the documentation.
When I touch a pin, the value goes up, not down, and the values are very different from what the documentation says.

tonhuisman, 2024-11-17:
> This seems to work for me. I think I was a bit confused because I think this contradicts the documentation. When I touch a pin, the value goes up, not down, and the values are very different from [...]

Partial comment from the framework source for `touchRead()`

```
 * Read touch pad (for ESP32 values close to 0 mean touch detected /
 * for ESP32-S2/S3 higher values mean touch detected)
```

As this plugin is initially coded ca. 5 years ago, when no S2/S3 was in sight, the documentation may be somewhat outdated.

chromoxdor, 2024-11-17:
Ok… so i will try to update the documentation in the coming days.

PR #5202 (2025-01-12): Wake/Longpress/Toggle support for Touch-Plugin on compatible ESP32 devices
When i resolve conflicts in VSCode and then push it i always have to do a new PR... Sorry, I am still not able to handle this.

Here the text from the old PR:
- added extra routine for esp32 s2 and s3
- added option to wake up the esp for supported devices (tested with s3 and classic)
- added second value "state" for easier handling
- added "toggle state" option
- ~~failed epically to add a long press event because I am just a simple person, but left the remains for someone else to pick up~~
- added "long press" event
- [... cut]
closed 2025-01-12 by TD-er