letscontrolit/ESPEasy #5166: Regulator - Level Control (P021) problem with control via HTTP

tomatos-st, 2024-11-19:
I updated a few devices and encountered a problem in new builds.
I have configured Regulator - Level Control to control radiators. 
I control them via HTTP from domoticz by changing the set temperature level.
_http://192.168.100.242/control?cmd=config,task,T.reg,setlevel,25.0_
after the update it stopped working and I get a response from ESP:

_"Command not allowed from this source!Command unknown: config,task,T.reg,setlevel,25.0"_

I started going back with versions and the last version that worked was: **20231130**

from Tools/Command Output command: "config,task,T.reg,setlevel,25.0"
it works properly, there is a problem with HTTP

I use:
Build: 20241025  - Mega32-c3
Binary Filename: ESP_Easy_mega_20241025_normal_ESP32c3_4M316k_LittleFS_CDC_ETH

tonhuisman, 2024-11-19:
"That's not a bug, that's a (security) feature!"

To enhance security, i.o.w. protect the unit from (easy) tampering, some low-level commands that change configuration have been classified as 'system' commands, that can only be executed from the unit itself, not from a remote, possibly insecure, source, like via HTTP.
You can however, still execute these kind of commands from rules. And you can trigger a rule from HTTP, so that circle is round again 😃 

The solution in this case is to add in rules something like:
```
on SetLevel do
  if %eventvalue1|-99% > 15 and %eventvalue1|-99% < 30 // Include some sanity-checks
    config,task,T.reg,setlevel,%eventvalue1%
  endif
endon
```
And from Domoticz you can set the value: `http://192.168.100.242/control?cmd=event,setlevel=25.0`

tomatos-st, 2024-11-19:
Ok thanks, it works.

In the future I have to remember not to update the software at 11:00 PM when the device is deep in the attic in a hard-to-reach place and there is a risk that I will stop connecting with it after the update. And it is crucial for the operation of the house (heating) 😃

closed 2024-11-19 by tomatos-st