letscontrolit/ESPEasy #5167: Analog Current Sensor: sct-013-000 on Wemos D1 Mini

ilsicca, 2024-11-20:
Board: Wemos D1 Mini ESP8266
ESPEasy version: ESP_Easy_mega_20240822_normal_IRext_no_rx_ESP8266_4M2M.bin (I need the IR Send)

Hi! How are they? I'm try to use a current sensor sct-013-000 on Wemos D1 Mini
[image]

When I add the analog input (internal device), I only get a constant reading of 547, approx. When I turn on or off the device that I need to measure the current, the analog reading does not change either.
However, when I take the readings without ESPEasy, I can read them correctly. This is the code:

https://github.com/Mottramlabs/MQTT-Power-Sensor/blob/master/Firmware/Version%201/Basic_Current_Sensor/Basic_Current_Sensor.ino

Thanks a lot!

ilsicca, 2024-11-20:
thanks @TD-er . Do you know another way to read sct-013-000 with ESPEasy?

TD-er, 2024-11-20:
Not at this moment, however it is possible to add an extra mode to the ADC plugin to read in bursts and take the peak-to-peak values as is done in your example code.

There is a pending pull request using such clamps, but it is using some extra electronics as frontend. 
See:
https://github.com/letscontrolit/ESPEasy/pull/4009

You can also look at those [PZEM-004Tv30 modules](https://espeasy.readthedocs.io/en/latest/Plugin/P102.html), which also do some pre-processing on measured data from current clamps.
This is much more reliable as the ESP isn't equipped to constantly monitor the signal comming from those current clamps at a high enough sample rate.