letscontrolit/ESPEasy #5168: [BUG] ESPEasy work with SDS010 only in hardware serial mode.

BsB5068, 2024-11-22:
my HW esp8266  use sht30 via i2c on:
sda gpio0
scl gpio2

and SDS010 soldered to
sds tx-esp gpio5
sds rx-esp gpio4
if i choose this GPIOs in configuration sensor never began to run.

when i solder sensor to GPIO13 and use hardware serial remapped to this GPIO it works fine.
so i guess the problem in software serial.

tried ESPEasy_mega_20240822_ESP82xx_binaries.zip
and older  ESPEasy_ESP82xx_mega-20221224.zip,  ESPEasy_mega-20180126.zip

tonhuisman, 2024-11-22:
Did you double-check the RX/TX pins? Wouldn't be the first time those are configured backward... 🤔

TD-er, 2024-11-22:
Also gpio 4 & 5 on ESP8266 are by default set to the I2C bus.
Make sure it isn't set to the I2C bus on the hardware tab in the ESPEasy configuration.

BsB5068, 2024-11-22:
yes, i checked it many times. 
my i2c bus configured as
SDA gpio0
SCL gpio2
and work excellent with sht30.
so gpio 4 and 5 can't be occupied by i2C

BsB5068, 2024-11-22:
i've got PCB for this project, but want to use espeasy instead of airRohr firmware.
[image]
[https://sensor.community/en/sensors/airrohr/](url)

TD-er, 2024-11-22:
Has this sensor been used with other firmwares?
I know that for example Tasmota changes some settings in the sensor (or at least they did do this) which may cause read issues when used with ESPEasy.
However you did mention it does work with HW serial, so I guess it might not be the issue here.

Not 100% sure what it is what Tasmota changes, but it could be some sleep setting...

Ton did mention you might also need to double check the RX and TX (or simply swap the defined pins when using Software Serial) as it is quite easy to make this mistake as you need to connect the RX of the ESP to the TX of the board and vice verse.

Not sure if there is a 'beta' ESPEasy build available for ESP8266 which has this plugin included. If there is, you could also try that one as it is using a different version of the SW serial library.