letscontrolit/ESPEasy #5170: P001 Switch plugin and MQTT problem

tomatos-st, 2024-12-01:
-  ESP type:  ESP32-S3
 - Build version: ESP_Easy_mega_20241025_normal_ESP32c3_4M316k_LittleFS_CDC_ETH

I have a problem with communication via MQTT (with Domoticz)
At first I detected strange "couplings": I turn on one GPIO and another one turns off additionally.
I created new switches in Domoticz and made a set "Switch input - Switch" in tasks 1-11 and started to find. Here's what I got:
[image]
[image]

if I do task1 GPIO-0, Task 2, GPIO-1... Task 11 GPIO-10 in sequence is ok, but when I shuffle it starts to get interesting..
- I send from domoticz via MQTT to address 750, GPIO-2 should be enabled on task 1, but GPIO-0 is enabled on task 3
- I send MQTT:749, GPIO-6 should be enabled on task 2, but GPIO-0 is enabled on task 2
- I send MQTT:748, GPIO-0 should be enabled on task 3, but GPIO-2 is enabled on task 1
- I send MQTT:747, GPIO-1 should be enabled on task 4, but GPIO-3 is enabled on task 10
- I send MQTT:746, GPIO-2 should be enabled on task 5 but GPIO-4 iis enabled on task 7
- I send MQTT:745 GPIO-9 should be enabled on task 6 but GPIO-5 is enabled on task 5
- I send MQTT:744 GPIO-4 should be enabled on task 7 but GPIO-6 is enabled on task 2
- I send MQTT:743 GPIO-10 should be enabled on task 8 but GPIO-7 is enabled on task 9
- I send MQTT:742 GPIO-7 should be enabled on task 9 but GPIO-8 is is enabled on task 11
- I send MQTT:741 GPIO-3 should be enabled on task 10 but GPIO-9 is enabled on task 6
- I send MQTT:740 GPIO-8 should be enabled on task 11, but GPIO-10 is enabled on task 8

After analysis, it turns out that the task number is confused with the port number:
and so, for example: MQTT:744 is on task 7 and instead of turning on the assigned port, it turns on the GPIO-6 port (task number -1 - ports are numbered from 0 and tasks from 1)... and so on

tonhuisman, 2024-12-01:
> After analysis, it turns out that the task number is confused with the port number:
> and so, for example: MQTT:744 is on task 7 and instead of turning on the assigned port, it turns on the GPIO-6 port (task number -1 - ports are numbered from 0 and tasks from 1)... and so on

I'm already working on the MQTT plugins, so I'll have a look at this issue.

tonhuisman, 2024-12-01:
Test build available from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12108829738)

tomatos-st, 2024-12-01:
It works :) 
Just so you know how much I struggled before I figured out what was wrong, I was looking for connections in Domoticz (rules, conditions, events, connections), nothing suited me, I wasted a whole day :(

TD-er, 2024-12-01:
> I'm going to bed. :)

Me too :)
Thanks for reporting

PR #5171 (2024-12-01): [C002] Bugfix: Use correct GPIO pin for controlling P001
Resolves #5170 

Bugfix:
- Domoticz MQTT should use the correct GPIO pin for controlling Switch plugin (P001)
closed 2024-12-01 by TD-er