letscontrolit/ESPEasy #5172: [BUG] AHT10 unable to initialize

zetapersei, 2024-12-02:
When i add the aht10 device at startup is "unable to initialize" (debug messages in the serial line).

I verified  i2c line  signals  between esp8266 and device. The initialization command (0xE1 0xBE 0x0800)  is not recognized.
I tried to replace this command with softreset command 0xBA modifing the P105_data_struct.cpp file:

[image]

after this change the device communicates regularly.

Used platform:

Build: 20241130 - Mega
System Libraries: ESP82xx Core 3.1.2, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.3 PUYA support
Git Build: My Build: Nov 30 2024 16:17:21
Build Platform:  Windows-10-10.0.19045-SP0

ESP Chip Frequency: 80 [MHz]
Flash Chip Real Size: 4096 [kB]
ESP Chip Model:  ESP8266EX
ESP Chip Cores: |1
ESP Board Name: Espressif Generic ESP8266 ESPEasy 1M Flash 128k FS

tonhuisman, 2024-12-02:
A few things to think about:
- AHT10 is not one of the most accurate sensors.
- That initialization sequence is obtained from a documented source, the manufacturer datasheet.
- There are many clones on the market, usually for a slightly lower price, that are _almost_ compatible. This might be one of those clones.

A possibly better solution is to add a setting to the plugin to use an alternative reset sequence, so original sensors are still usable, as your sensor seems to be an exception.

TD-er, 2024-12-02:
Please read this warning section in the docs regarding this AHT10 sensor: https://espeasy.readthedocs.io/en/latest/Plugin/P105.html#p105-page

In short: This specific sensor does NOT communicate according to the I2C standard and you cannot use it with anything else on the I2C bus.

So I guess there is not much we can do about this as this sensor is not obeying any standard.

zetapersei, 2024-12-02:
I agree, the my AHT10 is probabily clone of the original. I tried to use some libraries: if use the Adafruit library (https://github.com/adafruit/Adafruit_AHTX0) the sensor work fine but anyway is out of the standard....you cannot use it with anything else on the I2C bus.
 Adding a setting to the plugin to use an alternative reset sequence might be a solution.

tonhuisman, 2024-12-02:
That Adafruit lib uses basically the same initialization byte-sequence as ESPEasy does, a difference is that we don't wait indefinitely for the sensor to respond/report as ready, as that's against the time-scheduled way ESPEasy works.
So it's a bit surprising that it doesn't work with ESPEasy.

As TD-er said, you can not have other devices on the same bus when using an AHT10, that's a hardware/sensor-firmware issue that can't be solved by either Adafruit or us.
If you need to connect multiple devices on the I2C bus, an (untested here, but proven) solution is to use an I2C multiplexer and connect the AHT10 to its own, separate channel, and the other I2C devices to other channels of the multiplexer. That way it can't interfere with other devices.

TD-er, 2024-12-02:
Maybe we could add a checkbox or combobox to send only 1 byte as you suggested.
But still I have no idea what devices need this or whether it may cause issues on devices that don't need this but are supported by this plugin.

zetapersei, 2024-12-03:
I've observed some differences in the initialization procedure when comparing the sequence between ESPeasy and the Adafruit drive. The primary distinction lies in the soft reset during the startup process (I have attached the tabular CSV files  of the I2C line).
[aht_test_adafruit.csv](https://github.com/user-attachments/files/17991987/aht_test_adafruit.csv)

[aht_test_espeasy.csv](https://github.com/user-attachments/files/17991973/aht_test_espeasy.csv)

TD-er, 2024-12-03:
Can you also attach some other I2C device to the same bus to see if this version you have can play nice with other devices on the same bus?
I got the impression you might have some slightly different version at hand.

Preferrably a BME280 or SSD1306 OLED display as those are known to 'always just work'.

Things to check for:
- Does the I2C bus 'get stuck' at some point? (ESPEasy might want to mitigate this, so you can also disable this behavior on the tools->Advanced page)
- Does the 'other device' work? (OLED displays may work as you don't read from them, only write to them)
- Are all devices seen when scanning the I2C bus?
- Does the AHT10 work stable with some other device connected to it? Does the other one?
- Do you need to force the I2C bus at 100 kHz to make stuff work stable?
- Does the AHT10 keep working after you scan the I2C bus? (and other devices?)

zetapersei, 2024-12-03:
> Can you also attach some other I2C device to the same bus to see if this version you have can play nice with other devices on the same bus? I got the impression you might have some slightly different version at hand.
> 
> Preferrably a BME280 or SSD1306 OLED display as those are known to 'always just work'.
>  
> **I attached one SSD1306 togheter aht10 in the same bus**

> Things to check for:
> 
> * Does the I2C bus 'get stuck' at some point? (ESPEasy might want to mitigate this, so you can also disable this behavior on the tools->Advanced page)  --> **NO**
> * Does the 'other device' work? (OLED displays may work as you don't read from them, only write to them)  --> **YES**
> * Are all devices seen when scanning the I2C bus? --> **YES**
> * Does the AHT10 work stable with some other device connected to it? Does the other one? --> **YES**
> * Do you need to force the I2C bus at 100 kHz to make stuff work stable? --> **NO**
> * Does the AHT10 keep working after you scan the I2C bus? (and other devices?) --> **YES**

TD-er, 2024-12-03:
Hmm like I said, sounds like a different device.
Either it isn't a AHT10, but perhaps a different version, or one of those clones was not able to emulate the truly bad behavior of the original :)

tonhuisman, 2024-12-03:
I've added an option for alternative initialization for AHT10 sensors (only) sending the soft-reset as suggested.

@zetapersei Can you test with a build from this [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12148649634) once that's completed?

zetapersei, 2024-12-04:
Yes, of course

TD-er, 2024-12-04:
Can you also post some pictures of the AHT10 chip?
Just to make sure it actually _is_ an AHT10 and not some other variant.

zetapersei, 2024-12-04:
> Can you also post some pictures of the AHT10 chip? Just to make sure it actually _is_ an AHT10 and not some other variant.

I have made a purchase on Aliexpress at a very low price; theoretically, the chip should be from Asair, but I am not sure.

[image]

[image]

TD-er, 2024-12-04:
If you look at the picture in [this datasheet](https://www.handsontec.com/dataspecs/sensor/AHT10.pdf), it seems there should be 3 lines imprinted in the metal casing.

[image]

Not sure this is the correct datasheet though...

zetapersei, 2024-12-04:
https://www.aliexpress.com/item/1005006954884045.html?spm=a2g0o.store_pc_home.0.0.461ca2e32TAEla&pdp_npi=4%40dis%21EUR%21%E2%82%AC%205%2C76%21%E2%82%AC%201%2C92%21%21%2143.11%2114.36%21%40212fabfc17333270949107935dd618%2112000038842821660%21sh%21IT%21867816323%21X

zetapersei, 2024-12-05:
> I've added an option for alternative initialization for AHT10 sensors (only) sending the soft-reset as suggested.
> 
> @zetapersei Can you test with a build from this [Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/12148649634) once that's completed?

Result of my test:

console  with alternative no selected:
[image]

console with alternative selected:
[image]

This is my I2C data; I tested it at 100kHz and 400kHz, and it works fine for both.
[i2c_with_no_ck_alt.csv](https://github.com/user-attachments/files/18020347/i2c_with_no_ck_alt.csv)

[i2c_with_ck_alt.csv](https://github.com/user-attachments/files/18020361/i2c_with_ck_alt.csv)

PR #5173 (2024-12-09): [P105] Add Alternative initialization option for AHT10 clones
Resolves #5172 

Features:
- Add alternative initialization option to only send soft-reset to AHT10 clones
- Update documentation

TODO:
- [x] Testing by requester ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/5172#issuecomment-2519708760))
closed 2024-12-09 by TD-er