letscontrolit/ESPEasy #5174: [Test build] Testers for new upcoming build (X-mas '24)

TD-er, 2024-12-10:
I've just uploaded the [latest GH Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/12245479354)  to the [web flasher](https://td-er.nl/ESPEasy/latest/)

So I would like to summon the very appreciated testing crew :)
@fly74 @ghtester @chemmex @alex-kiev @iz8mbw @Wookbert @Dickesplumpel @thomastech 

I had planned to also have the complete network/WiFi rewrite in this build, but it is taking quite a bit longer than expected.
This is also because of other reasons which prevent me from working on ESPEasy for hours at a time. Since the rewrite takes quite a lot of time to get 'into it' again each time I work on it, it doesn't make sense to do anything on it when I only have an hour to spare.

Anyway there have been quite a lot of fixes and other changes since the last build, so let's prepare for some X-mas build.

To make clear what the differences are between SPIFFS and LittleFS:
- ESP32-xx LittleFS builds are based on ESP-IDF 5.3 code
- ESP32-xx SPIFFS builds are based on ESP-IDF 5.1 code

As can be seen, both do not differ a lot anymore and on the next build this difference might even be completely gone as I can hopefully then use pioarduino with the custom SDK flags made by Jason2866 and Staars, both from Tasmota.

Just as a reminder for those who want to upgrade from an April build:

> There are lots and lots of changes, but the main tricky one is that it might be impossible to OTA update larger builds from the > April release as there was a bug in the Arduino code which did not properly use timeouts when handling HTTP requests.
> 
> If this is an issue, to update any "max" builds, I will add intermediate builds for those typically 16M and 8M builds which do not contain any plugin or controller to make them as small as possible.
> This way it is more likely you can successfully OTA update such units without the need to hook them up via serial.
>
> So please let me know if this is indeed an issue (and maybe also for "4M builds LittleFS").
> N.B. this does only apply to "LittleFS" builds when OTA updating from the April release. (or test builds from around that time)

iz8mbw, 2024-12-10:
@TD-er instead to use the [latest GH Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/12245479354), may I build by myself a Custom Build using the latest commit available now on Github?

TD-er, 2024-12-10:
Sure
It is the same code base and you may find other issues which may not show when using the standard builds.

iz8mbw, 2024-12-10:
OK, going to build and test!

fly74, 2024-12-10:
Test activated for:

```
.
├── ./ESP32-D0WDQ5_neopixel_ESP32_4M316k_ETH
│   └── ./ESP32-D0WDQ5_neopixel_ESP32_4M316k_ETH/ESP_Easy_mega_20241210_neopixel_ESP32_4M316k_ETH.bin
├── ./ESP32-D0WDQ6_neopixel_ESP32_4M316k
│   └── ./ESP32-D0WDQ6_neopixel_ESP32_4M316k/ESP_Easy_mega_20241210_neopixel_ESP32_4M316k.bin
├── ./ESP32-D0WDQ6_normal_ESP32_4M316k
│   └── ./ESP32-D0WDQ6_normal_ESP32_4M316k/ESP_Easy_mega_20241210_normal_ESP32_4M316k.bin
├── ./ESP8266_climate_ESP8266_4M1M
│   └── ./ESP8266_climate_ESP8266_4M1M/ESP_Easy_mega_20241210_climate_ESP8266_4M1M.bin
├── ./ESP8266_minimal_core_312_1M_OTA_Domoticz_MQTT
│   └── ./ESP8266_minimal_core_312_1M_OTA_Domoticz_MQTT/ESP_Easy_mega_20241210_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT.bin
├── ./ESP8266_neopixel_ESP8266_4M1M
│   └── ./ESP8266_neopixel_ESP8266_4M1M/ESP_Easy_mega_20241210_neopixel_ESP8266_4M1M.bin
└── ./ESP8266_normal_ESP8266_4M1M
   └── ./ESP8266_normal_ESP8266_4M1M/ESP_Easy_mega_20241210_normal_ESP8266_4M1M.bin

```

iz8mbw, 2024-12-10:
All OK here on ESP32 boards (connected on WiFi) using Custum Builds!!!
Thanks @TD-er

Dickesplumpel, 2024-12-10:
OTA update from ESP_Easy_mega_20230623_normal_ESP32s2_4M316k_CDC to ESP_Easy_mega_20241210_normal_ESP32s2_4M316k_CDC
Fresh Webui install with ESP_Easy_mega_20241210_normal_ESP8266_4M1M on 12F and ESP_Easy_mega_20241210_normal_ESP32_4M316k on ESP32-WROOM_32

No issues so far. In the past I had often trouble setting up wifi settings with putty, now it seems working pretty good.

fly74, 2024-12-10:
> Test activated for from all.zip:
> 
> ```
> .
> ├── ./ESP32-D0WDQ5_neopixel_ESP32_4M316k_ETH
> │   └── ./ESP32-D0WDQ5_neopixel_ESP32_4M316k_ETH/ESP_Easy_mega_20241210_neopixel_ESP32_4M316k_ETH.bin
> ├── ./ESP32-D0WDQ6_neopixel_ESP32_4M316k
> │   └── ./ESP32-D0WDQ6_neopixel_ESP32_4M316k/ESP_Easy_mega_20241210_neopixel_ESP32_4M316k.bin
> ├── ./ESP32-D0WDQ6_normal_ESP32_4M316k
> │   └── ./ESP32-D0WDQ6_normal_ESP32_4M316k/ESP_Easy_mega_20241210_normal_ESP32_4M316k.bin
> ├── ./ESP8266_climate_ESP8266_4M1M
> │   └── ./ESP8266_climate_ESP8266_4M1M/ESP_Easy_mega_20241210_climate_ESP8266_4M1M.bin
> ├── ./ESP8266_minimal_core_312_1M_OTA_Domoticz_MQTT
> │   └── ./ESP8266_minimal_core_312_1M_OTA_Domoticz_MQTT/ESP_Easy_mega_20241210_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT.bin
> ├── ./ESP8266_neopixel_ESP8266_4M1M
> │   └── ./ESP8266_neopixel_ESP8266_4M1M/ESP_Easy_mega_20241210_neopixel_ESP8266_4M1M.bin
> └── ./ESP8266_normal_ESP8266_4M1M
>    └── ./ESP8266_normal_ESP8266_4M1M/ESP_Easy_mega_20241210_normal_ESP8266_4M1M.bin
> ```

No issues at all.

Dickesplumpel, 2024-12-10:
ESP_Easy_mega_20240331_normal_ESP32c3_4M316k_CDC to ESP_Easy_mega_20241012_normal_ESP32c3_4M316k_CDC on ESP32-c3 super mini
I'm missing ESP_Easy_mega_20241210_climate_ESP32solo1_4M316k_LittleFS, ESP_Easy_mega_20241210_climate_ESP32solo1_4M316k_LittleFS_ETH doesn't work on shelly-plus1, bootloop.

Dickesplumpel, 2024-12-10:
Update from ESP_Easy_mega_20231223_climate_ESP32solo1_4M316k_LittleFS to ESP_Easy_mega_20241210_climate_ESP32solo1_4M316k_LittleFS_ETH gone wrong. A flash with ESP_Easy_mega_20241210_climate_ESP32solo1_4M316k_LittleFS_ETH.factory doesn't work ether. I will try it again tomorrow with normal build.

TD-er, 2024-12-10:
Check!
I will later this evening update the ESP-IDF version for the solo1, and will post here when it is ready.

chemmex, 2024-12-10:
Ready to test specifically local AP restart after loss of main WiFi connection.

 ESP_Easy_mega_20241210_normal_ESP32c3_4M316k_LittleFS_CDC_ETH.factory.bin, doesn't restart so far

chemmex, 2024-12-10:
> Correct, as that is part of the WiFi/network rewrite which is not done yet

Oh, I see, I mistakenly thought you are doing it incrementally
Will wait for that fully armed.

iz8mbw, 2024-12-10:
and after the WiFi rewrite there will be a surprise/present made by @TD-er : the Bluetooth Low Energy support in ESPEasy :-)

Dickesplumpel, 2024-12-11:
Fresh clean install with ESP_Easy_mega_20241210_normal_ESP32solo1_4M316k_LittleFS_ETH.factory.bin, serial says this block in a loop:

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13056
load:0x40080400,len:4
load:0x40080404,len:2724
entry 0x40080534

TD-er, 2024-12-11:
Talked with Jason2866 and he pointed out the error I made.
So this build should hopefully fix the solo1 crashing: https://github.com/letscontrolit/ESPEasy/actions/runs/12273878227

fly74, 2024-12-11:
Is it only for the solo1 oder a globally change?

Dickesplumpel, 2024-12-11:
No, ESP_Easy_mega_20240822_normal_ESP32solo1_4M316k_LittleFS_ETH works fine, upgrade to Bin-ESP32SOLO1-normal_ESP32solo1_4M316k_LittleFS_ETH-20241211_PR_5175_12273878227 or fresh install with factory.bin gets same result as earlier described.

TD-er, 2024-12-11:
Jason also confirmed this build is failing on his Solo1.
So have to dig further to see why this is failing.

fly74, 2024-12-11:
It looks like we have an issue with sending values with http eg.

/usr/bin/curl http://esp/control?cmd=TaskValueSet,7,1,40
curl: (56) Recv failure: Connection reset by peer

often results in this error.

TD-er, 2024-12-11:
Which build?
LittleFS or SPIFFS?

fly74, 2024-12-11:
ESP_Easy_mega_20241210_normal_ESP32_4M316k.bin

TD-er, 2024-12-11:
Is it possible to test it on some (other) node using a LittleFS build?
This fix is not included in the SDK used for SPIFFS builds: https://github.com/espressif/arduino-esp32/pull/10640

fly74, 2024-12-11:
Back to ESP_Easy_mega_20240822_normal_ESP32_4M316k works fine.

fly74, 2024-12-11:
> Is it possible to test it on some (other) node using a LittleFS build? This fix is not included in the SDK used for SPIFFS builds: [espressif/arduino-esp32#10640](https://github.com/espressif/arduino-esp32/pull/10640)

no running ESP with SPIFFS atm, but can you build one for this bin?

fly74, 2024-12-11:
Getting the json and the webinterface is also from disconnects affected.

TD-er, 2024-12-11:
> > Is it possible to test it on some (other) node using a LittleFS build? This fix is not included in the SDK used for SPIFFS builds: [espressif/arduino-esp32#10640](https://github.com/espressif/arduino-esp32/pull/10640)
> 
> no running ESP with SPIFFS atm, but can you build one for this bin?

It is available: 
[image]

Also in the `all.zip` file on the webflasher page: https://td-er.nl/ESPEasy/latest/

If you want to convert a node, you should first download a tar file via "Backup Files" on the Tools page.
Make sure you actually have the tar file downloaded, as Chrome for example will pop-up a tooltip asking if you would like to keep the file and if you don't agree, it will not be downloaded.

Then you can flash the LittleFS build, which will format the file system and should keep the WiFi credentials so it will show up connected to your network.
Then you can via the "File Browser" button on the tools page upload the tar file and then reboot the node.

fly74, 2024-12-11:
The WIFI wasn't still there, but via AP I could restore the tar-file. But same issues with the unstable connection. How can I revert to the non SPIFFS from 240822?

fly74, 2024-12-11:
works with flash and AP restore.

fly74, 2024-12-11:
same with downloading backups:

wget http://esp/download -P SAVE --user=xxx --password=xxx --content-disposition
--2024-12-11 21:28:15--  http://esp/download
Resolving esp (esp)... 172.xxxxxx
Connecting to esp (esp)|172.xxxxxx|:80... connected.
HTTP request sent, awaiting response... 401 Unauthorized
Authentication selected: Digest realm="Login Required (default user: admin)", qop="auth", nonce="xxxxxxxxx", opaque="xxxxxxxxx"
Connecting to esp (esp)|172.xxxxxx|:80... connected.
HTTP request sent, awaiting response... Read error (Connection reset by peer) in headers.
Retrying.

TD-er, 2024-12-11:
And this was working well on the August build?

fly74, 2024-12-11:
yes, when I revert it works.

TD-er, 2024-12-11:
Can you disable the password to see if it makes a difference?

TD-er, 2024-12-11:
The thing is that this will be an extra call.
Just to know where we should start looking.

fly74, 2024-12-11:
You're right, without password it works.

fly74, 2024-12-11:
Can you reproduce it ?

fly74, 2024-12-13:
It also affects the MQTT traffic with loss of values.

TD-er, 2024-12-16:
@fly74 
I made a quick work-around to at least fix it for web page loading.
This also should fix it for the issue where you might not be able to save rules (mainly via Ethernet, but also sometimes via WiFi)

https://github.com/letscontrolit/ESPEasy/actions/runs/12356077020

There is a pending PR I made on the Arduino code, which should fix it for all network related issues.
See: https://github.com/espressif/arduino-esp32/pull/10737

fly74, 2024-12-16:
Does it fix the bad MQTT connections too?

fly74, 2024-12-16:
So I have to wait. All of them are MQTT nodes.

TD-er, 2024-12-16:
You also mentioned issues handling commands using curl, which should be fixed by this.

fly74, 2024-12-16:
I'll give it a short test with one node.

thomastech, 2024-12-16:
Sorry for being late to the party. @TD-er, It's not clear to me if you are looking for SPIFFS or LittleFS version testing. Or both.

In case it helps, I loaded  ESP_Easy_mega_20241210_collection_B_ESP32_4M316k and all is well.

- Thomas

fly74, 2024-12-16:
> @fly74 I made a quick work-around to at least fix it for web page loading. This also should fix it for the issue where you might not be able to save rules (mainly via Ethernet, but also sometimes via WiFi)
> 
> https://github.com/letscontrolit/ESPEasy/actions/runs/12356077020
> 
> There is a pending PR I made on the Arduino code, which should fix it for all network related issues. See: [espressif/arduino-esp32#10737](https://github.com/espressif/arduino-esp32/pull/10737)

look red:
[image]

TD-er, 2024-12-16:
Jason2866 did make a new platform build for me with the (now merged) PR I made into the Arduino repo.
So after putting Lilly to bed, I will make a new build using these new platform builds (hopefully also a fixed Solo1 build, though not tested yet)

TD-er, 2024-12-16:
Arduino 3.1.0 is now released: https://github.com/espressif/arduino-esp32/releases/tag/3.1.0
And my PR just made it in :)

Now building with the new Arduino release: https://github.com/letscontrolit/ESPEasy/actions/runs/12360555212

Hopefully this also fixes the Solo1 builds.

I will do some quick tests myself, then upload it to the web flasher and will let you all know as soon as it is ready to be tested.
Now I have to find the nodes which are hard to reach for serial flashing when things go wrong, just to be sure it is properly tested :)

TD-er, 2024-12-17:
@Dickesplumpel Can you also test to see if the Solo1 now works?
I have uploaded the latest build here: https://td-er.nl/ESPEasy/latest/

fly74, 2024-12-17:
> Uploaded latest build here: https://td-er.nl/ESPEasy/latest/
> 
> ESP32 SPIFFS builds also include the fix I discussed above, even though those are Arduino 3.0.x/ESP-IDF 5.1.x
> 
> * Loading pages should no longer stall when making multiple calls shortly after eachother (e.g. login, save rules, loading CSS)
> * MQTT (re)connect issues are hopefully fixed now as they probably suffered from the same bug.
> * Solo1 builds hopefully will work now (different issue, PEBKAC related, we don't talk about it....)
> 
> And some other fixes made by Ton as he continuously implements new features and fixes other bugs...
> 
> So please test whatever build you can find from the above URL. Currently it is the same code as in the `mega` branch as I just merged the PR used for this test build.

I tested:

ESP_Easy_mega_20241216_normal_ESP8266_4M1M
ESP_Easy_mega_20241216_normal_ESP32_4M316k

Webinterface responsibility and MQTT is looking good!

Dickesplumpel, 2024-12-17:
Solo1 no change, same bootloop.

fly74, 2024-12-17:
These also good:

ESP_Easy_mega_20241216_climate_ESP8266_4M1M
ESP_Easy_mega_20241216_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT
ESP_Easy_mega_20241216_neopixel_ESP32_4M316k

chemmex, 2024-12-17:
Is that normal that in custom builds the AP name is not taken from the DEFAULT_NAME, but uses "ESPEasy-" instead?

fly74, 2024-12-17:
ESP_Easy_mega_20241216_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT had an unexpected reboot, but the minimal has no logging.

chemmex, 2024-12-17:
Also self-compiled custom builds: when DEFAULT_USE_NTP is defined as true, it turns out to be unchecked in the Advanced menu, hence no NTP is activated

tonhuisman, 2024-12-17:
> the AP name is not taken from the DEFAULT_NAME define

> when DEFAULT_USE_NTP is defined as true, it turns out to be unchecked

Thanks for reporting/reminding of these issues. I assume these are older issues, not directly related to the current planned release.
I'll have a look at those later today.

I assume you are aware that these predefined settings are _only_ to be applied when starting with a blank filesystem (or erased flash).

TD-er, 2024-12-17:
> ESP_Easy_mega_20241216_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT had an unexpected reboot, but the minimal has no logging.

What does the sysinfo page tell you about the last action before reboot?
Did this node run stable on older builds? The reason I'm asking is because 1M devices may already be quite old and thus some components may start showing their age. Especially capacitors in net powered devices like Sonoff Basic units.

fly74, 2024-12-17:
> ESP_Easy_mega_20241216_minimal_core_312_ESP8266_1M_OTA_Domoticz_MQTT had an unexpected reboot, but the minimal has no logging.

Not seen again after a cold boot.

fly74, 2024-12-17:
> What does the sysinfo page tell you about the last action before reboot?
>
> Did this node run stable on older builds? The reason I'm asking is because 1M devices may already be quite old and thus some [...]

Was stable before. But as I wrote after a cold boot all is fine. Maybe the soft boot after flashing wasn't enough.

fly74, 2024-12-17:
> Was a previous build also using core 3.1.2?

Yes, from last official 0822.

iz8mbw, 2024-12-17:
Built a Custom Build on ESP32 and after some time the Web Interface does not respond:
```
root@rpi3:~# curl -v http://192.168.147.33/
* !!! WARNING !!!
* This is a debug build of libcurl, do not use in production.
* STATE: INIT => SETUP handle 0x556e886d58; line 1950
* STATE: SETUP => CONNECT handle 0x556e886d58; line 1966
* Added connection 0. The cache now contains 1 members
* STATE: CONNECT => CONNECTING handle 0x556e886d58; line 2002
*   Trying 192.168.147.33:80...
* connect to 192.168.147.33 port 80 from 192.168.147.2 port 53876 failed: No route to host
* Failed to connect to 192.168.147.33 port 80 after 3063 ms: Could not connect to server
* multi_done[CONNECTING]: status: 7 prem: 1 done: 0
* multi_done, not reusing connection=0, forbid=0, close=0, premature=1, conn_multiplex=0
* The cache now contains 0 members
* Curl_disconnect(conn #0, aborted=1)
* closing connection #0
* [CCACHE] closing #0
curl: (7) Failed to connect to 192.168.147.33
```

chemmex, 2024-12-17:
> Thanks for reporting/reminding of these issues. I assume these are older issues, not directly related to the current planned release. I'll have a look at those later today.

Sorry, I did not keep track of old issues, however, it might well be myself who reported the AP name issue in the first place

> I assume you are aware that these predefined settings are _only_ to be applied when starting with a blank filesystem (or erased flash).

Of course, I always clean the flash before testing new firmware.

I re-checked these issues on different boards and platforms. Only ESP8266 builds are affected, both custom_274_ESP8266_4M1M and custom_312_ESP8266_4M1M. It seems that new configuration is generated only partially as such items as DEFAULT_SERVER_HOST and others are applied, while DEFAULT_NAME and NTP setting are not

TD-er, 2024-12-17:
> Built a Custom Build on ESP32 and after some time the Web Interface does not respond:
> 

What is set in the Custom.h?
When was the build made? Yesterday there has been an important fix which for sure is needed or else all kinds of network related issues will occur.

[26 more replies]

closed 2024-12-23 by TD-er