letscontrolit/ESPEasy #5176: Controller MQTT

fly74, 2024-12-13:
Is it intended to have only host and ip connect the broker? 

[image]

I want to put the broker behind a ha-proxy with an URL like server:3333/broker eg.

tonhuisman, 2024-12-13:
Did some investigation, but it looks like the MQTT Client library (PubSubClient) does _not_ support the use of a full url. Only a `host:port` is supported (provided in 2 separate field), where the host is a domain name or an ip-address. Most likely this a 'thing' in the MQTT world.

AFAICS, your proxy should be able to proxy _any_ port to the external MQTT server on any port you want to connect to (probably using port 1883 or 8883 when using TLS). And you can choose any internal port to use on HA-Proxy, of course 😃.

fly74, 2024-12-13:
@tonhuisman Thanks, I thought this could be the reason.

closed 2024-12-13 by fly74